evcc-io/evcc #20422: kurzzeitiges Abfallen der Werte von SOC Hausakku und Erzeugung PV1

StevieC121176, 2025-04-05:
### Describe the bug

mit der neuen Nightly Version `0.202.1 (fd48bf35)` fällt alle 1-2 Minuten der Wert für den SOC meines Hausakkus auf 0% und die Erzeugung einer der Wechselrichter (an dem die Batterie hängt) auf 0W. nach einigen Sekunden werden die Werte wieder korrekt angezeigt. 
Ein Defekt ist auszuschließen da das mit der letzten EVCC staple Version nicht passiert. Ich kann auch zwischen den beiden Versionen wechseln und es wird immer nur bei der nightly `0.202.1 (fd48bf35)` dieser Fehler angezeigt

### Steps to reproduce

1.EVCC starten
2.kurz danach tritt der Fehler alle 1-2 Minuten auf
3.
...

### Configuration details

```yaml
2 Kostal Plenticore plus und KSEM Smartmeter

´´´meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: pv  
  modbus: tcpip
  endianness: little
  minsoc: 20
  maxsoc: 100
  watchdog: 60s
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.118  
  port: 1502  
  usage: battery  
  modbus: tcpip
  endianness: little 
  capacity: 12.8 
  minsoc: 20 
  maxsoc: 100  
  watchdog: 60s 
  name: battery
- type: template
  template: kostal-ksem
  id: 71
  host: 192.168.178.116
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv2´´´
```

### Log details

```text
logfiles sh. weiter unten
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1 (fd48bf35)

StevieC121176, 2025-04-05:
[evcc-20250405-202154-error.log](https://github.com/user-attachments/files/19618148/evcc-20250405-202154-error.log)
[evcc-20250405-202047-trace.log](https://github.com/user-attachments/files/19618149/evcc-20250405-202047-trace.log)

andig, 2025-04-05:
Und gibts einen offensichtlichen Fehler im Log?

StevieC121176, 2025-04-05:
> Und gibts einen offensichtlichen Fehler im Log?

Ich sehe da Timeout Fehler, aber wie gesagt, Sobald ich auf die EVCC stable Version wechsle funktioniert es ohne Probleme.

andig, 2025-04-06:
@StevieC121176 könntest du bitte für info@evcc.io eine Portweiterleitung einrichten?

StevieC121176, 2025-04-06:
> [@StevieC121176](https://github.com/StevieC121176) könntest du bitte für [info@evcc.io](mailto:info@evcc.io) eine Portweiterleitung einrichten?

Kann ich gerne machen, geht aber erst am Dienstag (bin unterwegs)

StevieC121176, 2025-04-08:
> [@StevieC121176](https://github.com/StevieC121176) könntest du bitte für [info@evcc.io](mailto:info@evcc.io) eine Portweiterleitung einrichten?

Ist erledigt, ich hoffe das funktioniert so.

closed 2025-04-09 by StevieC121176

andig, 2025-04-20:
Dann lass uns mal probieren, ob es so besser klappt- leider kann ich den Fehler den Du siehst remote nicht reproduzieren.

StevieC121176, 2025-04-20:
> Dann lass uns mal probieren, ob es so besser klappt- leider kann ich den Fehler den Du siehst remote nicht reproduzieren.

mit der Version 0.203.1 (12d8e448) funktioniert es bis jetzt ohne Aussetzer. Ich hab die Version 15.45 Uhr installiert und seit dem keine Probleme gehabt, weder bei PV noch bei Batterie Soc. (sh. Screenshot)

[image]