evcc-io/evcc #20435: Ladepunkt Priorität wird nicht berücksichtigt

docolli, 2025-04-06:
### Describe the bug

lp-1 ist das Auto und steht auf  PV, lp-3 ist mein Heizstab, steht auch auf PV.
lp-1 hat Priorität 20, lp-3 hat 1. Also lp-1 hat die höher Priorität als lp-3.

Es ist genügend PV Überschuss (>8000W) vorhanden, der lp-3 nimmt 5600W (max) auf, aber der lp-1 wird nicht gestartet. Auch lp-1 kurz auf Aus und dann wieder auf PV stellen funktioniert nicht.
Erst wenn lp-3 auf Aus steht und lp-1 die Ladung gestartet hat und ich dann lp-3 wieder auf PV stelle, nimmt lp-3 nur noch den "Rest" des PV Überschusses.

Ich hätte jetzt erwartet, dass auch wenn lp-3 bereits läuft, der lp-1 aufgrund seiner höheren Priorität die zur Verfügung stehende Leistung bevorzugt benutzen darf und erst dann lp-3 noch den Rest bekommt.

### Steps to reproduce

1. lp-3 und lp-1 auf PV und genügend PV Überschuss haben
2. lp-3 startet
3. lp-1 bleibt aus, obwohl höherer Priorität

### Configuration details

```yaml
circuits:
  - name: main
    title: Hauptsicherung
    maxCurrent: 48
  - name: garage
    title: Garage
    maxCurrent: 32
    parent: main

loadpoints:
- title: Garage
  circuit: garage
  charger: wallbox1
  vehicle: ev1
  mode: pv
  priority: 20
  soc:
    poll:
      mode: always
      interval: 720m

- title: AC•THOR Heizstab
  circuit: main
  charger: myPV_ac-thor
  vehicle: heating1
  mode: off
  priority: 1
  enable:
    threshold: -200 # einschalten, wenn 1 min. lang mind. 200W Überschuss vorhanden ist
    delay: 1m
  disable:
    threshold: 200
    delay: 10s

vehicles:
- name: ev1
  type: template
  template: carwings 
  title: Nissan Leaf  
  user: ***
  password: ***
  capacity: 40  
  phases: 1  
  icon: car  
  cache: 15m  
  #mode: pv  
  #minSoc: 30  
  #targetSoc: 80  
  minCurrent: 6  
  maxCurrent: 25  

- name: heating1
  type: custom
  title: Heizstab
  icon: heater
```

### Log details

```text
[lp-3 ] DEBUG 2025/04/06 11:45:58 charger status: C
[lp-3 ] DEBUG 2025/04/06 11:45:58 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 11:45:58 pv charge current: 33.2A = 15A + 18.2A (-4197W @ 1p)
[lp-3 ] DEBUG 2025/04/06 11:45:58 max charge current: 30A
[lp-3 ] DEBUG 2025/04/06 11:46:08 charge power: 5598W
[lp-1 ] DEBUG 2025/04/06 11:46:08 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:08 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:46:18 charge power: 5586W
[lp-1 ] DEBUG 2025/04/06 11:46:18 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:18 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:46:18 charge voltages: [233 234 234]V
[lp-1 ] DEBUG 2025/04/06 11:46:18 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 11:46:18 charge total import: 7196.191kWh
[lp-1 ] DEBUG 2025/04/06 11:46:18 charger status: B
[lp-1 ] DEBUG 2025/04/06 11:46:21 vehicle status: C
[lp-1 ] DEBUG 2025/04/06 11:46:21 vehicle soc: 88%
[lp-1 ] DEBUG 2025/04/06 11:46:21 vehicle range: 182km
[lp-3 ] DEBUG 2025/04/06 11:46:28 charge power: 5590W
[lp-1 ] DEBUG 2025/04/06 11:46:28 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:28 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:46:38 charge power: 5602W
[lp-1 ] DEBUG 2025/04/06 11:46:38 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:38 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:46:38 charger status: C
[lp-3 ] DEBUG 2025/04/06 11:46:38 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 11:46:38 pv charge current: 39A = 30A + 9A (-2070W @ 1p)
[lp-3 ] DEBUG 2025/04/06 11:46:48 charge power: 5602W
[lp-1 ] DEBUG 2025/04/06 11:46:48 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:48 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:46:54 set charge mode: pv
[lp-3 ] DEBUG 2025/04/06 11:46:54 charge power: 5604W
[lp-1 ] DEBUG 2025/04/06 11:46:54 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:54 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:46:54 charge voltages: [233 234 234]V
[lp-1 ] DEBUG 2025/04/06 11:46:54 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 11:46:54 charge total import: 7196.191kWh
[lp-1 ] DEBUG 2025/04/06 11:46:54 charger status: B
[lp-1 ] DEBUG 2025/04/06 11:46:54 next soc poll remaining time: 11h59m23s
[lp-1 ] DEBUG 2025/04/06 11:46:54 pv charge current: 33.5A = 0A + 33.5A (-7706W @ 1p)
[lp-1 ] DEBUG 2025/04/06 11:46:54 site power -7706W <= 0W enable threshold
[lp-1 ] DEBUG 2025/04/06 11:46:54 pv enable timer start: 1m0s
[lp-1 ] DEBUG 2025/04/06 11:46:54 pv enable in 1m0s
[lp-3 ] DEBUG 2025/04/06 11:46:58 charge power: 5599W
[lp-1 ] DEBUG 2025/04/06 11:46:58 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:46:58 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:46:58 charge voltages: [233 234 234]V
[lp-1 ] DEBUG 2025/04/06 11:46:58 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 11:46:58 charge total import: 7196.191kWh
[lp-1 ] DEBUG 2025/04/06 11:46:58 charger status: B
[lp-1 ] DEBUG 2025/04/06 11:46:58 next soc poll remaining time: 11h59m19s
[lp-1 ] DEBUG 2025/04/06 11:46:58 pv charge current: 33.4A = 0A + 33.4A (-7679W @ 1p)
[lp-1 ] DEBUG 2025/04/06 11:46:58 site power -7679W <= 0W enable threshold
[lp-1 ] DEBUG 2025/04/06 11:46:58 pv enable in 56s
[lp-1 ] DEBUG 2025/04/06 11:46:58 pv enable timer remaining: 56s
[lp-3 ] DEBUG 2025/04/06 11:47:08 charge power: 5597W
[lp-1 ] DEBUG 2025/04/06 11:47:08 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:08 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:47:18 charge power: 5602W
[lp-1 ] DEBUG 2025/04/06 11:47:18 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:18 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:47:18 charger status: C
[lp-3 ] DEBUG 2025/04/06 11:47:18 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 11:47:18 pv charge current: 39.1A = 30A + 9.09A (-2090W @ 1p)
[lp-3 ] DEBUG 2025/04/06 11:47:28 charge power: 5593W
[lp-1 ] DEBUG 2025/04/06 11:47:28 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:28 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:47:38 charge power: 5596W
[lp-1 ] DEBUG 2025/04/06 11:47:38 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:38 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:47:38 charge voltages: [233 234 234]V
[lp-1 ] DEBUG 2025/04/06 11:47:38 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 11:47:38 charge total import: 7196.191kWh
[lp-1 ] DEBUG 2025/04/06 11:47:38 charger status: B
[lp-1 ] DEBUG 2025/04/06 11:47:38 next soc poll remaining time: 11h58m40s
[lp-1 ] DEBUG 2025/04/06 11:47:38 pv charge current: 33.7A = 0A + 33.7A (-7747W @ 1p)
[lp-1 ] DEBUG 2025/04/06 11:47:38 site power -7747W <= 0W enable threshold
[lp-1 ] DEBUG 2025/04/06 11:47:38 pv enable in 16s
[lp-1 ] DEBUG 2025/04/06 11:47:38 pv enable timer remaining: 16s
[lp-3 ] DEBUG 2025/04/06 11:47:48 charge power: 5595W
[lp-1 ] DEBUG 2025/04/06 11:47:48 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:48 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:47:58 charge power: 5595W
[lp-1 ] DEBUG 2025/04/06 11:47:58 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:47:58 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:47:58 charger status: C
[lp-3 ] DEBUG 2025/04/06 11:47:58 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 11:47:58 pv charge current: 38.8A = 30A + 8.79A (-2022W @ 1p)
[lp-3 ] DEBUG 2025/04/06 11:48:08 charge power: 5603W
[lp-1 ] DEBUG 2025/04/06 11:48:08 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:48:08 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:48:18 charge power: 5598W
[lp-1 ] DEBUG 2025/04/06 11:48:18 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:48:18 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/06 11:48:18 charge voltages: [233 234 234]V
[lp-1 ] DEBUG 2025/04/06 11:48:18 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 11:48:18 charge total import: 7196.191kWh
[lp-1 ] DEBUG 2025/04/06 11:48:18 charger status: B
[lp-1 ] DEBUG 2025/04/06 11:48:18 next soc poll remaining time: 11h58m0s
[lp-1 ] DEBUG 2025/04/06 11:48:18 pv charge current: 33.6A = 0A + 33.6A (-7725W @ 1p)
[lp-1 ] DEBUG 2025/04/06 11:48:18 site power -7725W <= 0W enable threshold
[lp-1 ] DEBUG 2025/04/06 11:48:18 pv enable in 0s
[lp-1 ] DEBUG 2025/04/06 11:48:18 pv enable timer elapsed
[lp-3 ] DEBUG 2025/04/06 11:48:28 charge power: 5597W
[lp-1 ] DEBUG 2025/04/06 11:48:28 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:48:28 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:48:38 charge power: 5610W
[lp-1 ] DEBUG 2025/04/06 11:48:38 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 11:48:38 charge currents: [0 0 0]A
[lp-3 ] DEBUG 2025/04/06 11:48:38 charger status: C
[lp-3 ] DEBUG 2025/04/06 11:48:38 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 11:48:38 pv charge current: 39.2A = 30A + 9.24A (-2125W @ 1p)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

VolkerK62, 2025-04-06:
Leider hört das Log in dem Moment auf, wo LP1 eingeschaltet werden würde
`[lp-1 ] DEBUG 2025/04/06 11:48:18 pv enable in 0s`

Leider fehlen im Log die Werte von `site`, aber `priority` scheint berechnet zu werden

[lp-3 ] DEBUG 2025/04/06 11:47:18 pv charge current: 39.1A = 30A + 9.09A (**-2090W** @ 1p)
[lp-1 ] DEBUG 2025/04/06 11:47:38 pv charge current: 33.7A = 0A + 33.7A (**-7747W** @ 1p)

docolli, 2025-04-06:
Danke für die schnelle Antwort!

Ich versuche gleich noch einmal ein etwas längeres Logo zu erzeugen.

docolli, 2025-04-06:
```sh
[lp-3 ] DEBUG 2025/04/06 12:33:08 charge power: 5571W
[lp-1 ] DEBUG 2025/04/06 12:33:08 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:08 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:08 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:08 pv 1 power: 8900W
[site ] DEBUG 2025/04/06 12:33:08 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:08 grid power: -2512W
[site ] DEBUG 2025/04/06 12:33:08 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:08 site power: -7983W (including 5571W prioritized power)
[site ] DEBUG 2025/04/06 12:33:18 ----
[lp-3 ] DEBUG 2025/04/06 12:33:18 charge power: 5573W
[lp-1 ] DEBUG 2025/04/06 12:33:18 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:18 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:18 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:18 grid power: -2528W
[site ] DEBUG 2025/04/06 12:33:18 pv 1 power: 8898W
[site ] DEBUG 2025/04/06 12:33:18 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:18 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:18 site power: -2428W
[lp-3 ] DEBUG 2025/04/06 12:33:18 charger status: C
[lp-3 ] DEBUG 2025/04/06 12:33:18 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 12:33:18 pv charge current: 40.6A = 30A + 10.6A (-2428W @ 1p)
[site ] DEBUG 2025/04/06 12:33:28 ----
[lp-3 ] DEBUG 2025/04/06 12:33:28 charge power: 5560W
[lp-1 ] DEBUG 2025/04/06 12:33:28 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:28 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:28 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:28 grid power: -2484W
[site ] DEBUG 2025/04/06 12:33:28 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:28 pv 1 power: 8898W
[site ] DEBUG 2025/04/06 12:33:28 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:28 site power: -2384W
[site ] DEBUG 2025/04/06 12:33:38 ----
[lp-3 ] DEBUG 2025/04/06 12:33:38 charge power: 5556W
[lp-1 ] DEBUG 2025/04/06 12:33:38 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:38 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:38 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:38 pv 1 power: 8895W
[site ] DEBUG 2025/04/06 12:33:38 grid power: -2510W
[site ] DEBUG 2025/04/06 12:33:38 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:38 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:38 site power: -2410W
[lp-1 ] DEBUG 2025/04/06 12:33:38 charge voltages: [233 234 233]V
[lp-1 ] DEBUG 2025/04/06 12:33:38 detected connected phases: 3p
[lp-1 ] DEBUG 2025/04/06 12:33:38 charge total import: 7198.519kWh
[lp-1 ] DEBUG 2025/04/06 12:33:38 charger status: B
[lp-1 ] DEBUG 2025/04/06 12:33:38 next soc poll remaining time: 11h44m0s
[site ] DEBUG 2025/04/06 12:33:48 ----
[lp-3 ] DEBUG 2025/04/06 12:33:48 charge power: 5554W
[lp-1 ] DEBUG 2025/04/06 12:33:48 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:48 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:48 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:48 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:48 pv 1 power: 8898W
[site ] DEBUG 2025/04/06 12:33:48 grid power: -2550W
[site ] DEBUG 2025/04/06 12:33:48 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:48 site power: -8004W (including 5554W prioritized power)
[lp-1 ] DEBUG 2025/04/06 12:33:54 set charge mode: pv
[site ] DEBUG 2025/04/06 12:33:54 ----
[lp-3 ] DEBUG 2025/04/06 12:33:54 charge power: 5568W
[lp-1 ] DEBUG 2025/04/06 12:33:54 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:33:54 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:33:54 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:33:54 pv 1 power: 8905W
[site ] DEBUG 2025/04/06 12:33:54 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:33:54 grid power: -2657W
[site ] DEBUG 2025/04/06 12:33:54 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:33:54 site power: -8125W (including 5568W prioritized power)
[lp-1 ] DEBUG 2025/04/06 12:33:54 charge voltages: [233 234 233]V
[... 163 lines cut ...]
[site ] DEBUG 2025/04/06 12:35:48 site power: -8005W (including 5544W prioritized power)
[site ] DEBUG 2025/04/06 12:35:58 ----
[lp-3 ] DEBUG 2025/04/06 12:35:58 charge power: 5546W
[lp-1 ] DEBUG 2025/04/06 12:35:58 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:35:58 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:35:58 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:35:58 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:35:58 pv 1 power: 8838W
[site ] DEBUG 2025/04/06 12:35:58 grid power: -2543W
[site ] DEBUG 2025/04/06 12:35:58 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:35:58 site power: -2443W
[lp-3 ] DEBUG 2025/04/06 12:35:58 charger status: C
[lp-3 ] DEBUG 2025/04/06 12:35:58 charger soc limit: 75%
[lp-3 ] DEBUG 2025/04/06 12:35:58 pv charge current: 40.6A = 30A + 10.6A (-2443W @ 1p)
[site ] DEBUG 2025/04/06 12:36:08 ----
[lp-3 ] DEBUG 2025/04/06 12:36:08 charge power: 5545W
[lp-1 ] DEBUG 2025/04/06 12:36:08 charge power: 0W
[lp-1 ] DEBUG 2025/04/06 12:36:08 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/06 12:36:08 ext 1 power: 0W
[site ] DEBUG 2025/04/06 12:36:08 battery 1 power: -0W
[site ] DEBUG 2025/04/06 12:36:08 pv 1 power: 8857W
[site ] DEBUG 2025/04/06 12:36:08 grid power: -2401W
[site ] DEBUG 2025/04/06 12:36:08 battery 1 soc: 100%
[site ] DEBUG 2025/04/06 12:36:08 site power: -2301W
```

docolli, 2025-04-06:
[image]
Selbst nachdem lp-3 fertig ist (70°C erreicht), startet lp-1 nicht. 🤔

docolli, 2025-04-06:
Fun Fact: Selbst wenn ich jetzt lp-1 auf `Schnell` stelle, startet er nicht. Im Log fehlen auch für lp-1 so Einträge wie `set charge mode: off` . api/state zeigt mir aber die Änderung de mode an.

Reboot von evcc hat hierfür erstmal geholfen, jetzt lädt das Auto wieder.
Aber es bleibt meine Frage, was mit der Priorität bei mir los ist.

mfuchs1984, 2025-04-06:
- mit welcher Version genau testest du?
- Kannst du noch deine circuits mit ins log nehmen?

docolli, 2025-04-06:
Ich habe die 0.202.1 im Einsatz. Die circuits sind hier mit im Log.
Da mein Fzg an lp-1 jetzt bereits voll geladen ist, kann ich nur noch versuchen das Ganze am lp-2 nachzustellen:

```sh
[lp-3 ] DEBUG 2025/04/06 16:58:27 charge power: 2305W
[lp-2 ] DEBUG 2025/04/06 16:58:27 charge power: 0W
[lp-2 ] DEBUG 2025/04/06 16:58:27 charge currents: [0 0 0]A
[circuit-garage] DEBUG 2025/04/06 16:58:27 power: 0W
[circuit-garage] DEBUG 2025/04/06 16:58:27 current: 0A
[circuit-main] DEBUG 2025/04/06 16:58:27 power: 2305W
[circuit-main] DEBUG 2025/04/06 16:58:27 current: 26.9A
[site ] DEBUG 2025/04/06 16:58:27 ext 1 power: 0W
[site ] DEBUG 2025/04/06 16:58:27 grid power: -30W
[site ] DEBUG 2025/04/06 16:58:27 battery 1 power: -120W
[site ] DEBUG 2025/04/06 16:58:27 pv 1 power: 3471W
[site ] DEBUG 2025/04/06 16:58:27 battery 1 soc: 96%
[site ] DEBUG 2025/04/06 16:58:27 site power: -50W
[site ] DEBUG 2025/04/06 16:58:37 ----
[lp-3 ] DEBUG 2025/04/06 16:58:37 charge power: 2309W
[lp-2 ] DEBUG 2025/04/06 16:58:37 charge power: 0W
[lp-2 ] DEBUG 2025/04/06 16:58:37 charge currents: [0 0 0]A
[circuit-garage] DEBUG 2025/04/06 16:58:37 power: 0W
[circuit-garage] DEBUG 2025/04/06 16:58:37 current: 0A
[circuit-main] DEBUG 2025/04/06 16:58:37 power: 2309W
[circuit-main] DEBUG 2025/04/06 16:58:37 current: 26.9A
[site ] DEBUG 2025/04/06 16:58:37 ext 1 power: 0W
[site ] DEBUG 2025/04/06 16:58:37 pv 1 power: 3477W
[site ] DEBUG 2025/04/06 16:58:37 battery 1 power: -99W
[site ] DEBUG 2025/04/06 16:58:37 grid power: -30W
[site ] DEBUG 2025/04/06 16:58:37 battery 1 soc: 96%
[site ] DEBUG 2025/04/06 16:58:37 site power: -2338W (including 2309W prioritized power)
[lp-2 ] DEBUG 2025/04/06 16:58:37 set charge mode: pv
[site ] DEBUG 2025/04/06 16:58:37 ----
[lp-3 ] DEBUG 2025/04/06 16:58:37 charge power: 2309W
[lp-2 ] DEBUG 2025/04/06 16:58:37 charge power: 0W
[lp-2 ] DEBUG 2025/04/06 16:58:37 charge currents: [0 0 0]A
[circuit-garage] DEBUG 2025/04/06 16:58:37 power: 0W
[circuit-garage] DEBUG 2025/04/06 16:58:37 current: 0A
[circuit-main] DEBUG 2025/04/06 16:58:37 power: 2309W
[circuit-main] DEBUG 2025/04/06 16:58:37 current: 26.9A
[site ] DEBUG 2025/04/06 16:58:37 ext 1 power: 0W
[site ] DEBUG 2025/04/06 16:58:37 battery 1 power: -99W
[site ] DEBUG 2025/04/06 16:58:37 grid power: 19W
[site ] DEBUG 2025/04/06 16:58:37 pv 1 power: 3468W
[site ] DEBUG 2025/04/06 16:58:37 battery 1 soc: 96%
[site ] DEBUG 2025/04/06 16:58:37 site power: -2289W (including 2309W prioritized power)
[lp-2 ] DEBUG 2025/04/06 16:58:38 charge voltages: [232 235 234]V
[lp-2 ] DEBUG 2025/04/06 16:58:38 detected connected phases: 3p
[lp-2 ] DEBUG 2025/04/06 16:58:38 charge total import: 25.028kWh
[lp-2 ] DEBUG 2025/04/06 16:58:38 charger status: B
[lp-2 ] DEBUG 2025/04/06 16:58:38 next soc poll remaining time: 11h58m50s
[lp-2 ] DEBUG 2025/04/06 16:58:38 plan: charge 52m4s between 2025-04-07 06:00:00 +0200 CEST until 2025-04-07 07:00:00 +0200 CEST (power: 17250W, avg cost: 0.296)
[lp-2 ] DEBUG 2025/04/06 16:58:38 pv charge current: min 6A > 3.32A (-2289W @ 3p, battery: true)
[lp-2 ] DEBUG 2025/04/06 16:58:38 max charge current: 6A
[lp-2 ] DEBUG 2025/04/06 16:58:38 charger enable
[lp-2 ] DEBUG 2025/04/06 16:58:38 wake-up timer: start
[site ] DEBUG 2025/04/06 16:58:47 ----
[lp-3 ] DEBUG 2025/04/06 16:58:47 charge power: 2295W
[lp-2 ] DEBUG 2025/04/06 16:58:47 charge power: 873W
[lp-2 ] DEBUG 2025/04/06 16:58:47 charge currents: [5 1 1]A
[circuit-garage] DEBUG 2025/04/06 16:58:47 power: 873W
[circuit-garage] DEBUG 2025/04/06 16:58:47 current: 5A
[circuit-main] DEBUG 2025/04/06 16:58:47 power: 3168W
[circuit-main] DEBUG 2025/04/06 16:58:47 current: 31.9A
[site ] DEBUG 2025/04/06 16:58:47 ext 1 power: 0W
[site ] DEBUG 2025/04/06 16:58:47 grid power: -22W
[site ] DEBUG 2025/04/06 16:58:47 pv 1 power: 3471W
[site ] DEBUG 2025/04/06 16:58:47 battery 1 power: 546W
[site ] DEBUG 2025/04/06 16:58:47 battery 1 soc: 96%
[site ] DEBUG 2025/04/06 16:58:47 site power: -1671W (including 2295W prioritized power)
[... 184 lines cut ...]
[site ] DEBUG 2025/04/06 17:00:17 grid power: 5W
[site ] DEBUG 2025/04/06 17:00:17 pv 1 power: 3434W
[site ] DEBUG 2025/04/06 17:00:17 battery 1 power: 1899W
[site ] DEBUG 2025/04/06 17:00:17 battery 1 soc: 95%
[site ] DEBUG 2025/04/06 17:00:17 site power: 2004W
[lp-3 ] DEBUG 2025/04/06 17:00:17 charger status: B
[lp-3 ] INFO 2025/04/06 17:00:17 stop charging <-
[lp-3 ] DEBUG 2025/04/06 17:00:18 charger soc limit: 75%
[site ] DEBUG 2025/04/06 17:00:27 ----
[lp-3 ] DEBUG 2025/04/06 17:00:27 charge power: 0W
[lp-2 ] DEBUG 2025/04/06 17:00:27 charge power: 4294W
[lp-2 ] DEBUG 2025/04/06 17:00:27 charge currents: [6 7 7]A
[circuit-garage] DEBUG 2025/04/06 17:00:27 power: 4294W
[circuit-garage] DEBUG 2025/04/06 17:00:27 current: 7A
[circuit-main] DEBUG 2025/04/06 17:00:27 power: 4294W
[circuit-main] DEBUG 2025/04/06 17:00:27 current: 23A
[site ] DEBUG 2025/04/06 17:00:27 ext 1 power: 0W
[site ] DEBUG 2025/04/06 17:00:27 battery 1 power: 1890W
[site ] DEBUG 2025/04/06 17:00:27 pv 1 power: 3425W
[site ] DEBUG 2025/04/06 17:00:27 grid power: 11W
[site ] DEBUG 2025/04/06 17:00:27 battery 1 soc: 95%
[site ] DEBUG 2025/04/06 17:00:27 site power: 2001W
```

Hier hat die Priorisierung jetzt geklappt! Unterschied ist, dass das Fzg. an lp-1 1p-phasig lädt, an lp-2 lädt das Fzg. 3-phasig.

mfuchs1984, 2025-04-06:
Ich fragte nach den circuits, da beim anderen Fahrzeug, das einphasig lädt, Ströme größer 32 A in die circuit Logik gingen. Das sollte an sich kein Problem sein, aber wer weiß, das Log incl. circuit gibt noch etwas mehr Information und ich glaube, dass das load management in bestimmten Situationen mit den Prioritäten in Konflikt geraten kann. Beim anderen Auto jetzt passiert das nicht.

`[lp-1 ] DEBUG 2025/04/06 12:34:18 pv charge current: 34.6A = 0A + 34.6A (-7965W @ 1p)`
`[circuit-main] DEBUG 2025/04/06 17:00:27 current: 23A`

docolli, 2025-04-06:
Ich hoffe ich kann das diese Woche nochmals mit dem anderen Fzg. nachstellen. 

Ich habe jetzt für meine Heizung (Wärmepumpe+Puffer-Heizstab) noch einen extra circuit erstellt, die sind tatsächlich zusammen 3-phasig vom Verteilpunkt aus angeschlossen. Dort habe ich sogar ein Shelly 3EM, das könnte ich noch als `meter` für diesen circuit einbinden. Muss ich mir aber erst in der Doku anschauen, wie man das macht.
Mein Hausanschluß ist auch mit 64A abgesichert, das habe ich geändert. Die 48A sind per Copy&Paste aus dem Config-Beispiel rein kopiert worden.

```yaml
circuits:
  - name: main
    title: Hauptsicherung
    maxCurrent: 64
  - name: garage
    title: Garage
    maxCurrent: 32
    parent: main
  - name: heating
    title: Heizung
    maxCurrent: 32
    parent: main
 
loadpoints:
- title: AC•THOR Heizstab
  circuit: heating
  charger: myPV_ac-thor
  vehicle: heating1
  mode: off
  priority: 1
  enable:
    threshold: -200 # einschalten, wenn 1 min. lang mind. 200W Überschuss vorhanden ist
    delay: 1m
  disable:
    threshold: 200
    delay: 10s

- title: Wärmepumpe
  circuit: heating
  charger: wp_charger
  meter: wp_meter
  vehicle: heating2
  mode: now
  priority: 10
```

VolkerK62, 2025-04-06:
off topic:
>Mein Hausanschluß ist auch mit 64A abgesichert, das habe ich geändert. 

Dein main circuit hat kein meter. Das bedeutet, alle nicht durch evcc erfassten Verbraucher, werden nicht berücksichtigt.

docolli, 2025-04-06:
Okay, dann MUSS ich dem main circuit auch einen meter zuordnen? Also einmal bei einem circuit einen meter, dann ÜBERALL?

Was ist der richtige? Der, der den Hausverbrauch misst? Also ein Wert den mir meine PV-Anlage liefert?

Oder in meinem Fall keine Hierarchie, sondern einfach nur `garage` und `heating` definieren ohne `parent: main`?

VolkerK62, 2025-04-06:
aber, ich sehe im Log kein `grid currents`. Wenn das Grid Meter diese nicht liefert, kann man es nicht für MaxCurrent beim LM nutzen.

docolli, 2025-04-06:
Das denke ich nicht, da dieses nur den Netzbezug misst. Wenn die PV liefert, kommt u.U. nichts vom Netz. Wird eher der Wert "Hausverbrauch" meines E3DC HKW sein. Bin mir aber nicht sicher ob der gemessen oder nur berechnet wird.

Aber eigentlich sind wir jetzt gang schön OT. 😉 Da mache ich mal eine extra Diskussion dazu auf.
Hier soll es ja um die Priorität der loadpoints gehen.