evcc-io/evcc #20444: custom build x86 fails since v200

cc3neo, 2025-04-06:
### Describe the bug

Hallo zusammen,
seit der v200 funktioniert mein Build für x86 leider nicht mehr und es kommt zum trace (getestet zuletzt mit evcc  0.202.1).
Ich habe evcc seit ein paar Jahren erfolgreich auf meinem alten aber treuen x86 Notebook laufen, um diesem noch einen Zweck zu geben. Hierfür nutze ich ein eigenen Build vom Source-Code und lasse diesen direkt auf dem Gerät erstellen, was bis zur v0.133.0 einwandfrei funktionierte.
Dabei kommt es seit der Version >v200 beim rendern von `chunk 13` im `make` zu einem Trace.

```
<--- Last few GCs --->

[9406:0xe9212a0]   119178 ms: Scavenge 239.5 (253.5) -> 238.4 (254.0) MB, 7.17 / 0.00 ms  (average mu = 0.208, current mu = 0.108) allocation failure;
[9406:0xe9212a0]   119366 ms: Scavenge 240.0 (254.0) -> 238.6 (254.2) MB, 5.50 / 0.00 ms  (average mu = 0.208, current mu = 0.108) allocation failure;
[9406:0xe9212a0]   121907 ms: Mark-Compact 240.2 (254.2) -> 235.8 (254.4) MB, 2424.96 / 0.00 ms  (average mu = 0.295, current mu = 0.375) allocation failure; scavenge might not succeed

<--- JS stacktrace --->

FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory
----- Native stack trace -----

 1: 0x874bfb6 node::OOMErrorHandler(char const*, v8::OOMDetails const&) [node]
 2: 0x8aaaa1f v8::Utils::ReportOOMFailure(v8::internal::Isolate*, char const*, v8::OOMDetails const&) [node]
 3: 0x8aaace4 v8::internal::V8::FatalProcessOutOfMemory(v8::internal::Isolate*, char const*, v8::OOMDetails const&) [node]
 4: 0x8cb433c  [node]
 5: 0x8cb46b3 v8::internal::Heap::RecomputeLimits(v8::internal::GarbageCollector) [node]
 6: 0x8ccada3 v8::internal::Heap::PerformGarbageCollection(v8::internal::GarbageCollector, v8::internal::GarbageCollectionReason, char const*) [node]
 7: 0x8ccb458 v8::internal::Heap::CollectGarbage(v8::internal::AllocationSpace, v8::internal::GarbageCollectionReason, v8::GCCallbackFlags) [node]
 8: 0x8ca34ea v8::internal::HeapAllocator::AllocateRawWithLightRetrySlowPath(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node]
 9: 0x8ca4502 v8::internal::HeapAllocator::AllocateRawWithRetryOrFailSlowPath(int, v8::internal::AllocationType, v8::internal::AllocationOrigin, v8::internal::AllocationAlignment) [node]
10: 0x8c81fe6 v8::internal::Factory::AllocateRaw(int, v8::internal::AllocationType, v8::internal::AllocationAlignment) [node]
11: 0x8c70ffd v8::internal::FactoryBase<v8::internal::Factory>::AllocateRaw(int, v8::internal::AllocationType, v8::internal::AllocationAlignment) [node]
12: 0x8c7160d v8::internal::FactoryBase<v8::internal::Factory>::AllocateRawWithImmortalMap(int, v8::internal::AllocationType, v8::internal::Map, v8::internal::AllocationAlignment) [node]
13: 0x8c7449c v8::internal::FactoryBase<v8::internal::Factory>::NewRawTwoByteString(int, v8::internal::AllocationType) [node]
14: 0x8fddf0b v8::internal::String::SlowFlatten(v8::internal::Isolate*, v8::internal::Handle<v8::internal::ConsString>, v8::internal::AllocationType) [node]
15: 0x8fe10a6 v8::internal::String::ToNumber(v8::internal::Isolate*, v8::internal::Handle<v8::internal::String>) [node]
16: 0x90dba7c v8::internal::Runtime_StringToNumber(int, unsigned int*, v8::internal::Isolate*) [node]
17: 0x94cf46d  [node]
18: 0x945d709  [node]
19: 0xab64f083
20: 0xab6492b8
21: 0xab647084
22: 0xab56c7c0
23: 0xab6588e6
24: 0xab646fd8
25: 0x31423c1b
26: 0xab6588e6
27: 0xab652df4
28: 0xab6588e6
29: 0xab6486a9
30: 0xab567c3e
31: 0xab6588e6
32: 0xab568904
33: 0xab6588e6
34: 0x3cdb9ccf
35: 0xab563b83
36: 0xab567c3e
37: 0xab560d92
38: 0xab55cc11
39: 0xab56020b
40: 0xab568904
41: 0xab560d92
42: 0xab55cc11
43: 0xab56020b
44: 0xab56a67d
45: 0xab560d92
46: 0xab55cc11
47: 0xab56020b
48: 0xab56177b
49: 0xab5612d0
50: 0xab56c7c0
51: 0xab560d92
52: 0xab55cc11
53: 0xab56020b
54: 0xab5677d2
55: 0xab56773b
56: 0xab56787b
57: 0xab560d92
58: 0xab55cc11
59: 0xab56020b
60: 0xab56177b
61: 0xab563b83
62: 0xab567c3e
63: 0xab560d92
64: 0xab55cc11
65: 0xab56020b
66: 0xab568904
67: 0xab560d92
68: 0xab55cc11
69: 0xab56020b
70: 0xab56177b
71: 0xab5612d0
72: 0xab563a28
73: 0xab560d92
74: 0xab55cc11
75: 0xab56020b
76: 0xab568cb7
77: 0xab560d92
78: 0xab55cc11
79: 0xab56020b
80: 0xab56177b
81: 0xab563b83
82: 0xab567c3e
83: 0xab55cb83
84: 0xab56020b
85: 0xab568904
86: 0xab55cb83
87: 0xab56020b
88: 0xab5638d4
89: 0xab55cb83
90: 0xab56020b
91: 0xab568cb7
92: 0xab55cb83
93: 0xab56020b
94: 0xab56177b
95: 0xab563b83
96: 0x945b661  [node]
97: 0xab55cb83
98: 0xab56020b
99: 0xb1d51f2c
100: 0xab55cb83
101: 0xab56020b
102: 0x945b661  [node]
103: 0x46179637
104: 0x3141dd34
105: 0x38d36406
106: 0x948c2ec  [node]
107: 0x3cd8e881
108: 0x948c2ec  [node]
109: 0x5ce0dc46
110: 0x3f928b60
111: 0xb1d52498
112: 0x945b661  [node]
113: 0x945b661  [node]
114: 0x948a455  [node]
115: 0x951eac6  [node]
116: 0x947d0de  [node]
117: 0x9459ecd  [node]
118: 0x8c03c09  [node]
119: 0x8c04d9b v8::internal::Execution::TryRunMicrotasks(v8::internal::Isolate*, v8::internal::MicrotaskQueue*) [node]
120: 0x8c37aa4  [node]
121: 0x8c37e7b v8::internal::MicrotaskQueue::PerformCheckpoint(v8::Isolate*) [node]
122: 0x8950f2c  [node]
123: 0x8b0e9f1 v8::internal::FunctionCallbackArguments::Call(v8::internal::CallHandlerInfo) [node]
124: 0x8b0f1cb  [node]
125: 0x8b0f5f1 v8::internal::Builtin_HandleApiCall(int, unsigned int*, v8::internal::Isolate*) [node]
126: 0x94cf3ad  [node]
127: 0x3aec4912
128: 0x9459f59  [node]
129: 0x9459dcd  [node]
130: 0x8c03790  [node]
131: 0x8c04809 v8::internal::Execution::Call(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, int, v8::internal::Handle<v8::internal::Object>*) [node]
132: 0x8acb66e v8::Function::Call(v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*) [node]
133: 0x8790c98 node::InternalCallbackScope::Close() [node]
134: 0x879100a node::InternalMakeCallback(node::Environment*, v8::Local<v8::Object>, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*, node::async_context) [node]
135: 0x87a89ae node::AsyncWrap::MakeCallback(v8::Local<v8::Function>, int, v8::Local<v8::Value>*) [node]
136: 0x8997034 node::StreamBase::CallJSOnreadMethod(int, v8::Local<v8::ArrayBuffer>, unsigned int, node::StreamBase::StreamBaseJSChecks) [node]
137: 0x8999461 node::EmitToJSStreamListener::OnStreamRead(int, uv_buf_t const&) [node]
138: 0x899fda9 node::LibuvStreamWrap::OnUvRead(int, uv_buf_t const*) [node]
139: 0x89a01c4  [node]
140: 0x9447034  [node]
141: 0x94477ef  [node]
142: 0x944f2f2  [node]
143: 0x943b49a uv_run [node]
144: 0x8791c2b node::SpinEventLoopInternal(node::Environment*) [node]
145: 0x88cef9e  [node]
146: 0x88cf98b node::NodeMainInstance::Run() [node]
147: 0x8838ca1 node::Start(int, char**) [node]
148: 0x878727b main [node]
149: 0xb7a05ed6 __libc_start_main [/lib/i386-linux-gnu/libc.so.6]
Aborted
npm notice
npm notice New major version of npm available! 10.8.2 -> 11.2.0
npm notice Changelog: https://github.com/npm/cli/releases/tag/v11.2.0
npm notice To update run: npm install -g npm@11.2.0
npm notice
make: *** [Makefile:46: ui] Fehler 134
```
Das Gerät hat leider nur 1GB RAM (reichte in der Vergangingheit immer für evcc inkl. iobroker) und `node v20.18.3` mit ` npm 10.8.2` von https://unofficial-builds.nodejs.org auf `Debian 11.11` mit Kernel `Debian 5.10.234-1 (2025-02-24)`

Hat jemand ggf. einen Tip oder Parameter, wie ich das doch noch für x86 zum Laufen bekomme. Würde meine Hardware einfach gerne weiter einsetzen.

### Steps to reproduce

Make a custom build on x86 Debian, following the command:

```
wget https://github.com/evcc-io/evcc/archive/refs/tags/0.202.1.tar.gz
tar -xzf 0.202.1.tar.gz
cd evcc-0.202.1/
make install-ui
make install
npm install @rollup/wasm-node
cp -rp node_modules/@rollup/wasm-node/* node_modules/rollup
make patch-asn1-sudo #ich nutze eine Elli Wallbox
make
```

### Configuration details

```yaml
not relevant
```

### Log details

```text
not relevant
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

andig, 2025-04-06:
Wenn wir da rein schauen wollen müsstest du den exakten Commit raus finden, sonst würde ich da lieber keine Zeit investieren.

mh81, 2025-04-06:
Hi @cc3neo ,

falls ich Dich richtig verstehe ist das Problem nur das kompilieren und kein Trace beim Ausführen von evcc. Vielleicht ist es ja ein Problem bzgl. der Ressourcen auf dem Notebook.

Du könntest versuchen, auf einem anderen Rechner ein cross-compile für die x86 Architektur zu machen. Bin quasi n00b was go angeht, aber habe gute Erfahrungen gemacht indem ich das Repo in VS Code geöffnet habe, als Dev-Container (das bietet er beim Start an). 
Dann hab ich go in der Shell [manuell installiert ](https://go.dev/doc/install), und dann in der Shell einfach `GOOS=linux GOARCH=arm64 make` ausgeführt, was zu einer evcc binary für den Raspi 5 führt. Bei Dir wäre das dann soweit ich das sehe `GOOS=linux GOARCH=386 make`. Die evcc binary hab ich dann einfach kopiert. 

Falls das klappt wäre es dazu vermutlich noch schneller beim kompilieren auf der neuen Hardware 😃 Nur halt das manuelle kopieren der Binary ist dann noch "Aufwand".

Gruß
Marco

StefanSchoof, 2025-04-06:
Ist auch unter https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md#cross-compiling in der Dokumentation

cc3neo, 2025-04-07:
Danke für Eure Rückmeldung.
Ich habe es leider nicht geschafft den DevContainer bei mir zum Laufen zu bringen. Der Start bricht immer im `postCreateCommand` bei `make install` ab.
@andig was wäre denn zu tun, um den Commit herauszufinden? Bräuchte etwas Handreichung, da ich keine Entwicklererfahrung habe.

StefanSchoof, 2025-04-07:
Das geht am besten mit `git bisect`. Findet man gut Tutorials per Suchmaschinen.

closed 2025-04-08 by andig