evcc-io/evcc #20450: SMA support and lpc support together

hybe2022, 2025-04-06:
**Is your feature request related to a problem? 
I want to integrate my wallboxes into the SMA Portal am have configured hence the sunny home manager in HEMS section. This is only to have the consumption visible in SMA Portal or APP. It is not controlled by SMA. 
My plan was as well to use evcc for §14a as SMA system is no flexible enough. This is unfortunately also configured via HEMS section which only supports one entry.
Means I cannot use SMA and LPC together as this runs into an error which is a showstopper for me.

See also: 
https://github.com/evcc-io/evcc/issues/19216

**Describe the solution you'd like**
It should be possible to configure SMA home manager support and relay for LPC support together or to decouple both and create own sections for LPC and SMA home manager support in the configuration.

**Describe alternatives you've considered**
 no alternatives available 

**Additional context**
See 
https://github.com/evcc-io/evcc/issues/19216

premultiply, 2025-04-06:
Ich verstehe kein Wort. Versuche es mal bitte auf Deutsch.

evcc exportiert alle Ladepunkte via SEMP wenn der HEMS-Eintrag für SMA in der Konfiguration aktiv ist.
Dazu braucht es keinerlei weiter Angaben. Das kann (vollständig) fehlschlagen wenn die Dokumentation nicht gelesen wurde und evcc innerhalb eines Containers oder einen sonstigen fehlerhaften Netzwerkumgebung läuft bei dem der SMA SHM und evcc sich nicht via Multicast sehen und verständigen können.

closed 2025-04-06 by premultiply

hybe2022, 2025-04-06:
Es geht nicht darum, dass ich die Daten nicht bis SEMP ins SMA Portal bekomme, sondern wie oben und in dem erwähnten PR beschreiben, dass die Konfiguration des SMA home manager unter HEMS und gleichzeitig des Relais für LPC unter HEMS zu einem Fehler beim Starten führt und somit beides zusammen nicht nutzbar ist. 

Der PR wurde geschlossen und Andi sagt nun, dass es ein neues Feature wäre. Für mich ist es eher ein PR.

Siehe https://github.com/evcc-io/evcc/issues/19216

andig, 2025-04-06:
Es wäre prima wenn wir das nur in einem Issue diskutieren würden. Danke.