evcc-io/evcc #20479: openWB pro+ und version

ioBroker23, 2025-04-07:
### Describe the bug

die werte soc_value, vehicle_id und rfid_tag weren so die es aussieht nicht übernommen. 
abfrage in openWP pro+ zeigt sie 
evcc leider nicht

### Steps to reproduce

1. auto anschlissen
2. mit openWB prüfen
2.in evcc auf das auto gehen und info feld ist leer
3.
...

### Configuration details

```yaml
ist nur in oberfläche
```

### Log details

```text
leider nicht vorhanden da zur zeit mit der openWB oberfläche geladen wird
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

/0.202.1

andig, 2025-04-08:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-04-08 by andig

ioBroker23, 2025-04-08:
[main  ] INFO 2025/04/08 08:17:37 evcc 0.202.1
[main  ] INFO 2025/04/08 08:17:37 using config file: /home/pi/evcc.yaml
[db    ] INFO 2025/04/08 08:17:37 using sqlite database: /home/pi/.evcc/evcc.db
Power:          0W
Energy:         108.7kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 233V 233V 232V
Charge status:  B
Enabled:        false
Identifier:     <none>

openwb zeigt id und/oder rfid an

gibt es die übernahme überhaubt bei der openWB pro+ und openWB software 2 ??

andig, 2025-04-08:
Ich hab selbst eine OWB Pro, die funktioniert wunderbar.

ioBroker23, 2025-04-11:
ich nutze openWB pro+ und da ist von haus aus die software 2.1.8-Alpha.1 instaliert. warte auch auf antwort von openWB ob das richtig ist  oder ob ich auch auf 2.1.7 gehen kann.
muss in EVCC deshalb nicht openWB sondern V2 auswählen :-(
wenn ich pro wähle wird die pro+ nicht von EVCC gefunden :-(
schön wäre wenn vehicle_id	und rfid_tag übernommen werden würden.
die pro+ kann beides auslesen sowie auch den soc_value bei anstecken. die weiteren sco werte müssen aber weiterhin über andere wege kommen.
was auch genz toll wäre das wenn EVCC die openWB pro+ steuert die EVCC oberfläche zusätzlich auch auf der url die die openWB nutzt dargestellt würde :-)

andig, 2025-04-11:
Auf meiner OpenWB Pro läuft überhaupt kein OpenWB sondern sie wird direkt angesteuert. Das funktioniert über `openwbpro`. Die `openwb` Implementierung (=für die OpenWB v1 Software) von evcc kennt keine RFID Identifier. Die `openwb-v2` unterstützt RFID bereits. Separate vehicle_id ist mir nicht bekannt.

/cc @snaptec vielleicht könntest Du Licht ins Dunkel bringen?

snaptec, 2025-04-11:
Die pro+ ist eine vollwertige openWB. 
Also wie eine series2. Allerdings als Ladepunkt eine Pro angebunden. 

Die Pro liest du nicht direkt aus sonder wie eine series2 im secondary Mode.

andig, 2025-04-11:
@snaptec perfekt, das macht das `openwb-v2`. Dann fehlt uns aber anscheinend noch das `vehicle_id`. Die v2 steuern wir über Modbus an- wo würden wir das dann finden?

ioBroker23, 2025-04-11:
http://ip_openWB:8080/connect.php

gibt das aus. habe aber in EVCC nix gefunden um die werte z.b. mit ioBroker an EVCC zu übergeben

snaptec, 2025-04-11:
Das solltest du nicht nutzen. Da die software2 die Steuerung übernimmt. Auf die musst du gehen.

ioBroker23, 2025-04-12:
> Das solltest du nicht nutzen. Da die software2 die Steuerung übernimmt. Auf die musst du gehen.

schlecht ausgedrückt will über ioBrokker vehicle_id und rfid_tag bei der pro+ auslesen und dann an denn an EVCC übergeben. solange das nicht direkt geht oder die pro+ beim schnell laden die sB sperrt.
zur zeit passen beide systme nicht 100% ohne eigenen eingriff :-( 
zur zeit lade ich über openWB und sperre die sB mit dem ioBrocker beim laden über 10kW :-)

ioBroker23, 2025-04-13:
[image]

wennn da mal hand angelegt wird wäre es super toll wenn der start SCO auch mit übernommen würde und statt +6% dann z.b. 56% angezeigt wird.