evcc-io/evcc #20501: MQTT schreiben: Anpassung des Payloads per JQ

doits, 2025-04-09:
Ich würde gerne etwas in ein MQTT topic schreiben, den Wert aber anpassen. Soweit ich sehe kann ich mit `payload` den Wert z.B. in `1` oder `0` verwandeln (bei `enabled`). Aber den Wert dann weiter per `jq` anpassen geht leider nicht, z.B. so etwas:

```yml
chargers:
  - name: xxx
    enable:
      source: mqtt
      topic: 'a/b/c'
      jq: 'if . == 65 then 1 else 0 end'
    enabled:
      source: mqtt
      topic: 'a/b/c/set'
      payload: ${enable:%d}
      jq: 'if . == 1 then 65 else 50 end'
```

Beim lesen eines Wertes funktioniert es mit `jq`.

Könnte es eingebaut werden, dass auch der Wert der *geschrieben* wird, per jq modifiziert werden kann?

doits, 2025-04-09:
Ich habe einen Workaround mittels `source: script` und `mosquitto_pub` gefunden, aber schöner wäre es weiterhin mit mqtt, sodass man kein externes Programm aufrufen müsste:

```yml
enabled:
  source: script
  cmd: /bin/sh -c 'mosquitto_pub -m "$([[ ${enable:%d} -eq 1 ]] && echo -n 65 || echo -n 50)" -t "a/b/c/set"'

andig, 2025-04-09:
Wenn ich dich richtig verstehe möchtest Du, dass die Setter (mqtt, aus Konsistenzgründen auch http), den zu setzenden Wert vorab per JQ anpassbar machen? Das ist eine super Idee- willst Du einen PR versuchen? Es müsste aber tatsächlich konsequent passieren, nicht nur bei mqtt.

doits, 2025-04-09:
> Wenn ich dich richtig verstehe möchtest Du, dass die Setter (mqtt, aus Konsistenzgründen auch http), den zu setzenden Wert vorab per JQ anpassbar machen?

Korrekt!

> willst Du einen PR versuchen

Ja, kann ich machen, allerdings wird es vermutlich ein paar Wochen dauern, da ich aktuell leider wenig Zeit habe. Also bitte nicht zu schnell etwas erwarten. Ihr könnt mich gerne dem Issue assignen.

> Es müsste aber tatsächlich konsequent passieren, nicht nur bei mqtt.

Werde ich dann berücksichtigen.

doits, 2025-04-21:
doch doch

doits, 2025-04-27:
Erster Versuch in #20913. Brauche dort vermutlich etwas Feedback und Hilfe, ob das in die richtige Richtung geht 🙏.

andig, 2025-04-29:
Dein Beispiel oben ist übrigens falsch rum: den Setter gibts nur bei `enable`, nicht bei `enabled`.

andig, 2025-05-10:
Denkprozess aus https://github.com/evcc-io/evcc/pull/21147#discussion_r2081983159: 

>  jq: 'if . == 1 then 65 else 50 end'

M.E. braucht es das _gar nicht_ weil sich die Log leicht in das Template bauen lässt:

    payload: {{ if .enable }}65{{ else }}50{{ end }}

closed 2025-05-10 by andig