evcc-io/evcc #20502: Probleme mit Tibber Pulse: Too many open connections on this server

whisky2k2, 2025-04-09:
### Describe the bug

Hallo,

ich habe seit einigen Wochen Probleme mit dem Zusammenspiel aus EVCC und Tibber Pulse.
Leider werden vergleichbare Threads immer schnell geschlossen, so dass eine Diskussion der Probleme verebbt. Das Problem tritt manchmal innerhalb von Stunden auf, mal läuft es einige Tage rund.

Offtopic:
Die Anbindung an die Tibber API Lösung funktionierte über ein Jahr problemlos. Mir ist klar, dass das kein absolut zuverlässiger Datenlieferant ist, weswegen ich auch gerne noch mal das Thema lokale Abfrage ansprechen möchte. Feature Requests zu dem Thema sind leider versandet.

Ich habe zu diesem Zweck versuchweise Home Assistant aufgesetzt, bin aber weder in der Lage die dort eingesammelten Messwerte per MQTT oder über http an die separate EVCC Instanz (nicht in HA integriert) weiterzureichen.
Anbindung Tibber Pulse lokal im Netzwerk an HA: https://github.com/marq24/ha-tibber-pulse-local - darüber wird auch die SML übersetzt
Abruf der Werte via http/api von evcc über HA: https://github.com/marq24/ha-evcc/blob/main/HA_AS_EVCC_SOURCE.md (ich scheitere an der syntax mit jq: .state|tonumber)
Die Weiterleitung per MQTT scheitert bei mir am Sensor-Wert

### Steps to reproduce

1. Konfiguration gemäß https://docs.evcc.io/docs/devices/meters#tibber-pulse
2. Warten bis zum Fehler
3. Neues API Token bei Tibber generieren und EVCC mit neuem Token starten
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

# mqtt message broker
#mqtt:
#  broker: 192.168.11.237:1883
#  topic: pulse/wirkleistung

meters:
- type: template
  template: huawei-dongle 
  id: 1  
  host: 192.168.11.240  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 15s  
  name: pv1

#- name: grid1
#  type: custom
#  power:
#    source: http
#    uri: http://192.168.11.237:8123/api/states/sensor.ltibber_0100100700ff
#    method: GET
#    headers:
#    - Authorization: Bearer *****
#    insecure: true
[... 126 lines cut ...]
[pulse ] TRACE 2025/04/09 08:14:33 POST https://api.tibber.com/v1-beta/gql
[pulse ] TRACE 2025/04/09 08:14:33 {"query":"{viewer{websocketSubscriptionUrl}}"} -- {"data":{"viewer":{"websocketSubscriptionUrl":"wss://websocket-api.tibber.com/v1-beta/gql/subscriptions"}}}
[pulse ] TRACE 2025/04/09 08:14:33 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/09 08:14:33 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/09 08:14:33 {"id":"6a132063-388b-4c28-a37d-7bade6a2400f","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"5c5eea54-c342-4bb3-80a7-fe6adab180da"}}} client
[pulse ] TRACE 2025/04/09 08:14:36 {"id":"6a132063-388b-4c28-a37d-7bade6a2400f","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":916,"lastMeterConsumption":5562.093,"lastMeterProduction":7203.1367,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/09 08:14:39 {"id":"6a132063-388b-4c28-a37d-7bade6a2400f","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":919,"lastMeterConsumption":5562.093,"lastMeterProduction":7203.1377,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/09 08:14:42 {"id":"6a132063-388b-4c28-a37d-7bade6a2400f","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":917,"lastMeterConsumption":5562.093,"lastMeterProduction":7203.1387,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

machristoph1, 2025-04-10:
Ich kann leider bestätigen, dass die Meldung mit "Too many open connections on this server" in den letzten Wochen ein häufiger Zustand ist. Kommt täglich,  meist aber nur einmalig oder für ein paar Minuten, manchmal steht das Problem aber auch über Stunden an, zuletzt z. B. am 31.3/1.4 und am 08.04./09.04. 

EVCC funktioniert in der Zeit (logischerweise) nicht. Neustart (mit gleichem Token) hilft auch nicht. Evcc fängt sich allerdings irgendwann (evtl. wenn Tibber wieder funktioniert?). Es kommt nicht zu einem dauerhaften Funktionsverlust, das kann aber einige Stunden dauern. Aber irgendwann verschwinden die Meldungen und es ist wieder alles normal. 

Hier mal ein kurzes Event von heute 14:00 
```
Apr 10 14:10:55 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:10:55 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:10:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:10:59 ----
Apr 10 14:10:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:10:59 charge power: 0W
Apr 10 14:10:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:10:59 charge currents: [0 0 0]A
Apr 10 14:10:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:10:59 grid power: 0W
Apr 10 14:10:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:10:59 grid currents: [0 0 0]A
Apr 10 14:10:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:10:59 pv power: 0W
Apr 10 14:10:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:10:59 site power: 0W
Apr 10 14:10:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:10:59 charge total import: 15474.577kWh
Apr 10 14:10:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:10:59 charger status: A
Apr 10 14:11:00 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:00 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:05 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:05 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:10 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:10 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:15 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:15 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:20 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:20 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:26 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:26 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:29 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:29 ----
Apr 10 14:11:29 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:29 charge power: 0W
Apr 10 14:11:29 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:29 charge currents: [0 0 0]A
Apr 10 14:11:29 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:29 grid power: 0W
Apr 10 14:11:29 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:29 grid currents: [0 0 0]A
Apr 10 14:11:29 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:29 pv power: 0W
Apr 10 14:11:29 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:29 site power: 0W
Apr 10 14:11:29 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:29 charge total import: 15474.577kWh
Apr 10 14:11:29 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:29 charger status: A
Apr 10 14:11:31 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:31 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:36 raspicm evcc[30871]: [pulse ] ERROR 2025/04/10 14:11:36 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.>
Apr 10 14:11:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:59 ----
Apr 10 14:11:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:59 charge power: 0W
Apr 10 14:11:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:59 charge currents: [0 0 0]A
Apr 10 14:11:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:59 grid power: 0W
Apr 10 14:11:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:59 grid currents: [0 0 0]A
Apr 10 14:11:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:59 pv power: 0W
Apr 10 14:11:59 raspicm evcc[30871]: [site  ] DEBUG 2025/04/10 14:11:59 site power: 0W
Apr 10 14:11:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:59 charge total import: 15474.577kWh
Apr 10 14:11:59 raspicm evcc[30871]: [lp-1  ] DEBUG 2025/04/10 14:11:59 charger status: A
```

GrimmiMeloni, 2025-04-15:
Zu viele Verbindungen, sind halt zu viele Verbindungen. 🤷‍♂ 
@whisky2k2 hast Du andere Tools die sich gleichzeitig mit evcc auf der Pulse API anmelden?
Auch wäre es interessant mal ein Log zu sehen, bei dem auch `tibber` (also der Tariff) auf trace steht. Vielleicht gibt es da auch nochmal Erkenntnisse.

Ansonsten auch noch der Hinweis, das es im gezeigten Log, serverseitig Fehler hagelt. Hier war also defintiv auch bei Tibber etwas nicht in Ordnung.

whisky2k2, 2025-04-15:
> hast Du andere Tools die sich gleichzeitig mit evcc auf der Pulse API anmelden? Auch wäre es interessant mal ein Log zu sehen, bei dem auch `tibber` (also der Tariff) auf trace steht. Vielleicht gibt es da auch nochmal Erkenntnisse.
> 

Ich habe keine anderen Abfragen zur API, nur HA mit der lokalen Abfrage. 
Aktuell scheint es problemlos zu laufen. Ich melde mich  bei passender Gelegenheit mit einem Trace. 

> Ansonsten auch noch der Hinweis, das es im gezeigten Log, serverseitig Fehler hagelt. Hier war also defintiv auch bei Tibber etwas nicht in Ordnung.

Keine Frage und es geht mir auch nicht darum, hier einen schwarzen Peter zu verteilen. Ich brauche nur ein robustes System, weil ich nicht immer eingreifen kann und es für alle Mitbewohner funktionieren soll.

ja-felix, 2025-04-16:
Seit wenigen Wochen häufen sich bei mir auch die gleichen Probleme mit dem Tibber Pulse in Kombination mit Home Assistant und evcc.  Ich habe eben HA neu gestartet und daraufhin lief evcc auch wieder problemlos, aber die Tibber-Sensoren zeigen aktuell keine Werte mehr in HA.

whisky2k2, 2025-04-17:
Ich habe leider den Anfang verpasst, weil ich gerade mit der Tesla API zu kämpfen hatte. Vielleicht hilft es ja dennoch. Im Anhang ein Trace über alle Bereiche

[evcc-20250417-192835-trace.log](https://github.com/user-attachments/files/19799697/evcc-20250417-192835-trace.log)

jpvolkmann, 2025-04-25:
Just got an email from Tibber that my access token is used to connect to the tibber API more than 20 times / hour which is the rate limit.
Analyzing the source code (I'm not a GO developer...), I realized that the Tibber integration in EVCC is using `NewClient()` function 

https://github.com/hasura/go-graphql-client/blob/master/example/tibber/client.go

instead of "`graphql.NewSubscriptionClient`" used in the example of 

https://github.com/hasura/go-graphql-client/blob/3589de287865209ff835348469a34e610abeb038/example/tibber/client.go.

Tibber recommends to use the Subscription instead of connecting again for each request.

Schneeschuher, 2025-04-25:
Dasselbe Problem bei mir. Hab auch die Mail von Tibber bekommen.

GrimmiMeloni, 2025-04-25:
@jpvolkmann can you please elaborate further what exactly evcc is doing wrong?

https://github.com/evcc-io/evcc/blob/6f3321c709566424656770e74fb84ac716bd1049/meter/tibber-pulse.go#L74

whisky2k2, 2025-04-28:
Das Update von 203.1 auf 203.4 hat kurzzeitig das Thema too many connections geheilt (gestern 20 Uhr durchgeführt, seit 11 Uhr heute ein anderer Fehler)

[pulse ] TRACE 2025/04/28 11:22:57 {"id":"66623a17-8811-4f6d-a492-91b0eb69af40","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":426,"lastMeterConsumption":5711.0444,"lastMeterProduction":7582.8858,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/28 11:23:00 {"id":"66623a17-8811-4f6d-a492-91b0eb69af40","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":437,"lastMeterConsumption":5711.0444,"lastMeterProduction":7582.8863,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/28 11:24:30 failed to read JSON message: failed to get reader: context deadline exceeded server
[pulse ] TRACE 2025/04/28 11:24:36 {"type":"connection_init","payload":{"token":"***"}} client
[pulse ] TRACE 2025/04/28 11:24:37 {"type":"connection_ack"} server
[pulse ] TRACE 2025/04/28 11:24:37 {"id":"08f3620a-e072-4cbb-99da-be563ae38bb8","type":"subscribe","payload":{"query":"subscription ($homeId:ID!){liveMeasurement(homeId: $homeId){power,powerProduction,lastMeterConsumption,lastMeterProduction,currentL1,currentL2,currentL3}}","variables":{"homeId":"5c5eea54-c342-4bb3-80a7-fe6adab180da"}}} client
[pulse ] TRACE 2025/04/28 11:24:37 {"id":"08f3620a-e072-4cbb-99da-be563ae38bb8","type":"next","payload":{"errors":[{"message":"unable to start stream api:a1d68ecb64cc4c8caa75bc4ab6fbf24c for device a1d68ecb-64cc-4c8c-aa75-bc4ab6fbf24c; userId e6d2fab0-ee12-4ee0-bee0-c250f0ec4768","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server
[pulse ] ERROR 2025/04/28 11:24:37 Message: unable to start stream api:a1d68ecb64cc4c8caa75bc4ab6fbf24c for device a1d68ecb-64cc-4c8c-aa75-bc4ab6fbf24c; userId e6d2fab0-ee12-4ee0-bee0-c250f0ec4768, Locations: [{Line:1 Column:28}], Extensions: map[], Path: [liveMeasurement]
[pulse ] TRACE 2025/04/28 11:24:37 failed to read JSON message: failed to read: use of closed network connection client
[pulse ] TRACE 2025/04/28 11:24:42 {"type":"connection_init","payload":{"token":"***"}} client

jpvolkmann, 2025-04-28:
@toini might can update what's happening on server in this case?

TRACE 2025/04/28 11:24:37 {"id":"08f3620a-e072-4cbb-99da-be563ae38bb8","type":"next","payload":{"errors":[{"message":"unable to start stream api:a1d68ecb64cc4c8caa75bc4ab6fbf24c for device a1d68ecb-64cc-4c8c-aa75-bc4ab6fbf24c; userId e6d2fab0-ee12-4ee0-bee0-c250f0ec4768","locations":[{"line":1,"column":28}],"path":["liveMeasurement"]}]}} server

seems to be a message from the API

toini, 2025-04-28:
> [@toini](https://github.com/toini) might can update what's happening on server in this case?
>
> TRACE 2025/04/28 11:24:37 [...]

Yes, this seems to be a restart of a backend service that occurred around 11:20. Due to how the particular service is setup we might currently end up in a failure if a client tries to connect at the exact same time. We will be addressing the issue in the future.

Note however, that it was not the graphql-ws service that was restarted so any websocket connection you already had would have stayed up and new data would have been sent when the backend service is back up. Thus, if a client keeps on constantly reconnecting (e.g. every 10 seconds) you are bound to eventually hit this problem.

jpvolkmann, 2025-04-28:
Maybe timeout of 90s is not sufficient for the API. It is exactly timing out after 90, the currently hard-coded timeout.

[pulse ] TRACE 2025/04/28 11:**23:00** {"id":"66623a17-8811-4f6d-a492-91b0eb69af40","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":437,"lastMeterConsumption":5711.0444,"lastMeterProduction":7582.8863,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2025/04/28 11:**24:30** failed to read JSON message: failed to get reader: context deadline exceeded server

Unfortunately I don't know how to pass configuration parameters to the tibber modules to make this configurable. 
Might need support from an experienced EVCC developer to add this and test with an even longer timeout than 90s.

What kind of internet connection do you use? Does it sometimes re-connect or do you get a new IP address assigned? May check in your router...

SebStange, 2025-05-02:
Looks like 203.4 still has issues...

> We have detected a high number of connection initialization requests to the Tibber Websocket API (wss://websocket-api.tibber.com). You have likely set up a connection with a personal access token from Tibber developer console at developer.tibber.com or via a third-party OAuth2 service.

> Your client is currently using the user agent evcc/v0.0.0 hasura/go-graphql-client/v0.13.2 and exceeding the allowed rate of 20 new connections per hour.
It appears you are running evcc, that has known issue. Please make sure you are running the version 0.203.4 or later.

> This is a notification only — your connections have not been blocked yet. In the future we will start blocking your connections.

> [...]

> Rate limit exceeded:
> 2025-04-24T09:58:15.443Z
> 2025-04-28T11:27:41.434Z
> 2025-05-02T12:26:35.596Z

...anything specifically I can contribute in terms of logging etc.?

nc-kristema, 2025-05-02:
Das gleich e Problem habe ich auch !! Auch mit der neusten version 0.203.4
Ein paar Auszüge aus dem TRace/Debug.

[pulse ] ERROR 2025/05/02 16:41:29 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[pulse ] ERROR 2025/05/02 16:41:35 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[pulse ] ERROR 2025/05/02 16:41:40 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[pulse ] ERROR 2025/05/02 16:41:45 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[pulse ] ERROR 2025/05/02 16:41:51 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[pulse ] ERROR 2025/05/02 16:41:56 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"
[site ] DEBUG 2025/05/02 16:41:58 ----
[lp-1 ] DEBUG 2025/05/02 16:41:58 charge power: 0W
[lp-1 ] DEBUG 2025/05/02 16:41:58 charge currents: [0 0 0]A
[site ] ERROR 2025/05/02 16:41:58 grid power: outdated
[pulse ] ERROR 2025/05/02 16:42:01 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent go-graphql-client/0.9.0"

jpvolkmann, 2025-05-02:
nicht wirklich... Ich habe zuletzt direkt auf die Mail von Tibber geantwortet...

toini, 2025-05-02:
@nc-kristema you don't seem to be running the latest version since your user agent is set to go-graphql-client/0.9.0?

nc-kristema, 2025-05-02:
@toini die Version 0.203.4 funktioniert auch nicht somit liegt es daran vermutlich nicht.

jpvolkmann, 2025-05-02:
> [@toini](https://github.com/toini) die Version 0.203.4 funktioniert auch nicht, somit liegt es daran vermutlich nicht.

Bitte trotzdem die aktuelle Version installieren damit wir den anderen Fehler ausschließen können

nc-kristema, 2025-05-02:
Habe ich wieder installiert und prompt kommt die Meldung
failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = \"Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.13.2\

Mal bis morgen warten, vielleicht wird die 2 session nachts geschlossen. Das hatte ich schon mal, dass es dann wieder für eine Zeit funktioniert.

toini, 2025-05-02:
It appears there now are two connections open on (all?) the services on the cluster and they were not properly closed when evcc disconnected. We've added this protection since some clients open enormous amounts of connections. Primary fix would be to ensure the client gracefully closes the connection so that the server also becomes aware of that. Short term we can also restart the server so that the counters get reset. Can't do it now as im afk.

jpvolkmann, 2025-05-02:
> Habe ich wieder installiert und prompt kommt die Meldung failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on [...]

In welcher Umgebung läuft EVCC? Docker, Raspberry Pi ....? 
Wird EVCC korrekt runtergefahren oder einfach "der Stecker gezogen"?

Eventuell werden die offenen Verbindungen nicht korrekt geschlossen und somit das Limit erreicht.

toini, 2025-05-02:
If the laptop randomly shuts down, the connection might be left unterminated. @nc-kristema would you send your email address to me toni@tibber.com so i can take a look at the logs?

bytec77, 2025-05-04:
Ich habe auch das Problem das seit ein paar Tagen in HomeAssistant keine Werte mehr ankommen bei obiger Fehlermeldung. eVCC lief problemlos. Nun habe ich EVCC mal gestoppt (läuft bei mir in einem LXC also nur evcc) dann bekomm ich die Werte in HomeAssistant wieder angezeigt. 
Möchte ich aber jetzt wieder evcc starten bekomme ich in evcc die Meldung von zu vielen connections. Kann man auf die API jetzt nur noch mit einer Verbindung zugreifen ?

andig, 2025-05-04:
> It appears there now are two connections open on (all?) the services on the cluster and they were not properly closed when evcc disconnected. 

@toini I don't see how a stopped process should keep connections open? Imho OS would could any resource associated to a terminated service.

@bytec77 zuallererst bitte mal auf 0.203.4 aktualisieren.

bytec77, 2025-05-04:
Es lief schon die 0.203.4. Hm also wenn jetzt evcc läuft dann bekomme ich in HA keine Werte mehr. Stoppe ich EVCC dann gehts direkt wieder. Komisch ist erst seit ein paar tagen so.

SebStange, 2025-05-04:
Nach einem Neustart das HA Green ist EVCC nun nicht mehr startbar und hängt in einem Loop...

[49686a9f_evcc_2025-05-04T18-38-47.741Z.log](https://github.com/user-attachments/files/20029070/49686a9f_evcc_2025-05-04T18-38-47.741Z.log)

jpvolkmann, 2025-05-04:
> Nach einem Neustart das HA Green ist EVCC nun nicht mehr startbar und hängt in einem Loop...
> 
> [49686a9f_evcc_2025-05-04T18-38-47.741Z.log](https://github.com/user-attachments/files/20029070/49686a9f_evcc_2025-05-04T18-38-47.741Z.log)

Da scheinen noch weitere Probleme in der Netzwerkverbindung zu sein...

Verbindung zu github schlägt fehl, ebenso die ersten zwei Verbindungsversuche zu Tibber. 
Eventuell kommen diese dort an, erzeugen eine neue Connection, aber die antwort wird geblockt. 

Der dritte Verbindungsversuch endet dann mit "Too many connections"

```
[main  ] FATAL 2025/05/04 11:14:04 will attempt restart in: 15m0s
[main  ] ERROR 2025/05/04 11:14:09 version check failed: 

Get "https://api.github.com/repos/evcc-io/evcc/releases/latest": dial tcp 140.82.121.5:443: i/o timeout (installed: 0.203.4)

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/04 11:16:55 evcc 0.203.4
[main  ] INFO 2025/05/04 11:16:55 using config file: /config/evcc.yaml
[db    ] INFO 2025/05/04 11:16:55 using sqlite database: /data/evcc.db
[main  ] INFO 2025/05/04 11:17:02 listening at :7070
[main  ] FATAL 2025/05/04 11:17:07 meter [db:5] cannot create meter 'db:5': cannot create meter type 'template': cannot create meter type 'tibber-pulse': 

Message: Post "https://api.tibber.com/v1-beta/gql": dial tcp: lookup api.tibber.com: i/o timeout, Locations: [], Extensions: map[code:request_error], Path: []

[main  ] FATAL 2025/05/04 11:17:07 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/04 11:19:57 evcc 0.203.4
[main  ] INFO 2025/05/04 11:19:57 using config file: /config/evcc.yaml
[db    ] INFO 2025/05/04 11:19:57 using sqlite database: /data/evcc.db
[main  ] INFO 2025/05/04 11:20:07 listening at :7070
[main  ] FATAL 2025/05/04 11:20:12 meter [db:5] cannot create meter 'db:5': cannot create meter type 'template': cannot create meter type 'tibber-pulse': 

Message: Post "https://api.tibber.com/v1-beta/gql": dial tcp: lookup api.tibber.com: i/o timeout, Locations: [], Extensions: map[code:request_error], Path: []
[main  ] FATAL 2025/05/04 11:20:12 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
```

andig, 2025-05-05:
> Verbindung zu github schlägt fehl, ebenso die ersten zwei Verbindungsversuche zu Tibber.

Du hast vmtl. ein allgemeines Netzwerkproblem. Hat hiermit nichts zu tun.

toini, 2025-05-05:
> [@toini](https://github.com/toini) I don't see how a stopped process should keep connections open? Imho OS would could any resource associated to a terminated service.

No, it's actually the other way around: a connection needs to be closed in order for the server to notice it closed. During normal operation the connection is not "kept open" by anything, the server just sends data to the socket and doesn't know if a client is there or not. It's discussed for example [here](https://stackoverflow.com/questions/46111656/must-websockets-have-heartbeats). To detect a client is gone, there shuold probably be some kind a heartbeat mechanism (ping/pong) but we dont have such.

toini, 2025-05-05:
> If the laptop randomly shuts down, the connection might be left unterminated. [@nc-kristema](https://github.com/nc-kristema) would you send your email address to me [toni@tibber.com](mailto:toni@tibber.com) so i can take a look at the logs? I've restarted the servers now so any counters would get reset.

@nc-kristema I got your email addess and in the logs I your device is constantly disconnecting. It appears you are running the old version as user agent is reported as `go-graphql-client/0.9.0`. Could you make sure to upgrade to latest version?

[image]

SebStange, 2025-05-05:
Ended up disabling the Tibber query entirely for most of today, then updated to 0.203.5 and added pricing query before the actual grid/pulse query. Pricing requests worked w/o any issues, but as soon as I added the grid/pulse query again, the log starts showing issues again and I have no active pulse data in EVCC.

[evcc-20250505-162625-debug.log](https://github.com/user-attachments/files/20038573/evcc-20250505-162625-debug.log)

nc-kristema, 2025-05-05:
Ja, aktuelle habe ich eine ältere Version 0.200.1 drauf. Da frieren zumindest die Tibber werte nicht mehr  eine und das Auto kann Überschussladen vollziehen.
die aktuellste Version friert nach ca1-1,5h ein.

andig, 2025-05-06:
@SebStange Dein Log fängt sofort mit Fehlern an. Was auch immer das Problem ist- es hat schon vor dem Start begonnen und ist hier nicht ersichtlich.

@toini just to be sure: are you seeing any issues with the properly identified evcc 203.4 or 203.5 versions in your logging?

toini, 2025-05-06:
> [@toini](https://github.com/toini) just to be sure: are you seeing any issues with the properly identified evcc 203.4 or 203.5 versions in your logging?

I think @SebStange is talking about the api.tibber.com which used for querying e.g. prices. The issue in this ticket is about the websocket API. 

During the past 6 hours i can see several upgraded agents reconnecting websockets often:
* 1 unauthenticated running 2000 requests per hour
* 1 unauthenticated running 750 requests per hour
* 8 authenticated ones running 20-80 requests per hour

I also did confirm the fix helped several clients stop continuous reconnects.

[image]

toini, 2025-05-06:
Looking at the past 2 hours: i can actually see that it's 4 different ips with invalid tokens reconnecting at constant rate. 

[image]

SebStange, 2025-05-07:
Let's see if I can bring some more light into this...

In my statements above, I was referring to both Tibber endpoints, since I didn't know about the difference and to me it was unclear where the issues were with the Tibber integration in general. After playing with it, at some point I was neither getting pricing information (as stated by @toini before via api.tibber.com), nor was there any pulse/websocket API information. [Feel free to correct me on terminology, when I still confuse things and/or give me some pointers as to where I need to clarify.)

Yesterday I ended up removing both, doing soft (just EVCC, then EVCC & HA) and hard reboots of the host system (HA Green) and was able to get at least pricing back, but whatever I tried I couldn't get any pulse/websocket API data on actual consumption etc. at which point I decided to remove that entirely for now.

As you can see in the log below, EVCC logging connection errors (last one at ERROR 2025/05/05 22:23:11), until a final reboot at INFO 2025/05/05 22:23:15. Mind you, there was no pulse/websocket API info configured for more than 24h and had not been added then either. From then on I had no more connection errors, which I noticed just a few minutes ago and re-added the pulse/websocket API query at INFO 2025/05/07 14:03:08. The log has been clean since and I will update you/comment later on any further status changes.

[49686a9f_evcc_2025-05-07T12-04-04.796Z.log](https://github.com/user-attachments/files/20083462/49686a9f_evcc_2025-05-07T12-04-04.796Z.log)

toini, 2025-05-08:
> [@toini](https://github.com/toini) I don't see how that can happen. The token is required in any case to retrieve the `WebsocketSubscriptionUrl`. Without that we abort and don't even init the graphql client:

Could it be that the integration was already initialized and the token was revoked during operation, without restarting it?

andig, 2025-05-08:
> Could it be that the integration was already initialized and the token was revoked during operation, without restarting it?

I don't know. Can you revoke tokens on the Tibber side? Even if- that should not be a regularly occurring case? Technically I'm not sure if the graphql request would even send the token with each request or use an open connection? We could still debug it if there's any way to construct such a scenario.

toini, 2025-05-09:
> > Could it be that the integration was already initialized and the token was revoked during operation, without restarting it?
> 
> I don't know. Can you revoke tokens on the Tibber side? 

Yes, we can. But I suspect these devices are actually using the demo token available for anyone at developer.tibber.com. It was revoked and renewed just a week ago. That happens rarely so people might have been running the demo setup for a long time.

erdbeertoni, 2025-05-11:
Same here, since weeks not able to use evcc and getting Tibber warning mails even I am on the latest versions.

dundiffrunt, 2025-05-13:
Same here.

This morning I got errors every few seconds.

[pulse ] ERROR 2025/05/13 09:23:23 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"

toini, 2025-05-13:
> [@toini](https://github.com/toini) are you still seeing issues with authenticated clients of current version in monitoring?

Yes, I do still see many continuously reconnecting with clients set to `evcc/v0.0.0`. Here's top 10 during the past 1 day:

[image]

foto-andreas, 2025-05-13:
One of them its perhaps me. I got the mail today from tibber and I am using nightly. I‘ll dig into the logs next…

Andreas

foto-andreas, 2025-05-13:
Current Status:
* I am using docker container, running nightly. Updated normally every changed build with watchtower
* UI not available at the moment 404 due to traffic finding the evcc-container unhealthy
* Docker processes
````
 9375f0c35b15   evcc/evcc:nightly              "/app/entrypoint.sh …"   8 hours ago    Up 8 hours (unhealthy)   4712/tcp, 5353/udp, 7070/tcp, 7090/udp, 8899/udp, 8887/tcp, 9522/udp   evcc-py
2c918bbeded2   traefik:latest                 "/entrypoint.sh --lo…"   7 days ago     Up 7 days                80/tcp, 0.0.0.0:443->443/tcp, [::]:443->443/tcp                        traefik
f160096a63de   evcc/evcc:latest               "/app/entrypoint.sh …"   7 days ago     Up 4 minutes (healthy)   4712/tcp, 5353/udp, 7070/tcp, 7090/udp, 8899/udp, 8887/tcp, 9522/udp   evcc-wk
6848668c9ccc   willfarrell/autoheal:1.2.0     "/docker-entrypoint …"   3 months ago   Up 2 months (healthy)                                                                           autoheal
becd5795d15d   containrrr/watchtower:latest   "/watchtower"            3 months ago   Up 2 months (healthy)    8080/tcp                                                               watchtower
```
* Curent log status:
```
evcc-py  | [site  ] ERROR 2025/05/13 13:39:13 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 13:39:26 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 13:39:41 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 13:39:56 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 13:40:11 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 13:40:28 grid power: outdated
```
* Last interesting log lines:
```
evcc-py  | [pulse ] ERROR 2025/05/13 10:25:55 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:00 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:05 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [site  ] ERROR 2025/05/13 10:26:10 grid power: outdated
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:11 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:16 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:21 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:26 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [site  ] ERROR 2025/05/13 10:26:27 grid power: outdated
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:31 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:36 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [site  ] ERROR 2025/05/13 10:26:40 grid power: outdated
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:42 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:47 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:52 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
evcc-py  | [site  ] ERROR 2025/05/13 10:26:56 grid power: outdated
evcc-py  | [pulse ] ERROR 2025/05/13 10:26:57 failed to read JSON message: failed to get reader: received close frame: status = StatusCode(4429) and reason = "Too many open connections on this server; count 2; user agent evcc/v0.0.0 hasura/go-graphql-client/v0.14.0"
[... cut ...]
evcc-py  | [site  ] ERROR 2025/05/13 11:53:11 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 11:53:26 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 11:53:41 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 11:53:55 grid power: outdated
evcc-py  | [pulse ] ERROR 2025/05/13 11:54:04 status = StatusCode(4408) and reason = "Connection initialisation timeout"
evcc-py  | [site  ] ERROR 2025/05/13 11:54:10 grid power: outdated
evcc-py  | [site  ] ERROR 2025/05/13 11:54:26 grid power: outdated
```

toini, 2025-05-13:
> I'm wondering what these are. Either old or nightly versions. I'll need to check.

It's the user agent we get in here because the module does not contain version number, only build date and signature etc. https://github.com/evcc-io/evcc/blob/d14c9ee49bb5069257a5422c7631487c4d2eac57/meter/tibber-pulse.go#L29. It wasnt possible to get the version number with a function call as it resulted in a circular dependency somehow and build failed.

toini, 2025-05-13:
This is the value of  `info.Main.Version`: `v0.0.0-20250507212914-6bb7dda6322a+dirty`. Version is empty but there is a timestamp.

toini, 2025-05-14:
I've been debugging the client and found an issue that leads to troubles. I've it posted in [here](https://github.com/hasura/go-graphql-client/issues/177) and will also make adjustments to make the server more resilient in this scenario.

toini, 2025-05-14:
We deployed a fix in the Tibber websocket server so things should improve now.

Here's a PR to improve logging and avoiding evcc reconnecting frequently when unauthorized https://github.com/evcc-io/evcc/pull/21216

toini, 2025-05-14:
@andig the weird build error about circular deps i got when first tried to pull util.Version seems to have gone away. Here's pull request to use that https://github.com/evcc-io/evcc/pull/21217

andig, 2025-05-14:
I had refactored that but didn't find out (yet) why it wouldn't be used sofar. Thanks for the PR!

andig, 2025-05-15:
The improved logging is in the nightly. Does anybody still have issues and could share a log?

foto-andreas, 2025-05-15:
Aktuell bei mir keine Probleme.

LG Andreas

nc-kristema, 2025-05-15:
Weiter hin Probleme. es kommen immer  -> grid power: outdated 

Und im LOG:
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[httpd ] TRACE 2025/05/15 19:29:52 GET /api/system/log
[pulse ] TRACE 2025/05/15 19:29:52 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:51804->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server

toini, 2025-05-15:
@nc-kristema all of the log entries you post are from the same second. Could you post logs from a wider timespan including startup.

andig, 2025-05-15:
…and make sure to use nightly!

nc-kristema, 2025-05-16:
Hallo zusammen, 
ich habe heute es nochmals neugestartet, beobachtet und es läuft den ganzen Tag schon stabil ohne das die Leistungswerte von Tibber "einfrieren". Ich beobachte weiter.

VG, Markus

nc-kristema, 2025-05-16:
Zu früh gefreut...und bekomme das ...

[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server
[pulse ] TRACE 2025/05/16 16:38:36 failed to read JSON message: failed to get reader: failed to read frame header: read tcp 192.168.178.61:59238->34.251.87.49:443: wsarecv: Eine bestehende Verbindung wurde softwaregesteuert durch den Hostcomputer abgebrochen. server

andig, 2025-05-16:
> https://github.com/evcc-io/evcc/issues/20502#issuecomment-2884798008

?

andig, 2025-05-16:
@nc-kristema so what? There‘s not even an error visible. Don‘t make us chase red herrings.

nc-kristema, 2025-05-16:
@toini : Where the nightly version is?

[8 more replies]

closed 2025-06-06 by github-actions[bot]