evcc-io/evcc #20504: Renault integration

enboer, 2025-04-09:
### Describe the bug

When the car is asleep, the wake up command or alternate wakeup do not work anymore for certain models. At least the R5 E-tech and Scenic E-Tech.
The endpoint used are not available for these cars (forbidden). To start charging you need to send a certain json that disables all the programs (that is what the Renault app is doing). On the Renault API github this alternate way is described (https://gist.github.com/remo-tech/e9ffa42ddb19b94b21687ffd766dd910).
For me personally it is not important to keep the existing program, as I use EVCC as the master, car is in Always charging). Maybe make it for the use possible to have a custom .json file that EVCC will upload, gives more freedom to the user if he requires it.
So depending on the car type (or configurable??) it should use this endpoint commerce/v1/accounts/{renaultAccountId}/kamereon/kcm/v1/vehicles/{vin}/ev/settings?country=FR and uploads a .json with the schema (all disabled).
Stop charging is not possible (as of now) also not in the Renault App, but as EVCC controls the charger this is not an issue.

Second issue is when EVCC something does a request ( eg battery status when I connect the car) without using the authorisation procedure. This fails because of the expired tokens. Not sure if EVCC keeps track of the token expiration, but at least here is fails. Immediately it does the login and the request again (then succesfull), but it should have done this earlier. 
Also during charging the battery status is requested. But the token is 900sec valid, this request is done at 930s.
I see this with this command: commerce/v1/accounts/{id}/kamereon/kca/car-adapter/v1/cars/{vin}/cockpit?country=DE and commerce/v1/accounts/{account}/kamereon/kca/car-adapter/v2/cars/{vin}/battery-status?country=DE

### Steps to reproduce

1. evcc send the charge/pause-resume or actions/charging-start -> Forbidden

...

### Configuration details

```yaml
just the Renault vehicle template
```

### Log details

```text
[renault] TRACE 2025/04/01 04:54:35 POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/bb116d2a-b618-4b54-a254-b48af70dfd38/kamereon/kca/car-adapter/v1/cars/xxxxxx/actions/charging-start?country=DE
[renault] TRACE 2025/04/01 04:54:35 {"data":{"attributes":{"action":"start"},"type":"ChargingStart"}}
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0202.1

andig, 2025-04-09:
Happy to take a PR!

enboer, 2025-04-10:
> Happy to take a PR!

thanks

andig, 2025-04-14:
How would the full request look like then?

enboer, 2025-04-15:
What I did this morning is first read the ev-settings:
https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/{{account_id}}/kamereon/kcm/v1/vehicles/{{vin}}/ev/settings

Then write the same settings back, but with a modified lastSettingsUpdateTimestamp value, a few minutes later than the value read.
If this is not changed it is not saved to the car. Maybe to wake up the car this is not needed to change this, but to be on the safe side it better to change and it does not harm.
Secondly the programActivationStatus should be set to false (in my case that was already the case when reading the data). There can be zero of more programs.

[image]

POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/{{account_id}}/kamereon/kcm/v1/vehicles/{{vin}}/ev/settings {settings json file}

andig, 2025-04-15:
> What I did this morning is first read the ev-settings:

Verwenden wir heute nirgendwo, siehe Code. Gerne PR wenn du eine konkrete Vorstellung der notwendigen Änderung hast.

enboer, 2025-04-15:
I do not mean that EVCC should send the real charging plan to the car. But this is the way to wake up the car.

Not sure if this is what you require as proposal:

This current EVCC call is incorrect for R5, Scenic E-tech:
POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/{{account_id}}/kamereon/kca/car-adapter/v1/cars/{{vin}}/actions/charging-start?country=DE
This fails and EVCC will not start the charging.

I should be:
POST https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/{{account_id}}/kamereon/kcm/v1/vehicles/{{vin}}/ev/ {settings json file}

This is the post json data with lastSettingsUpdateTimestamp should be the current date/time UTC time (Z-time, so hour -2):

{
    "lastSettingsUpdateTimestamp": "2025-04-15T11:55:40.266Z",
    "delegatedActivated": false,
    "chargeModeRq": "SCHEDULED",
    "chargeTimeStart": "21:00",
    "chargeDuration": 1615,
    "preconditioningTemperature": 21.0,
    "preconditioningHeatedStrgWheel": false,
    "preconditioningHeatedRightSeat": false,
    "preconditioningHeatedLeftSeat": false,
    "programs": [

    ]
}

This can be hardcoded (except the time) or give the user the possibility to supply a json file to be more flexible.

Second (minor) issue is when the car is connected (so when I put in the plug) EVCC does:

GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXX/battery-status?country=DE

This is not correct, it first needs to go through the login procedure as it is not yet authorized.

enboer, 2025-04-17:
Just tested the last 2 days. And just sending a fix .json file wakes up the car, so no modification on the json file is needed.

What do you need as proposal for taking this into a PR as there are a few issues at the moment ? This json file is also needed to enable the preheating of the car. That will be a nice addon.

andig, 2025-04-17:
Afaikt this „json file“ (=HTTP request) will set a charge plan. That‘s not what we want to do. We want to start charging if it doesn‘t by itself, but not commit to any prolonged period of charging. We would also require the car to continue charging _after_ the plan (we may set a short 1m plan for that purpose) if wallbox is still enabled.

If that‘s confirmed happy to take a PR.

enboer, 2025-04-18:
Second request for PR is about getting the battery status:
GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/XXXXXX/kamereon/kca/car-adapter/v2/cars/XXXXXX/battery-status?country=DE

This is called and causing an error at because it did not go through the authorization process:
- When car is plugged in
- When charging start at a certain time, determined by EVCC

- During charging. Every 15min is check the battery status during the charging process. This should be a bit earlier as the token is only valid for 900secs. I see that it call is at about 15:30 intervals which is too late.

In all situations it will then go through the authorization process. this can be done more optimal.

enboer, 2025-04-25:
Can you update me on what is going to happen with this issue?

georgekue, 2025-04-30:
At least I can confirm that R5 does not wake up regardless of which wake-up mechanism I choose in evcc, at least when the vehicle is in standby mode for more than a couple of minutes (5 to 10 min, I'd say).

Renault support also adviced me to set a charging plan over the app, as this would wake up the car. I can confirm that this workaround really works.

So, @enboer 's RESTful solution seems the way to go for the time being, i.e. until Renault offers another option. 

I'm still in contact with Renault support in Germany, and they claim to discuss the issue with the developers in France, but are not confident that this issue can be resolved soon.

enboer, 2025-04-30:
I currently am running a cronjob every 3 minutes that checks the EVCC mqtt messages. It checks if offeredCurrent>0, chargerenabled and if the chargepower is still 0, then run the POST ...[vehicles/{{vin}}/ev/ {settings json file} command (and first the required commands to login, get token etc). If the car does not wakeup, it will try again after 3 minutes as the Chargepower is still 0)
This is for me working fine at the moment.

@georgekue What have you as charging plan in the car? Or do you create a plan based on what EVCC wants to do?

I read on the Renault hacf-fr/renault api (https://github.com/hacf-fr/renault-api/issues/1348)  that for a Scenic after the update 'Moteur électrique 3.1.0' there is no need to run the wakeup at all, it seems it does not go to sleep anymore

But anyway, the current EVCC wakeup command (needed or not) is not correct for (Scenic and R5).  So I hope someone it willing to take this PR.

enboer, 2025-05-07:
@georgekue Here is my script which runs via crontab. The json file is uploaded. the real data is not that important as the lastSettingsUpdateTimestamp is in the past it does not update the data at the car, but it triggers to wakeup. That is enough.

[wakeupR5-evcc.txt](https://github.com/user-attachments/files/20084191/wakeupR5-evcc.txt)

[evsettings.json](https://github.com/user-attachments/files/20084178/evsettings.json)

georgekue, 2025-05-07:
@enboer Wow, thanks a lot. I'll point Renault support to the assumption that the problem ist gone for Scenic cars, so obviously, someone in France knows how to fix it.

closed 2025-05-19 by github-actions[bot]

MobilettoSoft, 2025-05-22:
@enboer I got a Scenic e-tech too and I'm experiencing the same issue with the wake-up command. It's not always happening: sometimes the charge starts without issues, sometimes not, probably depending if the car is effectively in a deeper sleep state or not. I never received the 3.1.0 engine update, from what I can see from the OBD scanner I'm still on the 2.1.0 version.  However, I made a [fork](https://github.com/MobilettoSoft/evcc_mobiletto) of evcc  introducing the changes you suggested for the wakeup command. Instead of the "alternativewakeup" option in evcc.yaml config I used a "wakeupmode" key with three options: default, alternative and new, so you have to modify the evcc.yaml file accordingly.
Maybe give it a try; if it really works, maybe I can raise a pull request.

enboer, 2025-05-23:
@MobilettoSoft thanks for this! However I do not have setup my system to be able to easily build a docker from this. I will see how far I will get.