evcc-io/evcc #20517: wake-up vehilce funktioniert bei Version 202.1 nicht mehr

muemmel1, 2025-04-09:
### Describe the bug

Hallo zusammen,
die Version 202.1 funktioniert mit meinem Audi im "spontanen Ladebetrieb" sehr gut. Wenn ich das Auto aber anschließe, um es nach hinterlegter Planung nachts zu laden, passiert nichts. Im Log erkenne ich keine Fehlermeldung. Die Ladestartzeit wird erreicht - es wird versucht "wake-up vehicle" - aber es passiert nichts. Nach 4 Versuchen wird die Startzeit immer weiter nach vorne verlegt und kein Ladestart findet statt.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/04/09 19:55:24 evcc 0.202.1
[main  ] INFO 2025/04/09 19:55:24 using config file: /home/thomas/evcc.yaml
[db    ] INFO 2025/04/09 19:55:24 using sqlite database: /home/thomas/.evcc/evcc.db

Konfiguration (/home/thomas/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: rct-power 
  usage: grid  
  host: 192.168.178.198
  name: grid1
- type: template
  template: rct-power 
  usage: pv  
  host: 192.168.178.198 
  name: pv2
- type: template
  template: rct-power 
  usage: battery  
  host: 192.168.178.198  
  name: battery3

chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.178.109  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: audi 
  title: Audi Q5 TFSI e  
  user: *****
  password: *****
  vin: *****
  capacity: 17.9  
  phases: 3  
  icon: car  
  cache: 15m  
  mode: pv  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
#  phases: 3
#  mincurrent: 6
#  maxcurrent: 32
  enable:
    delay: 1m # threshold must be exceeded for this long
    threshold: -1000 # grid power threshold (in Watts, negative=export)
#  guardDuration: 5m # switch charger contactor not more often 

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100 

#tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
    type: fixed
    price: 0.3831 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.063 # EUR/kWh

#push messages
messaging:
  events:
    connect: # vehicle connect event
      title: Car connected
      msg: "at loadpoint:${title} in mode:${mode}"
  services:
    - type: custom
      send:
        source: script        
        cmd: /bin/sh /home/evcc/evcc_connect.sh
        timeout: 400s

Version: `0.202.1`
```

### Log details

```text
Apr 09 03:12:45 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:45 ----
Apr 09 03:12:46 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:46 charge power: 0W
Apr 09 03:12:46 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:46 charge currents: [0 0 0]A
Apr 09 03:12:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:46 grid power: -0W
Apr 09 03:12:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:46 battery 1 power: 281W
Apr 09 03:12:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:46 battery 1 soc: 57%
Apr 09 03:12:49 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:49 pv 1 power: 0W
Apr 09 03:12:49 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:12:49 site power: 381W
Apr 09 03:12:49 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:49 charge total import: 3633.800kWh
Apr 09 03:12:49 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:49 charger status: B
Apr 09 03:12:49 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:49 plan: charge 1h4m1s between 2025-04-09 03:12:50 +0200 CEST until 2025-04-09 04:15:00 +0200 CEST (overruns by 1m51s, power: 22080W, avg cost: 0.383)
Apr 09 03:12:49 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:12:49 wake-up vehicle, attempts left: 4
Apr 09 03:13:15 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:15 ----
Apr 09 03:13:15 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:15 charge power: 0W
Apr 09 03:13:16 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:16 charge currents: [0 0 0]A
Apr 09 03:13:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:16 grid power: -0W
Apr 09 03:13:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:16 pv 1 power: 0W
Apr 09 03:13:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:16 battery 1 power: 253W
Apr 09 03:13:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:16 battery 1 soc: 57%
Apr 09 03:13:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:16 site power: 353W
Apr 09 03:13:16 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:16 charge total import: 3633.800kWh
Apr 09 03:13:16 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:16 charger status: B
Apr 09 03:13:16 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:16 plan: charge 1h4m1s between 2025-04-09 03:13:17 +0200 CEST until 2025-04-09 04:15:00 +0200 CEST (overruns by 2m18s, power: 22080W, avg cost: 0.383)
Apr 09 03:13:45 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:45 ----
Apr 09 03:13:46 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:46 charge power: 0W
Apr 09 03:13:46 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:46 charge currents: [0 0 0]A
Apr 09 03:13:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:46 grid power: -0W
Apr 09 03:13:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:46 battery 1 power: 253W
Apr 09 03:13:46 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:46 battery 1 soc: 57%
Apr 09 03:13:47 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:47 pv 1 power: 0W
Apr 09 03:13:47 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:13:47 site power: 352W
Apr 09 03:13:47 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:47 charge total import: 3633.800kWh
Apr 09 03:13:48 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:48 charger status: B
Apr 09 03:13:48 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:13:48 plan: charge 1h4m1s between 2025-04-09 03:13:48 +0200 CEST until 2025-04-09 04:15:00 +0200 CEST (overruns by 2m49s, power: 22080W, avg cost: 0.383)
Apr 09 03:14:15 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:15 ----
Apr 09 03:14:15 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:15 charge power: 0W
Apr 09 03:14:16 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:16 charge currents: [0 0 0]A
Apr 09 03:14:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:16 battery 1 power: 253W
Apr 09 03:14:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:16 battery 1 soc: 57%
Apr 09 03:14:16 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:16 grid power: -0W
Apr 09 03:14:19 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:19 pv 1 power: 0W
Apr 09 03:14:19 raspberrypi evcc[19095]: [site  ] DEBUG 2025/04/09 03:14:19 site power: 352W
Apr 09 03:14:20 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:20 charge total import: 3633.800kWh
Apr 09 03:14:20 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:20 charger status: B
Apr 09 03:14:20 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:20 plan: charge 1h4m1s between 2025-04-09 03:14:20 +0200 CEST until 2025-04-09 04:15:00 +0200 CEST (overruns by 3m21s, power: 22080W, avg cost: 0.383)
Apr 09 03:14:20 raspberrypi evcc[19095]: [lp-1  ] DEBUG 2025/04/09 03:14:20 wake-up vehicle, attempts left: 2
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

202.1

andig, 2025-04-09:
Da können wir leider auch nichts machen- es sei denn jemand findet raus, warum das wakeup nicht funktioniert. Gehts dennmit

    evcc vehicle

?

StefanSchoof, 2025-04-10:
Es gab gerade auch eine neue AGB und man musste sich sogar in der App neu anmelden. Es kann sein, dass ein Neustart das Problem löst.

muemmel1, 2025-04-10:
Guten Morgen,
vielen Dank für Eure Antworten. Ja, "evcc vehicle" liefert korrekte Daten. @StefanSchoof  dann kann es ja eigentlich auch nicht an der App oder Anmeldung bei Audi liegen, oder? Ich prüfe das aber gleich nochmals ....

andig, 2025-04-10:
>  Ja, "evcc vehicle" liefert korrekte Daten

Die Frage war ob dort der Wakeup funktioniert- denn um den geht es ja?!

muemmel1, 2025-04-10:
ok, verstanden. Das müsste ich dann zu dem Zeitpunkt parallel machen, wenn der Startzeitpunkt erreicht ist, aber noch keine Ladung startet, richtig?

muemmel1, 2025-04-10:
Wenn ich "evcc vehicle --wakeup" absende erscheint sofort in der Audi App die Meldung "Laden wird gestartet ...". Ich vermute, dass ich am Ladepunkt zu dem Startzeitpunkt keine Mobilverbindung zu meinem Audi habe, denn ohne Internetverbindung am Fahrzeug wird der "wakeup" dort nicht ankommen. Richtig? Ich prüfe das nochmals.

andig, 2025-04-10:
Das ist gut möglich. Wenn Du siehst dass es per `evcc vehicle` funktioniert dann ist es erstmal richtig implementiert.

closed 2025-04-10 by andig