evcc-io/evcc #20523: Solcast: Prognosedaten verschoben

BrickTop87, 2025-04-10:
### Describe the bug

Inspiriert durch das Ticket #20371 habe ich meine Solcast-Vorhersage auch mal genau angeschaut, weil ich auch die Vermutung habe, dass sie eine Stunde zu früh kommt.
Für morgen sagt sie im Zeitraum 6-7 Uhr morgens z.B. zwischen 16 und 122 W voraus:
[image]

Auf der Solcast-Seite sehe ich solche Werte erst zwischen 7-8 Uhr morgens:
[image]

Die Realität der letzten Tage sagt mir, dass Solcast zeitlich eher richtig liegt.

Das Json von Solcast zu den relevanten Zeiten lautet:
`{"pv_estimate":0,...,"period_end":"2025-04-11T04:30:00.0000000Z","period":"PT30M"},
{"pv_estimate":0.0332,...,"period_end":"2025-04-11T05:00:00.0000000Z","period":"PT30M"},
{"pv_estimate":0.1141,...,"period_end":"2025-04-11T05:30:00.0000000Z","period":"PT30M"},
{"pv_estimate":0.1437,...,"period_end":"2025-04-11T06:00:00.0000000Z","period":"PT30M"}`
Das ist UTC: 144 W um 6 Uhr UTC --> 8 Uhr MESZ.

Gibt es hier vielleicht ein Problem bei der Berechnung UTC vs. Lokalzeit?
Dass Solcast Zeitpunkte mit Periodenende angibt ist wohl berücksichtigt. Aber wird UTC zu lokal richtig umgesetzt?

### Steps to reproduce

-

### Configuration details

```yaml
xxx@NAS:~$ timedatectl
      Local time: Do 2025-04-10 11:32:54 CEST
  Universal time: Do 2025-04-10 09:32:54 UTC
        RTC time: Do 2025-04-10 09:32:54
       Time zone: Europe/Berlin (CEST, +0200)
 Network time on: yes
NTP synchronized: yes
 RTC in local TZ: no

evcc.yaml:
tariffs:
  solar:
    - type: template
      template: solcast
      site: xxx # Ressource ID deiner Anlage
      token: xxx # Solcast API Token
      interval: 3h # max. 10 Abrufe pro Tag erlaubt
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.202.1

andig, 2025-04-10:
Soweit ich sehe (und da solcast in Go implementiert ist) nehmen wir 1:1 die Zeiten welche der Forecast liefert:

```go
type Forecast struct {
	PvEstimate   float64   `json:"pv_estimate"`
	PvEstimate10 float64   `json:"pv_estimate10"`
	PvEstimate90 float64   `json:"pv_estimate90"`
	PeriodEnd    time.Time `json:"period_end"`
	Period       Duration
}
```

Es kann aber gut sein, dass die Interpretation falsch ist! Aktuell tun wir so, als wäre `pv_estimate` der Wert zum Zeitpunkt `period_end - period`:

```go
start := now.With(r.PeriodEnd.Add(-r.Period.Duration())).BeginningOfHour().Local()
rr := api.Rate{
	Start: start,
	End:   start.Add(time.Hour),
	Value: r.PvEstimate * 1e3,
}
```

Das ist wohl falsch? Gerne PR zur Korrektur.

BrickTop87, 2025-04-10:
> Es kann aber gut sein, dass die Interpretation falsch ist! Aktuell tun wir so, als wäre `pv_estimate` der Wert zum Zeitpunkt `period_end - period`:

Wenn es so wäre, wäre die Interpretation m.E. falsch. Gemäß der Interpretation der Solcast GUI würde ich sagen, der Wert von `pv_estimate` ist zum Zeitpunkt `period_end`.

> start := now.With(r.PeriodEnd.Add(-r.Period.Duration())).BeginningOfHour().Local()
> rr := api.Rate{
> 	Start: start,
> 	End:   start.Add(time.Hour),
> 	Value: r.PvEstimate * 1e3,
> }

Wenn ich den Code richtig verstehe, passiert das hier:
- z.B. JSON: `{"pv_estimate":0.1437,...,"period_end":"2025-04-11T06:00:00.0000000Z","period":"PT30M"}`
- --> PeriodEnd = 6:00 UTC
- `-r.Period.Duration()` --> 5:00 UTC
- `.BeginningOfHour()` --> 5:00 UTC
- `.Local()` --> 7:00 MESZ
- --> `Start` = 7:00 MESZ
- --> `End` = 8:00 MESZ

In der GUI-Darstellung und weiteren Berechnung sagt er damit:
- 7:00 MESZ : 143.7 W (=Solcast-Wert 6:00 UTC)
- 8:00 MESZ: 1713 W (=Solcast-Wert 7:00 UTC)

Richtig wäre:
- **8:00 MESZ** : 143.7 W (=Solcast-Wert 6:00 UTC)
- **9:00 MESZ**: 1713 W (=Solcast-Wert 7:00 UTC)

Damit wäre der Fix:
`start := now.With(r.PeriodEnd).BeginningOfHour().Local()`

Oder übersehe ich da etwas?

BrickTop87, 2025-04-10:
Wenn man die API-Doku von Solcast liest, sind vermutlich beide Ansätze falsch, bzw. zu einfach.
[image]

Der Wert von `pv_estimate` ist nicht der Wert zu einem bestimmten Zeitpunkt, sondern der Mittelwert über den Zeitraum `period`.
Im obigen Beispiel von 5-6 Uhr UTC --> 7-8 Uhr MESZ. Mathematisch am saubersten als Darstellung als einzelner Datenpunkt wäre wohl eine Zuordnung auf den Zeitpunkt 7:30 MESZ...

andig, 2025-04-10:
...oder wir interpolieren.

BrickTop87, 2025-04-10:
Wäre auch eine Variante.

Höhere zeitliche Auflösung geht nicht, oder? Solcast liefert ja mit 30 Minuten Auflösung, das würde das Thema entschärfen. Aber ihr habt die Architektur mit 1h-Intervallen gemacht, richtig?

andig, 2025-04-10:
Man kann ja das eine abfrage und das andere ausliefern. Gerne PR.

TomF79, 2025-04-10:
Wäre es eventuell noch möglich die Daten (wie beim HomeAssistant AddOn) von Sonnenaufgang bis Sonnenuntergang abzufragen? Das würde für das API Limit hilfgreich sein. In der Nacht braucht man ohnehin keine Aktualisierung.

BrickTop87, 2025-04-11:
Ich habe die Lösung jetzt ausprobiert. Die Vorhersage passt jetzt besser zur Realität. Auch der Scheitelpunkt liegt jetzt dort, wo ich ihn bei meiner Ausrichtung erwarten würde.

[image]

PR ist erstellt.

userwithoutpassword, 2025-04-11:
> Wäre es eventuell noch möglich bitte die Daten (wie beim HomeAssistant AddOn) NUR von Sonnenaufgang bis Sonnenuntergang abzufragen? (das eingestellte intervall sollte sich dann nur auf diesen Zeitraum beschränken) Das würde für das API Limit hilfreich sein. In der Nacht braucht man ohnehin keine Aktualisierung.

Die Vorhersage kann sich doch in der Nacht ebenfalls ändern

TomF79, 2025-04-11:
> Die Vorhersage kann sich doch in der Nacht ebenfalls ändern

Kann sie ja, würde aber bestimmt mit Start bei Sonnenaufgang reichen wenn die Aktualisierung beginnt.

closed 2025-04-12 by andig

martinschrauberx, 2025-05-11:
Hallo @andig 

ich habe hier das gleiche Problem mit der Verschiebung um 1 Stunde, trotz deiner Änderung. Ich habe die aktuelle Nightly installiert.

Gruß Martin

andig, 2025-05-11:
@BrickTop87 @martinschrauberx es würde helfen wenn ihr mal Soll/Ist nebeneinander legen könntet?

martinschrauberx, 2025-05-11:
Hier mal mein IST und SOLL.

[image]

[image]

[image]

[image]

Ich nutze die Vorhersage per API auch in Grafana, wobei auch ein Versatz von 1 Stunde zur tatsächlichen Produktion vorhanden ist.

Gruß Martin

BrickTop87, 2025-05-11:
Wie ich etwas weiter oben schon geschrieben habe, stimmt die Darstellung auf der Solcast-Website nicht mit ihrer eigenen API-Beschreibung überein. Mit "beide Ansätze" habe ich den damaligen von evcc, aber auch die GUI der Solcast-Website gemeint.
> Wenn man die API-Doku von Solcast liest, sind vermutlich beide Ansätze falsch, bzw. zu einfach. [image]
>
> Der Wert von `pv_estimate` ist nicht der Wert zu einem bestimmten Zeitpunkt, sondern der [...]

Für evcc haben wir jetzt den Ansatz gewählt, der mathematisch die beste Approximation liefert.
Bei mir (evcc v0.203.5) finde ich den Abgleich exzellent:
[image]
Lila die heutige Vorhersage, gelb die tatsächliche Erzeugung.

Wie passt es bei dir zu den real erzeugten Leistungen?

martinschrauberx, 2025-05-12:
Bei mir sieht das Ganze heute so aus:

[image]

wobei es ab dem Scheitelpunkt ja recht gut passt. Ich glaub, ich muss bei Solcast nochmal meine Einstellungen prüfen.

martinschrauberx, 2025-05-12:
Ich habe 3 Dachausrichtungen belegt, die ich mir bei Solcast unter 3 Accounts angelegt habe. Ich habe jetzt den Azimut in allen Accounts angepasst, die ich erst nur grob (direkt Ost, Süd und West) eingestellt habe. Eine Verschattung habe ich nur im Westen für die letzten 1,5 Stunden.