evcc-io/evcc #20524: Add toggle to exclude immersion heaters from EV stats

LukasPeer, 2025-04-10:
Hallo zusammen 👋

Erst einmal vielen Dank für das tolle Projekt und die ganze Arbeit, die dahintersteckt!

Ich hätte einen kleinen Feature-Wunsch:
Wäre es möglich, eine generelle Option (z. B. ein Häkchen oder eine Filterfunktion) einzuführen, mit der man bestimmte Verbraucher wie Tauch­sieder (z. B. AC Thor) von den Wallbox-Statistiken ausschließen kann?

In unserem Fall verfälscht der AC Thor die Statistik zur geladenen Energie erheblich, da er wie eine Wallbox gezählt wird. Dadurch ist es schwierig, den tatsächlichen Stromverbrauch der einzelnen Ladepunkte oder den Solar-Anteil fürs Fahrzeugladen realistisch auszuwerten.

Ein Filter oder eine Markierung wie „Kein Fahrzeugverbrauch“ oder „Tauch­sieder“ würde da sehr helfen, um die Statistiken übersichtlicher und aussagekräftiger zu halten.

Gibt es hierzu bereits eine Lösung oder wäre das grundsätzlich umsetzbar?

Vielen Dank nochmals für eure großartige Arbeit 🙌

Viele Grüße

---------------------------------------------------------------------------------------------------------------------

Hi there 👋

First of all, thank you for the great project and all the effort you put into it!

I have a small feature request:
Would it be possible to add a general option (e.g., a checkbox or filter) to exclude certain consumers such as immersion heaters (e.g., AC Thor) from the wallbox charging statistics?

In our case, the AC Thor significantly distorts the energy statistics, as it is counted similarly to an EV wallbox. This makes it difficult to properly analyze the real energy usage per wallbox or the actual solar charging rate for vehicles.

A filter or tag like "non-EV consumer" or "immersion heater" would really help to make the statistics clearer and more meaningful.

Is there already a way to do this, or would this be something that could be implemented?

Thanks again for your great work 🙌

Best regards

[image]

VolkerK62, 2025-04-10:
Du bekommst doch für jeden Ladepunkt eine Auswertung. 
Geht es dir um den Gesamtwert?

Idee:
Du kannst  das Demo-Meter mit Power: 0 definieren und das dem Laodpoint zuweisen.
Dann sollte der in der Statistik nicht mehr auftauchen.
https://docs.evcc.io/docs/devices/meters#demoz%C3%A4hler

naltatis, 2025-04-10:
Dieses Problem mit der Auswertung betrifft gerade alle Geräte die keinen "Ladevorgang" haben sondern eher permanent aktiv sein können. Wir wollen das in https://github.com/evcc-io/evcc/issues/13050 adressieren.

closed 2025-04-10 by naltatis