evcc-io/evcc #20536: Mennekes Amtron 4You/Compact: Recover from error state via ModBus

supachris, 2025-04-10:
Problem: After restart of EVCC (e.g. by update) the Mennekes Amtrom 4You 310 wallbox is in error state due to missing heartbeat. To recover from this, the only way is currently to power cycle the charger.

Solution: According to the newest Modbus register description there is a way to recover:
_If the wallbox is in error state due to a missing energy manager or after a reboot, the registers 
0x0D00, 0x0D05 and 0x0302 need to be written to set the wallbox back in the “Idle” state._ 

Alternative: Frustrating power cycle of the wallbox.

See discussion here: https://github.com/evcc-io/evcc/discussions/19896

[Modbus_RTU_Description_AMTRON_4You300_Compact_2.0s_Start_2.0s-v2.4.pdf](https://github.com/user-attachments/files/19693983/Modbus_RTU_Description_AMTRON_4You300_Compact_2.0s_Start_2.0s-v2.4.pdf)

andig, 2025-04-10:
Thats horrible. It would be so easy to just accept the return of the heartbeat… anyway, PR welcome 👍🏻

supachris, 2025-04-28:
Hi. Is there any news about the feature? I saw the PR, but obviously it's not ready made. Unfortunately, I'm not fit in this language, so I cannot help by myself. Thanks to all of you!

PR #20579 (2025-05-03): Mennekes-Compact: Fix reboot and phase switching
Fixes #20536
Ref https://github.com/evcc-io/evcc/discussions/19896#discussioncomment-12754956
closed 2025-05-03 by andig