evcc-io/evcc #20537: SoC eGolf wird nicht angezeigt

chfaurie, 2025-04-11:
### Describe the bug

Der SoC des eGolf wird bei dem PlugIn von openWB mit folgender Fehlermeldung nicht mehr angezeigt.

Error from EVCC: 2025/04/11 07:18:20 rpc error: code = Unknown desc = Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVWZZZAUZLW912844/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements": unexpected status: 400 (Bad Request) (LV0)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Error from EVCC: 2025/04/11 07:18:20 rpc error: code = Unknown desc = Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVWZZZAUZLW912844/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements": unexpected status: 400 (Bad Request) (LV0)
```

### Log details

```text
Error from EVCC: 2025/04/11 07:18:20 rpc error: code = Unknown desc = Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVWZZZAUZLW912844/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements": unexpected status: 400 (Bad Request) (LV0)
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-11:
Neue Terms&Conditions?

chfaurie, 2025-04-11:
Waren neu sind aber bestätigt seit 2 Wochen ca.

closed 2025-04-11 by chfaurie

AES-256-GCM, 2025-04-11:
I also accepted the new terms and conditions but get the same error message. After accepting those my entities in Home Assistant were available, again. Since this isn't the case for evcc I'm assuming it's related to the api call evcc does:

```bash
[lp-1 ] ERROR 2025/04/11 08:30:09 vehicle odometer: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVG<redacted>/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements": unexpected status: 400 (Bad Request)
```

StefanSchoof, 2025-04-11:
With the current update if the new terms and conditions VW invalidated all Tokens. This time a restart of evcc is required to get a new Token.