evcc-io/evcc #20551: EVCC in OpenWB funktioniert nicht mehr

Ped68, 2025-04-11:
### Describe the bug

Seit einigen Tagen funktioniert die Abfrage des SoC über EVCC in meiner OpenWB für meinen VW e-up! nicht mehr. Die Fehlermeldung lautet:

Error from EVCC: 2025/04/10 08:51:40 rpc error: code = Unknown desc = Get "[https://emea.bff.cariad.digital/vehicle ... asurements](https://emea.bff.cariad.digital/vehicle/v1/vehicles/*********/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements)": unexpected status: 400 (Bad Request) (LV0)

(die Sternchen sind meine FIN)

Vermutlich hat das Problem etwas mit den geänderten Nutzungsbedingungen für die VW Connect Services zu tun, denen ich am  08.04.25 auf der VW-Website zugestimmt habe.

Meine OpenWB Version ist: 1.9.304.0

Könnt Ihr mich bitte unterstützen?

Herzliche Grüße 

Peter

### Steps to reproduce

1.xx
2.
3.
...

### Configuration details

```yaml
Xx
```

### Log details

```text
Xx
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-11:
Probiers bitte nochmal

Ped68, 2025-04-11:
Jetzt funktioniert es wieder
Herzlichen Dank für die schnelle Hilfe

closed 2025-04-11 by andig