evcc-io/evcc #20558: Battary charge from grid should be SoC aware

behdahh, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
Oftentimes it depends of the SoC of the home battery if I want to charge from grid, not only on the price, so I can make sure that I have a certain minimum charge level reached for more expensive hours. It's pointless for me though to charge form grid when I'm already 60% charged and that is sufficient to cover my energy demand during the night - and on the next day I can maybe charge for free with solar power. Also it's frustrating that I can't set the power level that is used for grid charging (for example only charge at 2kW from Grid and not a max power)

**Describe the solution you'd like**
Add a SoC threshold to the charge from grid options for the home battery. It should also be possible to define the charge power level, so it doesn't charge always at max power from grid (setting a percentage is already possible in config, but a power level would be much more useful since I usually don't know how much power 100% grid power actually is).

**Describe alternatives you've considered**
As a workaround, I'm currently using a python script that adjusts the cost threshold for battery charging to some really high price when the SoC is low and it's night time, to force a battery charge at night if SoC is low. In the morning, I lower the threshold prices again with that script.

andig, 2025-04-12:
https://github.com/evcc-io/evcc/pull/20455 addes an api to control grid charging externally. It feels your requirement is rather specific. Instead of adding a soc limit it seems to make more sense to have specific logic externally.

closed 2025-04-12 by andig

behdahh, 2025-04-12:
Thank for the hint about that feature, I will adjust my script accordingly. I find it interesting though that you consider having a minimum SoC setting for the home battery "rather specific" since I think anyone with dynamic power price and a home battery would face that issue. For the vehicle you need a minimum SoC to guarantee a certain range (which is already nicely implemented in the UI), and for the house you need a minimum SoC to guarantee that you're not using expensive grid power at peak hours.

andig, 2025-04-12:
It is specific since you might add any other factor to the decision, e.g. pV forecast erc…