evcc-io/evcc #20559: Shelly 1EM grid meter with negative power (feed-in to grid) is shown as positive power (get power from grid)

abroeders, 2025-04-12:
### Describe the bug

Grid meter with Shelly 1EM show negative (feed in to grid) as power from grid (it ignores the minus sign)

[image]
[image]

[image]

### Steps to reproduce

1. Configure shelly 1EM as grid meter
2. Feed in power from battery to grid
3. Grid meter (in webui of shelly 1EM device) sees power as negative number (this is good)
4. evcc sees grid power as IN from grid. (positive number, see attached screenshots)
...

### Configuration details

```yaml
Configured in database
```

### Log details

```text
no error in logfile
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.202.1

abroeders, 2025-04-12:
Additional info from the TRACE log:
`[shelly] TRACE 2025/04/12 19:05:11 {"wifi_sta":{"connected":true,"ssid":"Homenet-IoT","ip":"192.168.100.200","rssi":-58},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"19:05","unixtime":1744477510,"serial":36641,"has_update":false,"mac":"BCFF4DFCD370","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":-3561.72,"reactive":596.44,"pf":-0.99,"voltage":234.12,"is_valid":true,"total":8294787.2,"total_returned":1507357.2},{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":234.12,"is_valid":true,"total":0.0,"total_returned":0.0}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":33820,"fs_size":233681,"fs_free":157628,"uptime":5919343}
[site ] DEBUG 2025/04/12 19:05:11 grid power: 3562W
[shelly] TRACE 2025/04/12 19:05:11 GET http://192.168.100.200/status`

In the JSON: "power":-3561.72
grid power: grid power: 3562W.    --> Missing the minus sign

thierolm, 2025-04-13:
Hi @abroeders , we are currently working on a stepwise refactoring of the shelly logic. One result will be the usage dependent handling of the power value. In future only for pv usage, the sign will be forced to be positive.

abroeders, 2025-04-13:
Is it possible to use something like 'a dummy gridmeter' that I update from home assistant while the refactoring still not finished? Can someone advise me on a temporary fix to make the grid metering work.

VolkerK62, 2025-04-13:
You can use a custom meter. But this must be configured in the yaml file.

```
meters:
- name: grid
  type: custom
  power:
    source: http
    uri: http://192.168.100.200/status
    jq: .emeters[0].power

site:
  meters:
    grid: grid
```

abroeders, 2025-04-13:
Thanks! That was what i'm looking for.

VolkerK62, 2025-04-21:
@abroeders  die "große" Shelly Änderung ist im Master. Du kannst hier zu machen.