evcc-io/evcc #20560: Support Midea Air Conditions (Heat & Cool) based on ESPHome

stevoboy, 2025-04-12:
It would be great, to be able to control the Air Condition based on a ESPHome module.

:
Air Condition Modes: Off = 0. Auto 1
Target Temperature = ?
Actual Temperature = ?
IP Adress 

The Logic would be the following:

1. Threshold <xxxx kwh> exceeded? If yes go to 2.
2. Target temperatur different more than x degrees than Actual temperature then go to 3 else go to 4
3. Mode = 1
4. Mode = 0

[image]

andig, 2025-04-12:
So enable is 1, disable is 2? You should already be able to do this using a `custom` charger.

closed 2025-04-13 by andig

stevoboy, 2025-04-13:
Yes, but having a Custom charger in ConfigUI would still be nicer. That way you are Not Limited to pyhsical Relais Like Shelly 1 etc. But can set your one logics Zoon the Virtual EVCC switch.

andig, 2025-04-13:
Yes,    that's already on the todo list.

stevoboy, 2025-04-13:
That would be Great. 

Is there and plan to allow Alternative switch on/off Parameters in the Future for heating and cooling devices?

In heating, you want to enforce switch on if a certain minimum temperature, eg the reference room or the water temp, drops below a threshold - maybe switch on within low tibber prices in the next x hours.

Same for cooling - if the target / current gap becomes to big you want to send the switch on command - and only then.

You can realize both in nodered, must maybe there is a use case for this in EVCC.

stevoboy, 2025-04-13:
I guess it is a General question if a devices inbuilt switch on/off Logic shall be considered in EVCCs Strategy. 
The Dimplex heatpump clearly indicates its wishes but this is not known or Even considered by EVCC - only the Surplus threshold. Possibly considering this might Create Chaos.[image]

andig, 2025-04-13:
What question? Sgready is boost/ no boost which is equivalent and works fine.

stevoboy, 2025-04-13:
Take the Example of an EV, 

If EVCC knew that the EV is present, ready for charging, or not ready for charging would help in load management.

If the EV is not present or not ready for charging, then no need to consider that load point in planning.