evcc-io/evcc #20561: Absturz beim Start des Ladevorgangs - Zaptec Pro - VW ID3

lukylukiluky, 2025-04-12:
### Describe the bug

Hallo, nach dem Update auf die Version 0.202.1 stürzt das System ab, wenn ich meinen VW ID3 zum Laden anschließe. Ich habe das System bereits neu aufgesetzt und eine neue Konfiguration angelegt, leider ohne Erfolgt. Ich bin leider mit meinem Wissen am Ende. Ich hoffe, hier kann jemand helfen. Viele Grüße 

### Steps to reproduce

1. System nach Update neu aufgesetz und neue Konfiguration erstellt
2. Mehrfach neu Verbunden mit dem gleichen fehler 
3. Wallbox neu gestartet 
...

### Configuration details

```yaml
lukas@raspi:~ $ sudo nano /etc/evcc.yaml
  GNU nano 7.2                                                                                                                                            /etc/evcc.yaml
sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-gen24
  usage: grid
  host: 192.168.1.63
  port: 502
  name: grid1
- type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.1.63
  port: 502
  name: pv2
- type: template
  template: fronius-gen24
  usage: battery
  host: 192.168.1.63
  port: 502
  name: battery3

chargers:
- type: template
  template: zaptec
  id: xxx
  user: xxx
  password: xxx
  name: wallbox5

vehicles:
- type: template
  template: vw
  user: xxx
  password: xxx
  vin: xxx
  title: ID.3
  capacity: 58
  timeout: 60s
  name: ev4

loadpoints:
- title: Werkstatt
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: Zuhause
  meters:
[... 132 lines cut ...]
Apr 12 18:07:15 raspi evcc[4086]:         github.com/evcc-io/evcc/cmd/root.go:325 +0x3c
Apr 12 18:07:15 raspi evcc[4086]: created by github.com/evcc-io/evcc/cmd.runRoot in go>
Apr 12 18:07:15 raspi evcc[4086]:         github.com/evcc-io/evcc/cmd/root.go:324 +0x2>
Apr 12 18:07:15 raspi systemd[1]: evcc.service: Main process exited, code=exited, stat>
Apr 12 18:07:15 raspi systemd[1]: evcc.service: Failed with result 'exit-code'.
Apr 12 18:07:15 raspi systemd[1]: evcc.service: Consumed 4.353s CPU time.
Apr 12 18:07:25 raspi systemd[1]: evcc.service: Scheduled restart job, restart counter>
Apr 12 18:07:25 raspi systemd[1]: Stopped evcc.service - evcc.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.202.1

StefanSchoof, 2025-04-12:
> I have verified that the issue is reproducible with the latest nightly build

Really?

https://github.com/evcc-io/evcc/issues/20160

lukylukiluky, 2025-04-12:
Sorry, my fault, I haven't tested it with the latest nightly build. Should I try that?

lukylukiluky, 2025-04-12:
Updating to the latest Nightli version fixed the issue for me. Thank you!

closed 2025-04-12 by lukylukiluky