evcc-io/evcc #20573: Unable to fix config issue when configuration is in database

SvenBunge, 2025-04-13:
### Describe the bug

Updated the evcc instalaltion to 0.200.3 and receive following error:

[image]

The wallbox (ABL MH1) is connected via USB and worked till the update. The right parameter for the box is 8N1 and not 8E1 - I guess the config change came through the update. But as you see the wallbox is not displayed anymore. Also the vehicle is missing. So I'm not able to fix the configuration. Everything is db driven in this installation. the evcc.yaml is the debian base.

It's running on an RPI 2B+. Only evcc is installed.
I checked the box with the nightly because 0.203.0 has been released couple of hours ago.

### Steps to reproduce

Puuhh, good question. Upgrading from I guess 0.201.x to 0.203.0 with an USB dongle attached wallbox

### Configuration details

```yaml
Hmm, does this work with the database configuration?
```

### Log details

```text
[main  ] INFO 2025/04/13 15:11:49 evcc 0.203.0
[main  ] INFO 2025/04/13 15:11:49 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/13 15:11:49 using sqlite database: /var/lib/evcc/evcc.db
[main  ] INFO 2025/04/13 15:11:52 listening at :7070
[main  ] FATAL 2025/04/13 15:11:52 charger [db:2] cannot create charger 'db:2': cannot create charger type 'template': cannot create charger type 'abl': invalid comset: '8E1'
[main  ] FATAL 2025/04/13 15:11:52 will attempt restart in: 15m0s
time="2025-04-13T15:13:10+02:00" level=info msg="Connecting to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully connected to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2025-04-13T15:13:10+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2025-04-13T15:13:10+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2025-04-13T15:13:10+02:00" level=info msg="Connecting to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully connected to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2025-04-13T15:13:10+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2025-04-13T15:13:10+02:00" level=info msg="Connecting to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully connected to 192.168.167.115:5033"
time="2025-04-13T15:13:10+02:00" level=info msg="hiding auth request for security, use debug >= 99 to debug authentication"
time="2025-04-13T15:13:10+02:00" level=info msg="successfully authenticated (level: AUTH_LEVEL_USER)"
time="2025-04-13T15:13:10+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2025-04-13T15:13:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2025-04-13T15:13:25+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2025-04-13T15:13:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2025-04-13T15:13:25+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.0

VolkerK62, 2025-04-13:
Use `evcc config` to see the configured devices and `evcc config delete db:xx` to delete the invalid device.
Thereafter, the config should be avaiblable again

andig, 2025-04-13:
> guess the config change came through the update

Don‘t think so

SvenBunge, 2025-04-13:
Thanks. deleting of the wallbox and loadpoint did the trick to have the ability again.
After investigating: The update hasn't changed the configuration. The configuration was:

```
db:2 {Type:template Title: Icon: Product:} map[baudrate:38400 comset:8E1 device:/dev/ttyUSB0 modbus:rs485serial template:abl]
```

But trying and trying with 0.203.0: With 8N1 got a timeout. 8E1 is throwing the error above.
I downgraded to 0.202.1 and 8E1 is working.

So the error is, that 8E1 is not working since 0.203.0 and my box needs 8E1 to work:
[image]
With 8E1 the illegal Comset Error occours.

With 0.202.1 it works with 8E1
[image]

VolkerK62, 2025-04-13:
>Doesn‘t changing in ui work?

If the charger is not available, there is no loadpoint visible in ui-config. So nothing can be changed.

RTTTC, 2025-04-13:
These exact two issues mentioned above happens to me also. Heidelberg / 8E1 , and unable to edit UI charger because UI fails to start.

andig, 2025-04-13:
The bug here ist that 8E1 doesn‘t seem to work anymore?

/cc @naltatis regarding loadpoint.

RTTTC, 2025-04-13:
And - at least for now it seems that problem is only with Heidelberg ? No complains from owners of other chargers yet.

SvenBunge, 2025-04-13:
> And - at least for now it seems that problem is only with Heidelberg ? No complains from owners of other chargers yet.

No. Look, here it is the ABL Wallbox.

cjander, 2025-04-18:
I have a similar issue with my vestel wallbox (configured via ui), the ip of the wallbox changed and evcc fails to startup and does not show up half of the configs like the wallbox. so I cannot fix the config and have to delete the wallbox. Imo the ui should start up as usual and load all configurations so the users have a chance to fix it. with the "old" config file, one would just fix it there and restart, but the ui config is just inaccessible.

closed 2025-08-03 by andig