evcc-io/evcc #20576: Ladelimit (Ziel-SOC) in % beim Fahrzeug als Voreinstellung hinterlegen

universal-dilettant, 2025-04-13:
**Is your feature request related to a problem? Please describe.**
Der eCorsa (01/22) hat keine interne Ladelimit / Ziel-SOC Begrenzung und lädt damit, ohne äußere Eingriffe, immer auf 100%. Auch in der Opel-App gibt es keine Möglichkeit, dies zu begrenzen.

**Describe the solution you'd like**
Bereits heute kann man in der Fahrzeug-Konfiguration unter "erweitereten Einstellungen / Charge Settings" Parameter wie "default mode", "phases" und "min / max current" einstellen. Es würde sich anbieten, auch das "Ladelimit" fahrzeugspezifisch zu hinterlegen und als Voreinstellung beim Anstecken zu übernehmen. 

**Describe alternatives you've considered**

**Additional context**
.

VolkerK62, 2025-04-13:
Gar kein Problem. Ist nur etwas "versteckt".
Auto am Laodpoint auswählen.
Unten Mitte auf "Plan", dann Tab "Ankunft" und "Standard-Ladelimit" festlegen.
Das wird dann beim Fahrzeug gespeichert.

universal-dilettant, 2025-04-13:
Wow, da wäre ich nie drauf gekommen. Danke.

andig, 2025-04-14:
/cc @naltatis re discoverability

closed 2025-04-14 by andig