evcc-io/evcc #20581: 0.203 breaks custom Enphase IQ battery configuration

Neverminda, 2025-04-13:
### Describe the bug

Die Abfrage der Enphase Battery unter angegebener Konfiguration führt mit 0.203 zu einer Fehlermeldung ein downgrade 0.201.1 bei gleicher Konfiguration funktiniert weiterhin.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: Enphase_battery
  type: custom
  power:
    source: http
    uri: http://envoy.fritz.box/ivp/livedata/status
    method: GET
    auth:
      type: bearer
      password:  ***token***
    insecure: true
    jq: .meters.storage.agg_p_mw
    scale: 0.001
  soc:
    source: http
    uri: http://envoy.fritz.box/ivp/livedata/status
    method: GET
    auth:
      type: bearer
      password: ***token***
    insecure: true
    jq: .meters.soc
  capacity: 7 # kWh
```

### Log details

```text
[site ] ERROR 2025/04/13 23:33:28 2025-04-13 23:33:27.139946147 +0200 CEST m=+25.554833020 !! 456ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:27.620887464 +0200 CEST m=+26.035774337 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:27.664900294 +0200 CEST m=+26.079787166 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:27.702944663 +0200 CEST m=+26.117831646 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:27.763936392 +0200 CEST m=+26.178823265 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:27.839010219 +0200 CEST m=+26.253897092 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:28.068137784 +0200 CEST m=+26.483024675 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:28.230385049 +0200 CEST m=+26.645271922 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:28.667266705 +0200 CEST m=+27.082153577 !! 3ms unexpected status: 401 (Unauthorized) 2025-04-13 23:33:28.996058105 +0200 CEST m=+27.410944977 !! 3ms unexpected status: 401 (Unauthorized)
[site ] ERROR 2025/04/13 23:33:28 battery 1 power: unexpected status: 401 (Unauthorized)
[site ] ERROR 2025/04/13 23:33:29 battery 1 soc: unexpected status: 401 (Unauthorized)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.0

andig, 2025-04-14:
Replace password key with token!

closed 2025-04-14 by andig

Neverminda, 2025-04-14:
Danke, das war die Lösung