evcc-io/evcc #20585: BMW 330e no connection

luckytt69, 2025-04-14:
### Describe the bug

Hi, 
since e few weeks, my BMW 330e lost connection to EVCC. 

### Steps to reproduce

1.Updated EVCC to latest version V0.203.0
2. creatted new BMW hcaptcha
3. Restart of Raspberry 
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 3dfa0231aec4944902fdff4cc72abad80a7782b4367cfcdb96565fe20d967db0

interval: 30s # control cycle interval

sponsortoken: xxx>
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: go-e-controller
  usage: grid
  host: xx.xx.xx.xx
  name: grid1
- type: template
  template: sma-inverter-speedwire
  usage: pv
  host: xx.xx.xx.xx
  password: xx
  name: pv2

chargers:
- type: template
  template: go-e
  host: xx.xx.xx.xx
  name: wallbox4

vehicles:
- type: template
  template: bmw
  title: BMW E330
  user: xx
  password: xx
  vin: xx
  capacity: 12
  hcaptcha: xx
>  name: ev3
```

### Log details

```text
[site ] DEBUG 2025/04/14 09:38:27 ----
[... 233 lines cut ...]
[lp-1 ] DEBUG 2025/04/14 09:52:57 charge power: 0W
[lp-1 ] DEBUG 2025/04/14 09:52:57 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/14 09:52:57 pv 1 power: 1055W
[site ] DEBUG 2025/04/14 09:52:57 grid power: -611W
[site ] DEBUG 2025/04/14 09:52:57 site power: -611W
[lp-1 ] DEBUG 2025/04/14 09:52:57 charge total import: 1679.647kWh
[lp-1 ] DEBUG 2025/04/14 09:52:57 charger status: B
[lp-1 ] INFO 2025/04/14 09:52:58 vehicle updated: unknown
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

V0.203.0

andig, 2025-04-14:
> since e few weeks, my BMW 330e lost connection to EVCC.

Logfile doesn't show any of that.

closed 2025-04-14 by andig

luckytt69, 2025-04-14:
[image]

But program.

luckytt69, 2025-04-14:
Sorry, but why do you close this topic?
Did I made something wrong?
My issue isn't solved.

luckytt69, 2025-04-14:
> > since e few weeks, my BMW 330e lost connection to EVCC.
> 
> Logfile doesn't show any of that.

Why are you closing this topic?
Did I made something wrong?
My issue isn't solved.

andig, 2025-04-14:
You will need a log that actually shows the BMW.