evcc-io/evcc #20586: v0.203.1 Does not start

Lepidopteron, 2025-04-14:
### Describe the bug

Neither v0.203.0 nor v0.203.1 do finish startup in Dockercontainer

Had to revert to v0.202.1 to generate the config for you

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
---
network:
 schema: https
 host: xxxxxx
 port: 7070

# interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
interval: 30s  # control cycle interval

# sponsor token: *****
sponsortoken: *****

# log settings
log: info

# meter definitions
# name can be freely chosen and is used as reference
# when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
 - name: grid
   type: template
   template: solaredge-hybrid
   usage: grid
   # Modbus TCP
   modbus: tcpip
   id: 5
   host: 0.0.0.0  # Hostname
   port: 502  # Port 1502
 - name: pv
   type: template
   template: solaredge-hybrid
   usage: pv
   # Modbus TCP
   # modbus: tcpip
   modbus: tcpip
   id: 5
   host: 0.0.0.0  # Hostname
   port: 502  # Port 1502
   
 - name: ACThor
   type: template
   template: thor
   usage: aux
   host: 192.168.4.18 # IP-Adresse oder Hostname

 - name: battery
   type: template
   template: solaredge-hybrid
   usage: battery
   # Modbus TCP
   modbus: tcpip
   id: 5
   host: 0.0.0.0  # Hostname
   port: 502  # Port

site:
 - title: XXXX
   meters:
    grid: grid
    pv:
     - pv
     - pv
    battery: battery

chargers:
 - name: garage
   type: template
   template: alfen
   # Modbus TCP
   modbus: tcpip
   id: 1
   host: 192.168.4.20 # Hostname
   port: 502 # Port
   
 - name: heizstab
   type: template
   template: ac-thor
   # Modbus TCP
   modbus: tcpip
   id: 1
   host: 192.168.4.18 # Hostname
   port: 502 # Port

vehicles:
  - name: EQB350
    type: template
    template: mercedes
    title: "EQB 350" # Will be displayed in the user interface (optional)
    user: *****
    region: EMEA
    accesstoken: *****
    refreshtoken: *****
    vin: *****
    capacity: 69.7 # Battery capacity in kWh (optional)
    icon: car
    phases: 0
    mode: pv

# loadpoint describes the charger
# charge meter and connected vehicle
loadpoints:
 - title: Garage
   charger: garage
   mode: pv
   phases: 0
   priority: 2
   disable:
    threshold: 200  # maximum import power (W)
    delay: 1m
    
 - title: Heizstab # display name for UI
   charger: heizstab # Charger reference
   meter: ACThor
   mode: pv # charge mode (off, now, minpv, pv)
   priority: 0
   phases: 3
   mincurrent: 0.125 # 0.1 kW
   maxcurrent: 17 # 6.2 kW
   enable:
    threshold: -300 # einschalten, wenn 1 Minute lang mindestens 300 W Uberschuss vorhanden ist
    delay: 1m
   disable:
    threshold: 300  # maximum import power (W)
    delay: 1m

tariffs:
 currency: EUR  # (default EUR)
 grid:
  # static grid price
  type: fixed
  price: 0.42984  # [currency]/kWh
 feedin:
  # rate for feeding excess (pv) energy to the grid
  type: fixed
  price: 0.071442184  # [currency]/kWh
  
modbusproxy:
 - port: 5200
   uri: 0.0.0.0:502  # Hostname

Version: `0.202.1`
```

### Log details

```text
[main  ] INFO 2025/04/14 09:45:57 evcc 0.203.1
[main  ] INFO 2025/04/14 09:45:57 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/14 09:45:57 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/04/14 09:45:57 connecting evcc-2117800841 at tcp://192.168.5.206:1883
[main  ] INFO 2025/04/14 09:45:57 listening at :7070
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.1

andig, 2025-04-14:
I can't reproduce this on docker, may be configuraiton-specific. Could you please start with `--profile` and show the goroutine profile from `/debug/pprof`?

Lepidopteron, 2025-04-14:
> I can't reproduce this on docker, may be configuraiton-specific. Could you please start with `--profile` and show the goroutine profile from `/debug/pprof`?

```
/app/entrypoint.sh: exec: line 53: evcc --profile: not found
```

docker compose file:
```
services:
  evcc:
    command:
      - evcc --profile
    #image: evcc/evcc:0.202.1
    image: evcc/evcc:latest
    ports:
      - 502:502
      - 7070:7070/tcp
      - 8887:8887/tcp
      - 9522:9522/udp
      - 4712:4712/tcp
      - 5200:5200/tcp # proxy
    volumes:
      - /mnt/evcc/.evcc:/root/.evcc
      - /mnt/evcc/etc/machine-id:/etc/machine-id
      - /mnt/evcc/dbus/machine-id:/var/lib/dbus/machine-id
    network_mode: host
    configs:
      - source: evcc
         target: /etc/evcc.yaml
    deploy:
      mode: replicated
      replicas: 1
      placement:
        constraints: [node.role == worker]
        
configs:
  evcc:
    external: true
```

andig, 2025-04-14:
You should probably reproduce this without docker first.

chris-e-codes, 2025-04-14:
Same here. no start in docke since 0.203.1

Log only shows:
/app # evcc --profile
[main  ] INFO 2025/04/14 14:14:57 evcc 0.203.1
[main  ] INFO 2025/04/14 14:14:57 using config file: /etc/evcc.yaml

sonst nichts. Stille :(

andig, 2025-04-14:
@chris-e-codes we still need https://github.com/evcc-io/evcc/issues/20586#issuecomment-2800976653. "me too" does not help.

chris-e-codes, 2025-04-14:
> [@chris-e-codes](https://github.com/chris-e-codes) we still need [#20586 (comment)](https://github.com/evcc-io/evcc/issues/20586#issuecomment-2800976653). "me too" does not help.

/debug/pprof also runs in timeout? do i need to enable something?

chris-e-codes, 2025-04-14:
i started with --profile
Then? curl to /debug/pprof? or go tool pprof "http://127.0.0.1:7070/debug/pprof/goroutine"?
curl and go are not available in the docker image :(
from outside of the image the request to /debug/pprof gets timeout

Lepidopteron, 2025-04-14:
Container ist stopped. However, seems to be a problem with the keba-modbus ?

evcc -c /home/tmp/evcc.yaml      --database /home/tmp/evcc.db      --profile

```
[main  ] INFO 2025/04/14 15:14:13 evcc 0.203.1
[main  ] INFO 2025/04/14 15:14:13 using config file: /home/traefik/evcc.yaml
[db    ] INFO 2025/04/14 15:14:13 using sqlite database: /home/traefik/evcc.db
[mqtt  ] INFO 2025/04/14 15:14:13 connecting evcc-893092360 at tcp://192.168.5.206:1883
[main  ] INFO 2025/04/14 15:14:13 listening at :7070
[main  ] FATAL 2025/04/14 15:14:16 charger [db:1] cannot create charger 'db:1': cannot create charger type 'template': cannot create charger type 'keba-modbus': read tcp 192.168.5.49:34804->192.168.4.240:502: read: connection reset by peer
[main  ] FATAL 2025/04/14 15:14:16 will attempt restart in: 15m0s
```

TRIAG73, 2025-04-14:
Same for mee.
Updated from 0.203.0 
It does not start.
I use Home Assistant green with add-on

andig, 2025-04-14:
Closing until pprof available

closed 2025-04-14 by andig