evcc-io/evcc #20588: go-e-v3 suddenly uses wrong URL /status instead /api/status

DocBru, 2025-04-14:
### Describe the bug

When running for a longer period of time, I often notice that evcc suddenly does not work anymore in combination with my go-e gemini charger and reports 404 errors in the logs.

The logged URL `/status`, Is wrong, it should be `/api/status`, which is also correctly returning data. The `status` URL is correct for the old go-e charger as far as I remember.

### Steps to reproduce

Keep evcc running for a long time.

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  # host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

sponsortoken: *****
plant: *****

# log settings
log: error
levels:
  core: debug
  lp-1: debug

# meter definitions
meters:
  - name: grid
    type: custom
    power: # power reading
      source: http # use http plugin
      uri: http://192.168.178.49/solar_api/v1/GetPowerFlowRealtimeData.fcgi
      jq: .Body.Data.Site.P_Grid
  - name: pv
    type: custom
    power: # power reading
      source: http # use http plugin
      uri: http://192.168.178.49/solar_api/v1/GetPowerFlowRealtimeData.fcgi
      jq: .Body.Data.Site.P_PV // 0

# charger definitions
chargers:
  - name: goe
    type: template
    template: go-e-v3
    host: 192.168.178.77 # go-e ip address (local)

# site describes the EVU connection, PV and home battery
site:
  title: abc # display name for UI
  meters:
    grid: grid # grid meter
    pv: pv # pv meter

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
  - title: go-e # display name for UI
    charger: goe # charger
    enable: # pv mode enable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 8m # threshold must be exceeded for this long
      threshold: 500 # maximum import power (W)
```

### Log details

```text
evcc stops functioning and shows this in the log

[lp-1  ] ERROR 2025/04/14 10:00:06 charge power: Get "http://192.168.178.77/status": dial tcp 192.168.178.77:80: i/o timeout
[lp-1  ] ERROR 2025/04/14 10:00:07 charge currents: unexpected status: 404 (Not Found)
[lp-1  ] ERROR 2025/04/14 10:00:07 charge voltages: unexpected status: 404 (Not Found)
[lp-1  ] ERROR 2025/04/14 10:00:07 charge total import: unexpected status: 404 (Not Found)

After restart everything is fine:

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/04/14 10:03:08 evcc 0.202.1
[main  ] INFO 2025/04/14 10:03:08 using config file: /config/evcc.yaml
[lp-1  ] DEBUG 2025/04/14 10:03:09 charge total import: 1086.085kWh
[lp-1  ] INFO 2025/04/14 10:03:09 loadpoint 1:
[lp-1  ] INFO 2025/04/14 10:03:09   mode:        off
[lp-1  ] INFO 2025/04/14 10:03:09   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/04/14 10:03:09   meters:      charge ✓
[lp-1  ] INFO 2025/04/14 10:03:09     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2025/04/14 10:03:09 phase timer inactive
[lp-1  ] DEBUG 2025/04/14 10:03:09 pv timer inactive
[lp-1  ] DEBUG 2025/04/14 10:03:09 charge power: 0W
[lp-1  ] DEBUG 2025/04/14 10:03:09 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/04/14 10:03:09 charge total import: 1086.085kWh
[lp-1  ] DEBUG 2025/04/14 10:03:09 charger status: B
[lp-1  ] DEBUG 2025/04/14 10:03:09 charge start energy: 1086.085kWh
[lp-1  ] INFO 2025/04/14 10:03:09 car connected
[lp-1  ] DEBUG 2025/04/14 10:03:09 pv timer elapse
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.202.1

Sillium007, 2025-04-14:
On startup of evcc the module checks if the V2 api is available. If it is not available it will fallback to the V1 api. Therefore it looks like at the time the check was done the goE did NOT answer to the V2 api request for whatever reason.

Code: https://github.com/evcc-io/evcc/blob/master/charger/go-e/api.go#L56

As always a log for that time would be helpful.

DocBru, 2025-04-14:
Currently I can not get logs anymore, since I upgraded the home assistant addon. Not sure if I could go back so much in the logs also, since I do not know when it happened.

Would it be possible to make this configurable, so that it never tries to use the V1 API even?

andig, 2025-04-14:
> When running for a longer period of time, I often notice that evcc suddenly does not work anymore in combination with my go-e gemini charger and reports 404 errors in the logs.

I don't think evcc does that, if yes we need a logfile.

> Is a downgrade also happening when connectivity to V2 is lost?

No.

If Go-E turns of v2 api for no good reason, I'd suggest to open a ticket with Go-E.

closed 2025-04-14 by andig

Sillium007, 2025-04-14:
The check is not configurabe.
Yes phase switching is not possible with v1: https://github.com/evcc-io/evcc/blob/master/charger/go-e.go#L233

Yes you could activate V1 to prevent the car from not charging at all.

Without log we cannot say what happened. Maybe the box was "just" not reachable at all at that moment and not only the V2 Api did not react.

DocBru, 2025-04-14:
V2 API is running fine, there might have been only a temporary connectivity issue (Wifi or so). I will try to get logs when it happens the next time.