evcc-io/evcc #20601: After update to 0.202.1, 30.03.25 no information from Audi E-Tron odometer

DeBo6, 2025-04-14:
### Describe the bug

Hello, 
After update to version 0.202.1 (my update was happened on 31.03.25) 
no more information about audi odometer.
communication with Sungrow wallbox over OCCP 1.6

Nach dem update, auf die Version 0.202.1, durchgeführt am 31.03.2025.
Wird kein Tachostand vom Audi E-Tron mehr bei den Ladevorgängen zurück gegeben.
Verbindung über OCCP mit der Sungrow Wallbox (3Phasen,11KW)

z.B. vor dem 31.03.25

<html>
<body>

Ladepunkt | WallboxWallbox
-- | --
ETRONGastfahrzeugETRON
So., 30. März, 16:50So., 30. März, 17:36
5,1 kWh46:10 m (~6,6 kW)
95,7 % (4,9 kWh)
0,43 €8,4 ct/kWh
43.363 km
4.208,8 kWh4.214,0 kWh

</body>
</html>

Nach dem 31.03.2025:
<html>
<body>

Ladepunkt | WallboxWallbox
-- | --
ETRONGastfahrzeugETRON
Mo., 31. März, 18:37Di., 1. Apr., 15:26
16,2 kWh2:31 h (~6,4 kW)
96,1 % (15,5 kWh)
1,35 €8,4 ct/kWh
4.214,0 kWh4.230,3 kWh

</body>
</html>

### Steps to reproduce

1. try from 31.03.2025 until now

### Configuration details

```yaml
Last change yaml at 22.03.2025
- type: custom
  name: charge7
  power:
    source: http
    uri: http://192.168.178.169:8087/getPlainValue/0_userdata.0.EVCC.chargepower
    method: GET
  energy:
    source: http
    uri: 
http://192.168.178.169:8087/getPlainValue/modbus.10.inputRegisters.4126_positive_Gesamtwirkenergie
    method: GET

chargers:
- type: template
  template: ocpp
  name: wallbox5
  remotestart: true
  connecttimeout: 2m
  meterinterval: 10s

vehicles:
- type: template
  template: audi
  user: *************
  password: *************
  vin: WAUZZZ***********
  title: ETRON
  capacity: 64
  name: ev4

loadpoints:
- title: Wallbox
  charger: wallbox5
  meter: charge7
  vehicle: ev4
  mode: pv #now -ist Sofort laden (Schnell)
  phases: 3
  mincurrent: 6
  maxcurrent: 16
```

### Log details

```text
no logfile at moment
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-14:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-04-14 by andig

DeBo6, 2025-04-22:
After updating to version 0.203.1 the odometer data is again available in the loading list.

DeBo6, 2025-04-22:
Issue can be closed now.