evcc-io/evcc #20608: Adjusted forecast totally off after evcc restart

CiNcH83, 2025-04-14:
### Describe the bug

Based on discussion #20535.

Adjusted forecast is totally off after evcc restart:
[image]

[evcc-20250410-220721-debug.log](https://github.com/user-attachments/files/19693915/evcc-20250410-220721-debug.log)

Comment by @VolkerK62:

die value `produced` is growing up every interval
So it is no problem with the forecast, it is a probem with the adjustment
```
[site  ] DEBUG 2025/04/10 20:20:06 solar forecast: accumulated 0.037kWh, produced 0.780kWh, scale 20.866
[site  ] DEBUG 2025/04/10 20:20:26 ----
[site  ] DEBUG 2025/04/10 20:20:27 pv 1 power: 0W
[site  ] DEBUG 2025/04/10 20:20:28 battery 1 power: 2225W
[site  ] DEBUG 2025/04/10 20:20:31 solar forecast: accumulated 0.037kWh, produced 0.800kWh, scale 21.401
```
diff 20Wh in 30s = 2400Wh in 1h

```
[site  ] DEBUG 2025/04/10 21:33:31 solar forecast: accumulated 0.037kWh, produced 2.570kWh, scale 68.751
[site  ] DEBUG 2025/04/10 21:33:56 ----
[site  ] DEBUG 2025/04/10 21:33:56 pv 1 power: 0W
[site  ] DEBUG 2025/04/10 21:33:57 battery 1 power: 1237W
[site  ] DEBUG 2025/04/10 21:34:01 solar forecast: accumulated 0.037kWh, produced 2.580kWh, scale 69.019
```
diff 10Wh in 30s = 1200Wh in 1h

correlation vs causality:
it seems, that the battery discharge power is taken into accout.

### Steps to reproduce

1. docker stop evcc
2. docker start evcc
3. check forecast

### Configuration details

```yaml
solar:
  - type: template
    template: open-meteo
    lat: 47.8146
    lon: 9.6238
    dec: 25
    az: -10
    kwp: 6.075
    #alphatemp: -0.350
    #efficiency: 90
    interval: 1h
  - type: template
    template: open-meteo
    lat: 47.8146
    lon: 9.6238
    dec: 6
    az: -10
    kwp: 3.240
    #alphatemp: -0.350
    #efficiency: 90
    interval: 1h
```

### Log details

```text
full log attached
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.0

andig, 2025-04-14:
> it seems, that the battery discharge power is taken into accout.

Then your meter setup is broken. You can check using

    evcc meter

andig, 2025-04-14:
Closing since meter config not available

closed 2025-04-14 by andig

CiNcH83, 2025-04-14:
```
Pi@raspberrypi5:~ $ sudo docker exec -it evcc /bin/sh
/app # evcc meter
[main  ] INFO 2025/04/14 16:13:06 evcc 0.203.0
[main  ] INFO 2025/04/14 16:13:06 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/14 16:13:06 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2025/04/14 16:13:06 Local SKI: 4081993d045b8582135f3185d462b911ee74354b
[eebus ] ERROR 2025/04/14 16:13:06 websocket server error: listen tcp :4712: bind: address already in use
[main  ] FATAL 2025/04/14 16:13:06 [db:7] cannot create meter 'db:7': cannot create meter type 'template': invalid key: timeout
```

May this stem from my IAMMETER tests with the external template? Is there some corpse left in the DB?

Here is the meter config from evcc.yaml:
```
meters:

- name: dtsu666
  type: template
  template: huawei-sun2000
  usage: grid

  modbus: tcpip
  id: 1
  host: 192.168.1.15
  port: 502
  timeout: 15s
  #modbus: rs485serial
  #id: 1
  #device: /dev/ttyUSB0
  #baudrate: 19200
  #comset: "8N1"

- name: sun2000
  type: template
  template: huawei-sun2000
  usage: pv

  modbus: tcpip
  id: 1
  host: 192.168.1.15
  port: 502
  timeout: 15s
  #modbus: rs485serial
  #id: 1
  #device: /dev/ttyUSB0
  #baudrate: 19200
  #comset: "8N1"

- name: luna2000
  type: template
  template: huawei-sun2000
  usage: battery
  storageunit: 1
  capacity: 5

  modbus: tcpip
  id: 1
  host: 192.168.1.15
  port: 502
  timeout: 15s
  #modbus: rs485serial
  #id: 1
  #device: /dev/ttyUSB0
  #baudrate: 19200
  #comset: "8N1"

#- name: wem3080t
#  type: template
#  template: iammeter
#  usage: grid
#  host: 192.168.1.16
```

CiNcH83, 2025-04-14:
Meter export power look sane. Have to check import power as well when the sun is down and the battery is empty?

How about this error when executing the `evcc meter` command:
```
[main  ] FATAL 2025/04/14 16:13:06 [db:7] cannot create meter 'db:7': cannot create meter type 'template': invalid key: timeout
```

I removed the `timeout` param in the IAMMETER template which I contributed which may be what the error is about? I think I tested the template (via the new external template feature) with the param inside. Somehow this meter is still in the db?

Probably have to delete evcc.db and start from scratch?

CiNcH83, 2025-04-14:
I started evcc.db from scratch. Something has not been sane bacause of my IAMMETER testing which I created a template for.

The `evcc meter` command runs successfully now. Will have to check whether the forecast adjustment works properly now.

CiNcH83, 2025-04-14:
[image]

Still far off.

```
/app # evcc meter
[main  ] INFO 2025/04/14 19:31:35 evcc 0.203.0
[main  ] INFO 2025/04/14 19:31:35 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/14 19:31:35 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2025/04/14 19:31:35 Local SKI: 4081993d045b8582135f3185d462b911ee74354b
[eebus ] ERROR 2025/04/14 19:31:35 websocket server error: listen tcp :4712: bind: address already in use
sun2000
-------
Power:  200W
Energy: 14383.2kWh

dtsu666
-------
Power:          552W
Energy:         13666.6kWh
Current L1..L3: -3.3A 8.59A -2.85A

luna2000
--------
Power:        2500W
Energy:       4319.0kWh
Soc:          73%
Capacity:     5.0kWh
Controllable: true

```

Looks like this always happens when battery is discharging.

CiNcH83, 2025-04-14:
Forcast is adjusted to 383,8 kWh for tomorrow (+844 %) and is increasing as long as battery is discharging it seems.

andig, 2025-04-15:
> Forcast is adjusted

The adjustment is just a result of the accrued actual or forecast energy. You'll have find out if the accrual is correct an if not why. It may also help to answer the questions (https://github.com/evcc-io/evcc/issues/20608#issuecomment-2802455845).

andig, 2025-04-15:
Guessing won‘t help. See https://github.com/evcc-io/evcc/issues/20608#issuecomment-2804061413

CiNcH83, 2025-04-15:
I think I’ll have to pass. I am not able to follow your comments. I have no clue how this is supposed to work and how to debug it. I don’t even know if there is anything to debug or whether this is expected behavior and the control loop just does not have enough information in the beginning and needs to settle first. Just felt like an adjustment of several hundred percent did not make a ton of sense.

Suppose the [debug log](https://github.com/user-attachments/files/19693915/evcc-20250410-220721-debug.log) (also see initial posting) did not reveal anything? The adjusted kWh seems to increase at night and becomes more and more off when the battery discharges (see previous two screenshots of the forecast and how the adjusted kWh increased). The [comment](https://github.com/evcc-io/evcc/discussions/20535#discussioncomment-12798977) made by @VolkerK62 doesn't help either?

andig, 2025-04-16:
There is only a single todo: check if the delta in accrued PV energy matches the pv meter power/energy. If it does and the result is wrong then the meter is wrong. There's nothing more I can add, it's really simple...

CiNcH83, 2025-04-17:
I am still not sure about what you want me to check even though I am trying hard. You really make me feel stupid.

What I can say is that the actual production calculated via the PV meter matches the Open-Meteo forecast pretty well. An efficiency parameter of 80% would probably work very well most of the time.

I haven't yet figured out what makes the adjusted forecast for tomorrow drift off so tremendously. I do believe that it is a combination of restarting evcc in the evening (as it loses the tracked PV production? I can't see that it is persisted in the db?) and the home battery discharging at a high rate. Not too easy and very time consuming to investigate further. Will do my best.

CiNcH83, 2025-04-23:
Putting this issue to rest now (it is closed anyway, so...)

I am pretty sure that it is caused by restarting evcc in the evening, losing all historical data about past energy production which is used for the adjusted forecast and the battery discharging. The adjusted forecast settles the next day already and gets better over time. I think that this will be fixed as soon as this data is persisted.

I turned off the adjusted forecast for now as I am restarting evcc on a daily basis. An efficiency factor of 80% turned out to be pretty accurate for the Open-Meteo forecast in my case.

andig, 2025-04-23:
As I‘ve said before: as long as the calculation is correct that‘s expected.