evcc-io/evcc #20625: FR: Zielladen mit minPV

RenatusRo, 2025-04-15:
**Is your feature request related to a problem? Please describe.**

Kontext: Dynamischer Strompreis
Taeglich ein oder zwei Autos laden, Warmwasser, elektrische Fahrraeder, Spuelmaschine

Im Winterbetrieb funktioniert das alles mit (extern automatisierten) Ladezielen/plaenen
wunderbar. Der billigste Strom ist meist Nachts.

In der jetzigen,sonnigen Uebergangszeit ist der billigste Strom ueber Mittag.
Es ist global ueber den Tag auch genuegend PV verfuegbar , um alles ausreichend 
zu bedienen.
Stehen alle loadpoints auf PV werden die niedrig priorisierten 
jedoch durch das Auto aus dem Markt gedraengt und bekommen nichts ab 
(oder nur die Stunde mit < 1.3kW).
Wenn die automatisierten Ladeplaene eingestellt werden, feuern 
die alle ueber Mittag auf volle Pulle und ziehen (teilweise) aus dem Netz.

Gaebe es nun die Moeglichkeit, um per loadpoint den Ladeplan optional 
mit minPV statt Fast zu betreiben, wuerden die minPV Ladeplaene sich
auch in die Mittagszeit draengeln aber in der Summe < PV sein.
Und das Auto wuerde dann alle RestPV schlucken und Grid auf +/- Null
regeln.
(Im Handbetrieb funktioniert das einwandfrei.)

**Describe the solution you'd like**

In der Ladeplan UX eine Option 'minPV statt Fast'. 
Erweiterung der planning API um diese Option.

**Describe alternatives you've considered**
The usual suspects: EMHASS, ... HA, Automatisierungen. Ganz ohne die geht es doch nicht.
Aber mit der vorgschlagenen Erweiterung wuerden die Sozialpartner weniger duch magische  Hintergundzaubereien ueberrascht.

**Additional context**
https://github.com/evcc-io/evcc/discussions/20619
fragt etwas Aehnliches

naltatis, 2025-04-16:
Ich verstehe das "Min+PV" hier noch nicht.
Wenn ich das richtig verstehe ist hier der Wunsch, dass es mehrere Ladepläne gibt, die sich die begrenzte Ressource "PV Überschuss" sinnvoll und geplant aufteilen. Also dass die Prognoseenergie mit im Planer berücksichtigt wird. Das kann dann zur Folge haben, dass wir (im Gegensatz zu heute) vielleicht auch geplante Slots haben in denen wir den Ladepunkt nicht die maximale Leistung ziehen lassen um das optimale Ergebnis zu erreichen.

RenatusRo, 2025-04-16:
Dass die Ladeplanung die begrenzte Ressource prognosebasiert sinnvoll und geplant aufteilt (und dann noch Circuit MaxPower/MaxCurrent beruecksichtigt) ist das getraeumte Endziel und ja in https://github.com/evcc-io/evcc/issues/14261 beschrieben. Aber das wird wohl noch ein bissel dauern.

Der Vorschlag hier intendiert, dass bei den mit minPV aktiven Laedeplaenen kein Netzbezug stattfindet (oder eben nur, wenn die Summe der minCurrents nicht mehr aus PV gedeckt werden kann).

Praktisches Beispiel:
Die eBikes kommen jeden Morgen mit leeren Batterien an. Der Ladeplan sagt: bis 17Uhr Ziel 1.3kWh, das macht derzeit  automatisch von 11 bis14:30Uhr (Viel Sonne, Strom Mittags billig).
Eines der Autos laedt  immer mit PV und saugt allen Ueberschuss. 
Die eBikes wuerden also mit PV niemals auf ihre 3.5 Stunden Ladezeit kommen. Deshalb Ladeplan, weil taeglich. Wenn dann der eBike-Ladeplan feuert , kommen die 500W aus dem Grid. Das ist  doof: ich schalte den Ladeplan aus und stelle um auf minPV.
Ergebnis: Gruener Balken
[image]

batmanvane, 2025-04-18:
Es geht doch auch jetzt schon approximativ automatisiert, um das  losfeuern zu vermeiden: Einfach mehrer Ladepläne mit unterschiedlichen SOCs. Minimalbeispiel, um bei 8 Uhr 80% zu haben, aber von 6-8 halt mit PV, statt Nachts mit maximaler Leistung: 

Ladeplan 1: bis 6 Uhr zB SOC=0.5
Ladeplan 2: bis 8 Uh zB SOC=0.8

Ladeplan 2 wird erst nach 1 ausgeführt mit Morgensonne.

Bei uns geht das dann so: Auto kommt gegen 16 mit SOC=0.2 an. Dann bekommt man noch ein wenig über PV. Rest regelt Ladeplan 1 (und 2, hab das noch nicht getestet, müsste aber so gehen).

Setzt natürlich voraus, dass man die Zeit zum Lastverteilen hat.

RenatusRo, 2025-04-22:
Ja, sowieso geht alles. Und mit ein bissel externer Logik noch viel mehr. Und ich habe auch kein nicht-loesbares Problem.
Im Wesentlichen will ich erreichen, dass die tadellos funktionierende Regelung Gridbezug vermeidet (innerhalb der regelbaren Grenzen).

Und wenn wir das eben noch etwas erweitern:
https://github.com/evcc-io/evcc/discussions/20762
haette gerne einen planbaren PV-modus.

Und, regelmaessig kommt die Frage, um die Energiemessung fuer integrateddevices taeglich  auf 0 zu setzen. Hierfuer ist dann immer die Loesung per API eben auf Off zu schalten. 

Alles extern machbar. Aber wenn doch schon ein scheduler in evcc ist: warum nicht auch hierfuer nutzen?

andig, 2025-04-29:
Es scheint als bräuchte es einen Mechanismus, mehrere Ladepläne zu koordinieren. Zusaätzliche Einflussgrössen sind dann maximale Leistung und PV Vorhersage. Aktuell haben wir dafür keinen Algorithmus. Eine Idee könnte eine zusätzliche Kostenfunktion sein. Wie die zu rechnen hätte wäre eine separate Diskussion.

closed 2025-04-29 by andig