evcc-io/evcc #20629: mg2mqtt force funktioniert nicht

Studhey, 2025-04-15:
So wie ich das verstehe und auf dem MQTT beobachen kann wird zum aufwachen ein "force" in das topic geschrieben, was auch grundsätzlich richtig ist. 

`  wakeup:
    source: go
    script: |
      "force"
    out:
      - name: wakeUp
        type: string
        config:
          source: mqtt
          topic: saic/{{ .user }}/vehicles/{{ .vin }}/refresh/mode/set
          timeout: {{ .timeout }}`

Das funktioniert aber nicht, solange in dem Topic noch ein "force" steht. Ich lösche das force und dann schreibe ich "force" was eigentlich immer zum erfolg führt. 

Im ioBroker kann ich das Topic z.B. nach 3s automatisch löschen lassen. Dann gibt es in einem anderen Topic kurz eine Meldung, dass "nichts" nicht supported ist, aber das hat weiter keine Auswirkung. 

Wäre es möglich das force nach ein paar Sekunden wieder zu entfernen?

andig, 2025-04-15:
/CC @tisoft würde eine `sequence` helfen?

tisoft, 2025-04-16:
Hmm, also bei mir funkioniert das auch wenn da schin ein force im topic drin steht. Egal ob das retained ist oder nicht. Ich nutze mosquitto als mqtt server. Vielleicht verhalten sich da verschiedene server unterschiedlich?

Studhey, 2025-04-16:
Ich habe noch ein Einstellung "Publizieren/Aktualisieren nur im Falle einer Änderung" gefunden, ich habe da mal den Haken rausgenommen. Vielleicht macht das schon den Unterschied. Mosquitto wird direct für ioBroker nicht angeboten, dann müssten ich den parallel laufen lassen. 

[image]

andig, 2025-04-20:
>Ich habe noch ein Einstellung "Publizieren/Aktualisieren nur im Falle einer Änderung" gefunden, ich habe da mal den Haken rausgenommen.

Es wäre wunderbar, wenn nicht jedes MQTT Plugin versuchen würde das Rad neu zu erfinden...

closed 2025-04-20 by andig

Studhey, 2025-04-20:
Hilft aber nicht. Ich habe gerade mal das topic mit einem leeren Wert geschrieben und gewartet, bis evcc dort force reinschreibt. Das funktioniert ohne Probleme. Das Auto war 2:10h angeschlossen und ich geh davon aus, dass evcc schon mehrmals force probiert hatte, was bis dann nicht funktionierte. Ich werde mir ein script machen was 5s nach dem setzen von force diesen wieder löscht, dann braucht das nicht in evcc passieren.

andig, 2025-04-20:
Das ist unlogisch. Der Empfänger sieht immer die neu geschriebene Nachricht. Falls das nicht funktioniert brauchts ein Ticket bei mq2mazda, nicht bei evcc.