evcc-io/evcc #20631: New Volvo Connected integration throws an error

baldrick123, 2025-04-15:
### Describe the bug

Updated to the latest version of EVCC in the hopes of using the newly published Volvo Connected integration. 
When following all steps, getting developer credentials etc and populating the values as follows in the yaml, an error occurs when trying to regenerate tokens. 

**Yaml:**

vehicles:
  - name: Volvo
    type: template
    template: volvo-connected
    title: Volvo XC40
    capacity: 67
    vccapikey: 31******
    clientId: dc******
    clientSecret: Dw*******
    accessToken: 
    refreshToken: 
    vin: Y********

**Refreshing tokens:** 

/app # evcc token Volvo
[main  ] INFO 2025/04/15 13:08:12 evcc 0.203.1
[main  ] INFO 2025/04/15 13:08:12 using config file: /etc/evcc.yaml
[main  ] FATAL 2025/04/15 13:08:12 decoding failed due to the following error(s):

'' has invalid keys: capacity, icon, title, tokens, vccapikey, vin
/app # 

### Steps to reproduce

1. Update to latest EVCC
2. Follow steps for Volvo Connected vehicle integration, entering vccapikey, client ID, and client secret
3. Run the evcc token command to try and refresh the tokens
4.  Command fails, tokens are not refreshed therefore integration fails. 

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****
sponsortoken: *****
interval: 30s # control cycle interval

meters:
- type: template
  template: homewizard 
  usage: grid  
  host: ******
  name: grid1
- type: template
  template: enphase 
  usage: pv  
  host: ******
  token: *********
  cache: 1s  
  name: pv2

site:
  title: Home
  meters:
    grid: grid1
    pv:
      - pv2

vehicles:
  - name: Volvo
    type: template
    template: volvo-connected
    title: Volvo XC40
    capacity: 67
    vccapikey: *****
    clientId: *****
    clientSecret: *****
    accessToken: 
    refreshToken: 
    vin: *****

  - name: Opel
    type: template
    template: opel
    title: Opel Corsa E
    capacity: 50 # Battery capacity in kWh (optional)
    user: *****
    accesstoken: *****
    refreshtoken: *****

loadpoints:
  - title: Driveway # display name for UI
    charger: Alfen # charger
    vehicle: Volvo # default vehicle
 
chargers:
  - name: Alfen
    type: template
    template: alfen
        
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: *****
    port: 502 # Port
```

### Log details

```text
[main  ] ERROR 2025/04/15 13:12:55 creating vehicle Volvo failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': missing token credentials
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.1

andig, 2025-04-15:
Tokens are missing in your config.

/cc @lehmanju

baldrick123, 2025-04-15:
> Tokens are missing in your config.
> 
> /cc [@lehmanju](https://github.com/lehmanju)

But instructions tell me to generate those tokens using the command that is failing. Where else can I get the token values from? Putting a dummy value results in the same error.

lehmanju, 2025-04-15:
Yeah, that token generation command is suboptimal. To generate a token you need an empty Volvo vehicle config entry. 
` ` ` 
- name: my-volvo
  type: volvo-connected
` ` ` 
Then call the token generation command with ` my-volvo`  and paste the tokens into your existing volvo config/new volvo via UI.

baldrick123, 2025-04-15:
> Yeah, that token generation command is suboptimal. To generate a token you need an empty Volvo vehicle config entry.
> 
> ```
> - name: my-volvo
>   type: volvo-connected
> ```
> 
> Then call the token generation command with ` my-volvo` and paste the tokens into your existing volvo config/new volvo via UI.

I tried that just now. It then progresses on the "evcc token my-volvo" command, asking me for client-id and client secret. Entering those gives me a URL to visit. That however loads a volvo webpage stating the following:

_We're sorry!
An error has occurred.

Invalid Session.

TrackingId:
eizlb_shN9r8bM7AZ5VGeeHKWts_

I can confirm both client ID and client secret are correct.

lehmanju, 2025-04-15:
It should work, I just used it yesterday successfully ;). If your client_id and client_secret (not vccapikey) are correct, then the URL should be correct. I do not know, however, what you used as redirect_uri, because it is missing from this request. You should set it to "http://localhost:7070/callback" on app publication. Apparently I hardcoded it and didn't update the docs: https://github.com/evcc-io/evcc/blob/23c7459cbf0c378d041f0ee2e3636ada474dd69d/vehicle/volvo/connected/identity.go#L20

baldrick123, 2025-04-15:
Ah, ok. I followed the doc indeed. Let me try to republish with the updated callback to see what happens. Will report shortly.

oskarfessel, 2025-04-15:
I just ran into this same problem - since there seems to be no way to update the publishing details, do i need to publish a new application or am i missing something?

oskarfessel, 2025-04-15:
Just before i run into some other problem, can i include conve:fuel_status and conve:battery_charge_level?  this is where my car publishes the distance to empty and battery level - these are not availible on the energy API.  This is for older Sensus connect hybrid vehicles...

baldrick123, 2025-04-15:
I had to delete the existing app and publish a new one with the updated callback URL. That resulted in the Volvo webpage loading properly. 
Now, it tells me to enter the code that is to be extracted from the webpage, linking the PSA instructions (which by coincidence I used before, as we also own an Opel). 
I am assuming the "callback" URL the volvo page opens has the code, and it's the piece after **callback?code=** and before **&state=.** 
Putting that in, resulted in a token string and a refresh string. But when I input those in the yaml and restart EVCC, I get the following error:

_[main  ] ERROR 2025/04/15 14:41:54 creating vehicle Volvo failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_client" "unable to parse Authorization header."_

lehmanju, 2025-04-15:
@oskarfessel right now, only energy API is used. Theoretically, Connected Vehicle should be possible as well, as long as the scopes do not need manual approval. That is not implemented, however.

lehmanju, 2025-04-15:
@baldrick123 yes, you copied the correct code and the fact that you got two tokens means that it worked. The error you posted is related to an invalid client_id/client_secret. Maybe you forgot to change them after you published a new app?

baldrick123, 2025-04-15:
Hmm, I am positive I copied them both to the letter. Just to rule everything out though I will delete the published API and start from scratch. Will update in the next few minutes.

baldrick123, 2025-04-15:
Well, I just recreated the API on the volvo site using a different name to be sure.
Wrote down the primary key, new Client ID and new Client Secret. 

Entered both when prompted in the **evcc token my-volvo** command. 
Tokens were generated. 
Input tokens in the yaml by copy pasting the output of evcc token my-volvo like so:

vehicles:

  - name: Volvo
    type: volvo-connected
    tokens:
      access: eyJh***1Iow
      refresh: PiU7****vNs

Restarted EVCC:

[main  ] ERROR 2025/04/15 14:54:16 creating vehicle Volvo failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_client" "unable to parse Authorization header."

lehmanju, 2025-04-15:
Well, from your config the client_id, client_secret and vccapikey are missing. These need to be present as well.

baldrick123, 2025-04-15:
Doh! I forgot to add those back in when I started out with your instructions to just use the two lines to generate the tokens. 
I just added the three values in like this:

  - name: Volvo
    type: volvo-connected
    capacity: 67
    vccapikey: *******
    clientId: dc-***********
    clientSecret: rM*************
    tokens:
      access: **********
      refresh: Pi**********

Restarted, now it throws this:

_[main  ] FATAL 2025/04/15 15:08:53 vehicle cannot create vehicle 'Volvo': cannot create vehicle type 'volvo-connected': decoding failed due to the following error(s):
'' has invalid keys: clientid, clientsecret_

Any idea why it would still complain about that?

lehmanju, 2025-04-15:
Just use the UI :) Wrong format, if you use `type: volvo-connected` then use [this](https://github.com/evcc-io/evcc/blob/23c7459cbf0c378d041f0ee2e3636ada474dd69d/templates/definition/vehicle/volvo-connected.yaml#L46), otherwise if `type: template` then you need to use `clientId` and so on. Seriously, just use the UI, it's going to be the default way anyway.

oskarfessel, 2025-04-15:
> Just use the UI :) Wrong format, if you use `type: volvo-connected` then use [...]

But how can i create a token using the ui?  I will try using the yaml for the token and then the UI for the full set of parameters (after deleting the vehicle from the yaml, of course)

lehmanju, 2025-04-15:
You can't create a token using the UI, yet. However, I believe that feature is planned for other cars as well.

lehmanju, 2025-04-15:
> @oskarfessel right now, only energy API is used. Theoretically, Connected Vehicle should be possible as well, as long as the scopes do not need manual approval. That is not implemented, however.

See here, only energy api is implemented. What vehicles it supports, I don't know.

FrankBuchholz, 2025-04-16:
evcc v0.203.01 - Volvo
Hier mein aktuelle Protokoll - leider mit Fehler bei oauth2:

1. Für die Nutzung mit EVCC benötigst du einen Volvo Account und einen Volvo Connected Car API Key.
Erstelle dazu auf der Account Seite eine neue Applikation und speichere den primären VCC API Key ab.
Veröffentliche nun deine Applikation und wähle unter "Scopes" die Berechtigungen "Connected Vehicle API - conve:vehicle_relation" und "Energy API - (alles)" aus.
Als Redirect URL muss zur Zeit `http://localhost:7070/callback` angegeben werden. 
Sobald die Applikation erstellt ist, wird sie als "Publication under Review" angezeigt. Das ist nicht weiter schlimm, es funktioniert trotzdem.
VCC API key - Primary     1ae4d***
VCC API key - Secondary   33927***
Client ID                 dc-p8***
Client Secret             uYRxl***

2. Lege eine evcc.yaml Datei an:
```
vehicles:
- name: EX30
  type: volvo-connected
``` 

3. Erstelle danach einen Token mit  `evcc token EX30`  und speichere die Tokens in der Konfiguration. 
Nach Eingabe von  client id  und  client secret  wird eine URL angezeigt, die man im Browser aufruft:
https://volvoid.eu.volvocars.com/as/authorization.oauth2?client_id=dc-p8***&code_challenge=42zoX***&code_challenge_method=S256&redirect_uri=http%3A%2F%2Flocalhost%3A7070%2Fcallback&response_type=code&scope=openid+conve%3Avehicle_relation+energy%3Arecharge_status+energy%3Abattery_charge_level+energy%3Aelectric_range+energy%3Aestimated_charging_time+energy%3Acharging_connection_status+energy%3Acharging_system_status&state=Utxh6***

4. Auf diese Seite meldet man sich mit der Volvo id an und bestätigt die nächste Seite:
Allow access for:
	Request for approval
	EVCC would like to get the following permissions:
	Vehicle permissions
	Application can see information about the battery state of charge
	Application can see the status of the charging handle connection with the car
	Application can see the status of the charging system
	Application can see the estimated time to fully charge the battery
	Application can see the estimated distance to empty based on battery capacity
	Application can see car status (incl. battery charge level, charging state and range)
	Fleet permissions
	Application can see your connected cars

5. Die nächste Seite liefert zwar einen Fehler, in der URL wird jedoch der code gezeigt, den man als nächstes eingeben muss:
http://localhost:7070/callback?code=jZBwCte***&state=Utxh***

6. Nun erhält man die erforderlichen token:
Add the following tokens to the vehicle config:
```
    type: volvo-connected
    tokens:
      access: ***ziemlich*lang***
      refresh: BLWfh***
```

Das sieht dann so aus:
```
vehicles:
- name: EX30
  type: volvo-connected 
  title: Volvo EX30  
  vccapikey: 1ae4d***
  tokens:
    access: ***ziemlich*lang***
    refresh: BLWfh***   
  vin: YV12*** 
  capacity: 69
```

7. Zum Schluss kann man die neue Konfiguration mit  `evcc vehicle`  testen.

Leider bekomme ich  nun folgende Meldung:
[main  ] ERROR 2025/04/16 17:22:15 creating vehicle EX30 failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_client" "unable to parse Authorization header."
.Soc:     vehicle not available: cannot create vehicle type 'volvo-connected': oauth2: "invalid_client" "unable to parse Authorization header."
Capacity: 69.0kWh
Features: [Offline Retryable]

baldrick123, 2025-04-16:
I think your issue is the same as I had. Take a look up higher in this thread and look at the two different options to structure the yaml. Or better do as lehmanju suggested and add your Volvo via the UI using these tokens.

FrankBuchholz, 2025-04-16:
Thank you!

Step by step...

```
vehicles:
- name: EX30
  type: volvo-connected 
  title: Volvo EX30  
  vccapikey: 1ae4d***
  credentials:
    id: dc-p8***
    secret: uYRx*** 
  tokens:
    access: ***quite*long***
    refresh: BLWfh***   
  vin: YV12*** 
  capacity: 69
```

Now `evcc vehicle` on the laptop shows the car! Yabadabadoo!!!
Soc:           88%
Capacity:      69.0kWh
Charge status: B
Range:         315km
Finish time:   2025-04-16 18:33:00 +0200 CEST

After uploading `evcc.yaml` to the NAS, and restarting the EVCC container I still need some more tuning (but tomorrow...):
```
[main ] INFO 2025/04/16 19:38:38 evcc 0.203.1
[main ] INFO 2025/04/16 19:38:38 using config file: /app/evcc.yaml
[db ] INFO 2025/04/16 19:38:38 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2025/04/16 19:38:39 listening at :7070
[main ] ERROR 2025/04/16 19:38:39 creating vehicle EX30 failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"
..
```.

lehmanju, 2025-04-16:
After you have used your configuration you cannot reuse the token elsewhere afterwards. This is because as soon as evcc starts, it refreshes the token and saves it in its database, invalidating the token from the configuration file.

yareblo, 2025-04-26:
It worked very well for a week. Then, the connection to Volvo was lost (something with "expired token"). I followed the steps above again, but this time without success:

sudo evcc --database /var/lib/evcc/evcc.db  vehicle

[main  ] INFO 2025/04/26 15:57:02 evcc 0.203.3
[main  ] INFO 2025/04/26 15:57:02 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/26 15:57:02 using sqlite database: /var/lib/evcc/evcc.db
[main  ] ERROR 2025/04/26 15:57:03 creating vehicle my_volvo failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"

my_volvo
--------
.Soc:     vehicle not available: cannot create vehicle type 'volvo-connected': oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"
Capacity: 18.8kWh
Features: [Offline Retryable]

Do I need to clean something in the database? How?

lehmanju, 2025-04-26:
Yeah, I have the same issue. Wanted to wait whether this happens for other people as well. Will investigate.

setje, 2025-04-30:
Happening here as well. 

> [lp-1 ] ERROR 2025/04/30 08:18:02 vehicle soc: Get "https://api.volvocars.com/energy/v1/vehicles/ABCABCABCABC/recharge-status": oauth2: "invalid_grant" "unknown, invalid, or expired refresh token

I guess the token just has to be renewed. But doing this every now and then manually will be a bit uncomfortable.

lehmanju, 2025-04-30:
Refresh token rotation should happen automatically, otherwise it becomes invalid after 7 days. This does seem to be the case here. https://developer.volvocars.com/apis/docs/authorisation/

lehmanju, 2025-05-01:
I checked that renewed access tokens are saved correctly to settings and the refresh token is renewed as well. If a request happens and the token is expired, it is renewed. 

My theory is that if no request happens for at least 7 days, the refresh token becomes invalid since no refresh is triggered and the integration doesn't work anymore. Evcc should not be restarted during this time since that triggers a refresh as well.

@setje @yareblo can you confirm/check whether that is the case? Does this error also occur if the vehicle is charging regularly (at least once a week)? Can it be fixed by setting the update interval on your loadpoint to "always"?

JeeDeWee, 2025-05-02:
Had exactly the same issue and attempted to refresh the token by executing the command `evcc token xc60`. I then enter the creditions as they are listed in evcc.yaml into the responses. Open the URL and cut and paste the code from the URL that does not load.
It will then show new access and refresh tokens which I copy paste into evcc.yaml.

However, this does not work. Still throws the error:
```
[main  ] ERROR 2025/05/02 19:32:24 creating vehicle xc60 failed: cannot create vehicle type 'volvo-connected': oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"
```

The way to fix this is create a complete new application in developer.volvocars.com and follow the process again.

yareblo, 2025-05-04:
On my side, the token expired again yesterday. After following the reset mechanism (generating the token again and deleting the old token from the DB), it worked.

I was charging my vehicle on a daily basis last week, so there were several requests per day to access the volvo api. 

I saved the current database state. once the token expires again (probably in 7 days), I check if it has been updated in the db.

A slightly other topic, but might be connected: I noticed that the charge state is not refreshed as expected. I need to change from "PV" to "off" and back to "PV" to refresh the charge state. When I stay on "PV", evcc still things the vehicle is charged fully (from the last charging) and does not start. After switchin to "off" and "PV", it grabs the real state and starts charging. Is this expected bahaviour, or should I open a new topic for tthis?

yareblo, 2025-05-04:
@JeeDeWee: Did you delete the old token from evcc.db as stated in https://github.com/evcc-io/evcc/issues/20631#issuecomment-2832225379? On my side, it worked without recreating a completely new application. Deleting the token in the db was sufficient for a reset.

lehmanju, 2025-05-04:
I have created a fix [commit](https://github.com/lehmanju/evcc/commit/ff24c87d0071cc3128a2269f88868ac155508003) and uploaded a corresponding [docker-image](https://github.com/users/lehmanju/packages/container/package/evcc).

I will test this locally for the next week. Just a guess, but this copies code from the Tesla integration in the hopes that it fixes the issue.

JeeDeWee, 2025-05-04:
> [@JeeDeWee](https://github.com/JeeDeWee): Did you delete the old token from evcc.db as stated in [#20631 (comment)](https://github.com/evcc-io/evcc/issues/20631#issuecomment-2832225379)? 
I have not tried. I am running my EVCC as a home-assistant add-on. Stopping evcc will automatically restart the container and lose shell access into the container so things will get weird.

setje, 2025-05-05:
> I checked that renewed access tokens are saved correctly to settings and the refresh token is renewed as well. If a request happens and the token is expired, it is renewed.
>
> My theory is that if no [...]

Sorry for the belated reply. The car is connected every day, so it also expired then. Do you need further information?

lehmanju, 2025-05-10:
> I have created a fix [commit](https://github.com/lehmanju/evcc/commit/ff24c87d0071cc3128a2269f88868ac155508003) and uploaded a corresponding [...]

It runs stable for over a week now, pretty confident this fixed it. I'll integrate it into #21003. Then cumbersome token generation is gone as well.

lehmanju, 2025-05-16:
Just a heads up: Delete your Volvo Vehicle configuration once you upgrade. Otherwise you will get an error that evcc can not start. This is because the configuration changed internally.

andig, 2025-05-16:
@lehmanju lets fix that- we just need a new template name and restore plus deprecate the old one.

lehmanju, 2025-05-16:
ok, fixed in #21267

Aretus84, 2025-05-18:
"Is it possible to get a 'for dummies' guide on how to connect the Volvo app via UI? I've tried following all the instructions I could find, but I keep encountering the same issue: 'cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': cannot get vehicles: Get "https://api.volvocars.com/connected-vehicle/v2/vehicles": missing token credentials'."

garosp, 2025-05-18:
I think I followed your instruction but without - but with the same result "https://api.volvocars.com/connected-vehicle/v2/vehicles": missing token credentials

andig, 2025-05-18:
As usual, a trace log may help. "me too" doesn't. @lehmanju do we have trace logging on the OAuth exchange?

lehmanju, 2025-05-18:
What is your RedirectURL? Did you look in for /oauth/login?id=... in your log file, copied it into your browser and completed the login?

yareblo, 2025-05-18:
I also have trouble. Log-File:

[main ] INFO 2025/05/18 16:45:21 listening at :7070
[oauth2redirect] INFO 2025/05/18 16:45:22 registering oauth provider at /oauth/login?id=oauth.dc-******
[main ] ERROR 2025/05/18 16:45:22 creating vehicle my_volvo failed: cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': cannot get vehicles: Get "https://api.volvocars.com/connected-vehicle/v2/vehicles": missing token credentials

When entering the following URL in the browser:
http://openhabian:7070/oauth/login?id=oauth.dc-******

I am directed to the Volvo-Site. I cannot log in, but I get the error:

We're sorry!
An error has occurred.
Invalid Session.

Aretus84, 2025-05-18:
Here are all the logs I have found on the topic.
[image]
[image]

lehmanju, 2025-05-18:
@Aretus84 your clientsecret is clearly wrong because its the same as clientid. reread the docs again, after the first try to add the vehicle, look in the logs for /oauth/login?id=oauth. open that in a new tab, login and then try to add your vehicle again.

nheyser, 2025-05-18:
Just for your information:
I couldn't create a new Application within the Volvo Developer Portal with an Redirect URL with "http://".
I had to use "https://" instead.

After a few restarts the vehicle is now connected, but I needed several tries.

Thank you 👍

Aretus84, 2025-05-18:
> [@Aretus84](https://github.com/Aretus84) your clientsecret is clearly wrong because its the same as clientid. reread the docs again, after the first try to add the vehicle, look in the logs for /oauth/login?id=oauth. open that in a new tab, login and then try to add your vehicle again.

That part has been fixed, but it still doesn't work. :( I have no idea what to do with this."
[oauth2redirect] INFO 2025/05/19 00:27:42 registering oauth provider at /oauth/login?id=oauth.dc-lju4crracvxxxxxxx]

Aretus84, 2025-05-18:
@lehmanju That part has been fixed, but it still doesn't work. :( I have no idea what to do with this."
[oauth2redirect] INFO 2025/05/19 00:27:42 registering oauth provider at /oauth/login?id=oauth.dc-lju4crracvxxxxxxx]

garosp, 2025-05-18:
THx for your help and your great work,

for me it seems to be working :

What I do :

first I can't register my app with the right URI at Volvo - I could only use http://localhost:7070/oauth/callback - otherwise no registration possible
I check that all data clientID, clientSecret is correct. I use redirectUri: http://localhost:7070/oauth/callback -> same as above in app and I comment out the old accessToken and refreshToken

- Then I take the part /oauth/login?id=oauth.dc-xxxxxxx from the log and submit it to my application for example http://192.168....:7070/oauth/login?id=oauth.dc-xxxxxxx.

192.168.... is the hostname/IP of my evcc and /oauth.... is the string from Log.

- Then I was prompted to lock in at Volvo and allow the access.

- The response in the browser was http://localhost:7070/....................................................... as assumed
- I take this url and change http://localhost:7070 with the right URL http://192.168...:7070/ and submit
- then I restart and get
-  [oauth2redirect] INFO 2025/05/19 01:17:50 registering oauth provider at /oauth/login?id=oauth.dc-................
- The car is connected.

My question is - what is necessary in the future when the access is expired ? Will it now be automatically refreshed ?

Best regards

Sicher versendet mit [Proton Mail](https://proton.me/mail/home).

erniemctoeff, 2025-05-20:
Still some problems here :( first of all thank you for your great work!

When publishing my Volvo app I can’t use the correct redirecturl. There is an error on the Volvo page then. When using https instead of http, the publishing works. But then when logging in at Volvo with link from the log I can admit but then get an error as the url I used as redirect url (https) doesnt exist. When changing it by myself in the Adress by deleting the s, i get to the Evcc page. When i click on Konfiguration nothing appears on the page. When going to the correct evcc link and adding the Volvo ex30 i still get the error: cannot create vehicle type ‚template‘: cannot create vehicle type ‚Volvo-connected‘: cannot get vehicles: get ‚https://api.volvocars.com/connected-vehicle/v2/vehicles‘:missing token credentials

Can anyone help?

andig, 2025-05-20:
> There is an error on the Volvo page then.

What error?

> When using https instead of http, the publishing works.

Is your redirect url served on https? What's the url?

>  But then when logging in at Volvo with link from the log I can admit but then get an error as the url I used as redirect url (https) doesnt exist.

If you don't serve it on https you can't use https!

lehmanju, 2025-05-20:
i think you did it correctly, mostly. i think you have to do the login process in a new tab and leave the tab with your add vehicle window open. after the redirect is successful, go to the original tab and retry validation.

drojasvela, 2025-05-20:
I'm also trying to config my Volvo car from the UI **using EVCC integration in HA**. I get the `registering oauth provider at /oauth/login?id=oauth.dc-xxxxxxxxx` error in the log, but I don't know how to construct the URL I need to complete the authentication. I've tried `http://homeassistant.local:8123/49686a9f_evcc/ingress/oauth/login?id=oauth.dc-xxxxxxxx` and similar variations but doesn't work.

Aretus84, 2025-05-20:
I finally got this working. Basically, I had to install RealVNC on the Raspberry Pi and use a remote connection to access the Raspberry Pi’s own browser to complete the /oauth/login?id=oauth.dc process. So, I had to do it on the device where EVCC is installed.

garosp, 2025-05-21:
I don't use HA, but I think that http://homeassistant.local:8123 is your local HA host/address. I think you must root the "http://homeassistant.local:8123/49686a9f_....." direct to your evcc-target. Something like "http://evcc.local:7070/..... or direct the ip-address of your evcc-instance. Please try to access your evcc directly from your browser with "http://evcc.local:7070" or something similiar perhaps with your IP-Adress not over HA -> then you know the way to access evcc -> and that is the way you have to route the oauth statements.

Sicher versendet mit [Proton Mail](https://proton.me/mail/home).

erniemctoeff, 2025-05-21:
> What error

‚Something Went wrong, please try again later‘

It works, when i type ‚https://…‘ instead of ‚http://…‘

> Is your redirect url served on https? What's the url?

No it’s not. URL is http://evcc:7070/#/

andig, 2025-05-22:
> No it’s not. URL is http://evcc:7070/#/

@erniemctoeff that is not the valid redirect or of evcc!

erniemctoeff, 2025-05-22:
> > No it’s not. URL is http://evcc:7070/#/
> 
> [@erniemctoeff](https://github.com/erniemctoeff) that is not the valid redirect or of evcc!

I just copied the adress from the browser when using evcc. How has the valid redirect to look like?

m03pchen, 2025-05-23:
Then Volvo seems to have changed something again. Currently you can't publish an app if you don't have ‘https://’ in the link. This is currently my problem too.

I don't have a https:// instance of EVCC on Homeassistant, so my address is always http://myipadresse:7070/oauth/callback

But Volvo does not accept this address - error message ‘Something went wrong, please try again later’.
The error message only goes away if you enter an httpS url.

However, logging in via (for example) http://myipadresse:7070/oauth/login?id=oauth.dc-oaivik8xxxxxxxxx62z3i then no longer works -> Volvo message: ‘An error has occurred. Invalid Session.’

Translated with DeepL.com (free version)

lehmanju, 2025-05-23:
Docs:
> redirectUri: # Redirect URI of your EVCC instance, format: `http://evcc.local:7070/oauth/callback`. Must match the redirect URI set in your Volvo Developer App.

If only https works, then you must use https. However, the two URIs must still match.

[27 more replies]

PR #21003 (2025-05-15): Volvo Connected: add oauth using generic plugin
Fixes #20986 
Fixes #20631 

Ein generisches OAuth2 plugin, dass die verschiedenen "Identities" in den `vehicle` APIs vereinheitlichen soll.
closed 2025-06-28 by andig