evcc-io/evcc #20638: Zendure panic: runtime error:

bsmout, 2025-04-15:
### Describe the bug

Hello,

I added 2 Zendure Hypers 2000 to EVCC and since then I'm receiving the following errors in the log:
panic: runtime error: invalid memory address or nil pointer dereference

[signal SIGSEGV: segmentation violation code=0x1 addr=0x98 pc=0x1282810]

goroutine 89 [running]:

github.com/evcc-io/evcc/meter/zendure.(*Connection).handler(0x4000f48540, {0x4000f9c500, 0x127})

	github.com/evcc-io/evcc/meter/zendure/connection.go:73 +0xa0

github.com/evcc-io/evcc/plugin/mqtt.(*Client).listen.func1({0x0?, 0x4000da9e80?}, {0x31f3c60?, 0x40011200f0?})

	github.com/evcc-io/evcc/plugin/mqtt/client.go:238 +0x228

github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2()

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:218 +0x7f4

created by github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch in goroutine 83

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:173 +0x200

### Steps to reproduce

1. Add 2 Hyper 2000 systems to EVCC
2.
3.
...

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

sponsortoken: *****
telemetry: true
plant: *****
site:
  title: ***masked*** # display name for UI
  meters:
    grid: fluvius # grid meter
meters:
  - name: fluvius
    type: template
    template: homewizard
    usage: grid
    host: 10.44.1.50

Version: `0.203.1`
```

### Log details

```text
panic: runtime error: invalid memory address or nil pointer dereference

[signal SIGSEGV: segmentation violation code=0x1 addr=0x98 pc=0x1282810]

goroutine 115 [running]:

github.com/evcc-io/evcc/meter/zendure.(*Connection).handler(0x4000b59ac0, {0x4000c937c0, 0x127})

	github.com/evcc-io/evcc/meter/zendure/connection.go:73 +0xa0

github.com/evcc-io/evcc/plugin/mqtt.(*Client).listen.func1({0x0?, 0x0?}, {0x31f3c60?, 0x40006e4140?})

	github.com/evcc-io/evcc/plugin/mqtt/client.go:238 +0x228

github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2()

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:218 +0x7f4

created by github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch in goroutine 93

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:173 +0x200
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.1

StefanSchoof, 2025-04-15:
See https://github.com/evcc-io/evcc/issues/20632

andig, 2025-04-16:
> have verified that the issue is reproducible with the latest nightly build

Are you sure?

closed 2025-04-16 by andig

bsmout, 2025-04-16:
Yes but I was using the docker nightly image which wasn't yet updated.

andig, 2025-04-16:
It is updated just the same…