evcc-io/evcc #20642: Verbindung schlägt fehlt mit evcc iOS App

userwithoutpassword, 2025-04-15:
### Describe the bug

Wenn man mit der iOS App seinen evcc server sucht klappt das. Es wird evcc.local vorgeschlagen und man kann sich auch damit verbinden und die Verbindung speichern.

Wenn man die App schließt und wieder öffnet direkt funktioniert das auch weiterhin.

Wenn man die App aber einige Zeit (vielleicht halbe Stunde) nicht genutzt hat schlägt die Verbindung fehlt und es erscheint dauerhaft die Einblendung "Search for connection..." 

Ein erneutes hinzufügen klappt zwar, aber auch dann hat man nach einiger Zeit wieder dasselbe Problem.

Warscheinlich irgendein Problem mit mDNS???

Finde ich aber merkwürdig weil grundsätzlich findet er den Server ja und kann sich auch verbinden...

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
siehe oben
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

detlefheese, 2025-04-15:
Kannst du nicht deine feste IP einstellen?

userwithoutpassword, 2025-04-15:
habe ich noch nocht ausprobiert. Aber wieso nicht mDNS nutzen wenns da ist. Finde es halt nur merkwürdig dass die Verbindung beim ersten mal klappt, später dann aber nicht mehr.