evcc-io/evcc #20647: Heidelberg Energy Control - keine Modbus VErbindung nach Update auf 0.203.0

twingo667, 2025-04-16:
### Describe the bug

Setup: Vecchio auf Raspberry Pi mit zwei Heidelberg Energy Control per USB/RS485 Adapter. Einwandfreie Funktion bis zum Modbus Update (0.203). Nach dem Update findet er die Wallboxen nicht mehr. Downgrade auf 0.201 führt wieder zu sofortiger Funktionalität.

### Steps to reproduce

1. Evcc V. 0.201. läuft einwandfrei
2. Update auf evcc V 0.203. Wallboxen werden nicht mehr gefunden. Fehlermeldung, Abbruch beim Starten des Dienstes
3. Downgrade auf V 0.201: Wallboxen funktionieren wieder einwandfrei
...

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203

VolkerK62, 2025-04-16:
nimm mal die neueste version

andig, 2025-04-16:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

twingo667, 2025-04-16:
> nimm mal die neueste version

yepp, mit 0.203.1 geht es wieder. Irgendwie mochte er die Comset Einstellung nicht in der 0.203.0. Besten Dank für den Hinweis.