evcc-io/evcc #20648: Polestar API Auth Changes

fblix, 2025-04-16:
### Describe the bug

The current implementation of evcc throws another authentication error when trying to connect with the Polestar API.

It seems Polestar have changed their auth flow again.

### Steps to reproduce

1. Start evcc with a Polestar plugged in.
2. See the error message pop up.

### Configuration details

```yaml
vehicles:
  - name: polestar
    type: template
    template: polestar
    title: Polestar
    user: ***
    password: ***
    capacity: 82
    icon: car
    phases: 3
```

### Log details

```text
[main ] INFO 2025/04/16 09:40:25 evcc 0.203.1
[main ] INFO 2025/04/16 09:40:25 using config file: /config/evcc.yaml  
[main ] INFO 2025/04/16 09:40:26 listening at :7070  
[main ] ERROR 2025/04/16 09:40:26 creating vehicle polestar failed: cannot create vehicle type 'template': cannot create vehicle type 'polestar': login failed: could not find resume path
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.203.1

andig, 2025-04-16:
Somebody will need to find out _what_ has changed...

loebse, 2025-04-16:
Not again... 😒

supergroover, 2025-04-16:
I'm not a coder, in the integration [Polestar EV](https://github.com/pypolestar/pypolestar) there was a change yesterday that made it work again. Maybe someone can look into this.

see [commit](https://github.com/pypolestar/pypolestar/commit/144a5142925d7ef0718e1ae5ce6d48a711d12e76)

loebse, 2025-04-16:
Changed the regex that we are using to find the resume path again:

https://github.com/evcc-io/evcc/pull/20656

closed 2025-04-16 by andig