evcc-io/evcc #20653: panic: runtime error: invalid memory address or nil pointer dereference

scharmerd, 2025-04-16:
### Describe the bug

EVCC läuft in einer Docker Umgebung auf einer Synology NAS
Seit dem Upgrade auf 0.203.0 stürtzt der EVCC Container jedes Mal ab.

### Steps to reproduce

1. Update 0.202.1 -> >= 0.203.0 oder 0.203.1 Fehler
2. Update 0.202.1 startet problemlos
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxxxxxxxxxxxxxxxx

interval: 30s # control cycle interval

sponsortoken: xxxxxxxxxxxxxxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.1.xxx 
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 12

site:
  title: Daheim
  
influx:
  url: http://192.168.1.XXX:8087
  database: evcc # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: xxxxxxxxxxxxxxxxx
  org: xxxxxxxxxxxxxxxxx

Webgui Konfig (aufgenommen in 0.202.1)
Ladepunkt / Auto (Tesla): https://imgur.com/EEKlLP7
Netzanschluss / PV / Batterie https://imgur.com/mHdLphM
Andere Integrationen: https://imgur.com/MBXYG9E
```

### Log details

```text
[main  ] INFO 2025/04/16 12:03:53 evcc 0.203.1

[main  ] INFO 2025/04/16 12:03:53 using config file: /etc/evcc.yaml

[db    ] INFO 2025/04/16 12:03:53 using sqlite database: /root/.evcc/evcc.db

[mqtt  ] INFO 2025/04/16 12:03:54 connecting evcc-917480754 at tcp://192.168.1.xxx:1883

[mqtt  ] DEBUG 2025/04/16 12:03:54 tcp://192.168.1.xxx:1883 connected

[main  ] INFO 2025/04/16 12:03:54 listening at :7070

[proxy-502] DEBUG 2025/04/16 12:03:54 modbus proxy for 192.168.1.xxx:1502 listening at :502

[zendure] INFO 2025/04/16 12:03:54 connecting  at tcp://mqtt-eu.zen-iot.com:1883

[zendure] DEBUG 2025/04/16 12:03:54 tcp://mqtt-eu.zen-iot.com:1883 connected

[zendure] DEBUG 2025/04/16 12:03:54 tcp://mqtt-eu.zen-iot.com:1883 subscribe w3D6Yj03/#

[lp-1  ] DEBUG 2025/04/16 12:03:56 charge total import: 5864.814kWh

panic: runtime error: invalid memory address or nil pointer dereference

[signal SIGSEGV: segmentation violation code=0x1 addr=0x98 pc=0x17b0841]

goroutine 98 [running]:

github.com/evcc-io/evcc/meter/zendure.(*Connection).handler(0xc000dfae00, {0xc000ffa500, 0x127})

	github.com/evcc-io/evcc/meter/zendure/connection.go:73 +0xa1

github.com/evcc-io/evcc/plugin/mqtt.(*Client).listen.func1({0x0?, 0xc00057a0b8?}, {0x36bb3a0?, 0xc000a8f720?})

	github.com/evcc-io/evcc/plugin/mqtt/client.go:238 +0x20e

github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2()

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:218 +0x8f0

created by github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch in goroutine 47

	github.com/eclipse/paho.mqtt.golang@v1.5.0/router.go:173 +0x211
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-16:
> I have verified that the issue is reproducible with the latest nightly build

Really?

closed 2025-04-16 by andig

scharmerd, 2025-04-16:
> > I have verified that the issue is reproducible with the latest nightly build
> 
> Really? This checkbox is there for a reason ;)

Hi, sorry, bei Portainer hat er scheinbar ein Problem beim abholen des Builds gehabt.
Sprich nach der Umstellung startete der Container schon gar nicht.

Hast recht, mit dem Nightly geht es.

Und sorry für das doppelte eröffnen.