evcc-io/evcc #20654: Daikin homehub integration

tbiest, 2025-04-16:
**Is your feature request related to a problem? Please describe.**
Daikin has developed a “homehub” to make a modbus integration with their heat pumps (for instance the daikin altherma 3). 

**Describe the solution you'd like**
Integration with modbus is a more reliable and advanced way to integrate instead of the SG contacts. This way you can modulate the consumed power and get information back from the device. The modbus protocol and prameters are well described by Daikin and other solutions like Loxone are already capable.

**Describe alternatives you've considered**
Using SG relay contacts

**Additional context**

andig, 2025-04-16:
If you have the register description you can build a tempalte. Note this is nothing the core team will be working on.

hybe2022, 2025-04-17:
Das Modbus IF/die Register sind im folgenden Installationshandbuch im Kapitel 9 beschrieben:

https://www.daikin.de/content/dam/document-library/Installer-reference-guide/heat/accessories/ekrhh/EKRHH_Installer%20reference%20guide_4PDE744838-1D_German.pdf

[EKRHH_Installer reference guide_4PDE744838-1D_German.pdf](https://github.com/user-attachments/files/19791690/EKRHH_Installer.reference.guide_4PDE744838-1D_German.pdf)

tbiest, 2025-04-25:
Looking at the existing templates I think we need a new kind of combination between setting the maximum available solar power (like Kermi) and the SG ready template (like Stiebel Eltron). I'm not familiar with the Go programming language, maybe some community member can think with us on this?

The Daikin home hub also has a Modbus address to lookup the 3-way valve setting (register 37). With that we should be able to  decide automatically if we show the boiler temperature or the room temperature (in case the unit is performing room buffering).

In the case of Stiebel Eltron, a different mapping and channel setting is used for activating the smart grid mode: multiple channels must be set. In case of Daikin, only channel 56 is needed. 0 = normal, 1 = forced off (might be useful for dimming § 14a EnWG), 2 = Recommended on (this is what we need for our PV), 3 = Forced on.

andig, 2025-05-04:
@tbiest we can add this as optional enhancement to the SG Ready charger. Before I'd start working on this I'd like to see there's a working SG ready implementation for this WP though. The power recommendation is optional, so we can do this afterwards. To you want to try a template?

tbiest, 2025-05-04:
Hello @andig agreed, let's first set up the basic template and validate it against a real Daikin homehub. Below is the idea I have so far, based on existing templates and the Daikin manual. You indicate that the power limitation is optional. I included it in the idea below, I'm not sure it is supported already?

`template: daikin-homehub
products:
  - brand: Daikin
    description:
      generic: Daikin-HomeHub (SG Ready)
group: heating
# requirements:
#   evcc: ["sponsorship"]
params:
  - name: modbus
    choice: ["tcpip"]
  - name: tempsource
    type: choice
    choice: ["warmwater", "buffer"]
    description:
      de: "Temperaturquelle"
      en: "Temperature source"
render: |
  type: sgready
  getmode:
    source: map
    values:
      0: 2 # Normal
      1: 1 # Forced off
      2: 3 # SG recommended on
      3: 3 # SG forced on
    get:
      source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 55
        type: input
        encoding: int16
  setmode:
    source: switch
    switch:
    - case: 2 # normal
      set:
        - source: const
          value: 0
          set:
            source: modbus
            {{- include "modbus" . | indent 10 }}
            register:
              address: 55
              type: writeholding
              encoding: int16
    - case: 3 # boost
      set:
        - source: const
          value: 2
          set:
            source: modbus
            {{- include "modbus" . | indent 10 }}
            register:
              address: 55
              type: writeholding
              encoding: int16
    - case: 1 # dimm
      set:
        - source: const
          value: 1
          set:
            source: modbus
            {{- include "modbus" . | indent 10 }}
            register:
              address: 55
              type: writeholding
              encoding: int16
  {{- if .tempsource }}
  temp:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: {{ if eq .tempsource "warmwater" -}} 42 {{ else }} 49 {{- end }}
      type: input
      encoding: int16
    scale: 0.1
  {{- end }}
  power:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: 50
      type: input
      encoding: int16
    scale: 0.1
  setmaxpower:
    timeout: 30s
    set:    
      source: modbus
      {{- include "modbus" . | indent 4 }}
      register:
        address: 56 # PV surplus power
        type: writeholding
        decode: int16
      scale: 10.0
`

andig, 2025-05-04:
Can you please open a PR and make sure to test the tempalte?

tbiest, 2025-05-04:
@andig yes I will do that. It will take some time before I have time available but you will see the PR after testing.

Raudi1, 2025-05-16:
My Home Hub should be operational in a couple of hours. If you want to test something, just let me know.
I have a multi spilit system with 5 indoor units.

tbiest, 2025-05-16:
Hi @Raudi1 that would be awesome. I don't know if you are able to setup the template in a test environment at your site? Otherwise, maybe we can do some custom tests in your evcc.yaml. F.i. test the connection and see if we can get the consumed power by configuring the device as a modbus power meter?

Raudi1, 2025-05-16:
It's all installed and I can control it with the custom Home Assistant integration. If you want me to test something, I can do it for you next week. I only have Windows or the homeassistant OS server.

tbiest, 2025-05-16:
[daikin_evcc.txt](https://github.com/user-attachments/files/20258418/daikin_evcc.txt)

Hi @Raudi1 

In the attachment you will find the additions for your evcc.yaml. I extracted the code snippets from the draft template to test.

You need to replace the ip-address with your own instead of the 192.168.2.11 address that my mockup server uses. Also, I assume your unit ID is 1 and you use the 502 port (otherwise replace this also).

You need to add the home hub to the loadpoints and to the chargers section.

If all is well, it will show up nicely and display the power and temperature of the room. I'm curious if it works and if the scaling of the measurements are correct.

andig, 2025-05-17:
See https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md#device-templates for testing templates.

tbiest, 2025-05-19:
#21298 merged pull request

Raudi1, 2025-05-19:
I took a look at the merged template. In its current state it doesn't make any sense for my case. I only have air conditioning units and no boiler (use case 4 in the Daikin manual). So there's no water temperature for me etc. and I would need to use different registers.
What is possible in my case is to limit the power draw directly via holding register 1002. So it would be possible to set that value to the pv surplus value.

Raudi1, 2025-05-19:
The template would need different registers depending on use cases, or at least a disclaimer about it being limited to a certain use case.

tbiest, 2025-05-19:
Yes I agree. By the way, are you sure that you have the article "EKRHH" from Daikin? The modbus registers on that device are harmonized between the supported devices (Altherma3 H HT, 3 H MT, 3 R, 3 R MT and 3 M). For Airco units, you probably use another hub type then? It would be good to let evcc users know what units are compatible and how you can see what hardware is installed in your home/business.

Raudi1, 2025-05-19:
Yes, I have the same device. My use case is even mentioned in the same manual you linked above. The Home Assistant integration also has a checkbox for air to air systems and just gives me the ability to set the SG mode and to limit the power. So everything as it should be according to the manual. There it also only lists two registers for my use case as you can see below:

[image]

The manual also states for use case 3:

> In this mode the Daikin HomeHub is used to
control the Daikin Altherma heat pump from a home automation or home energy
management (HEM) system through a local interface.

And for use case 4

> This use case provides Smart Grid (SG) and Demand Control functionality for air-to-
air heat pumps. This makes it possible for energy utilities to communicate with air-
to-air heat pumps.

I only have the home hub because of §14a compliance in Germany. But I'm thinking about removing it, since I can control the AC units directly.

tbiest, 2025-05-19:
Ahh yes I see, for use case 4 you have a different set of modbus registers, you are correct. So then, we need to explicitly state that this first version of the template is limited to Daikin Altherma3 with use case 3. And make a second one for use case 4 if there is demand for it (since it differs quite a bit from use case 3).

andig, 2025-05-19:
> Yes the power limit would be great, also for my Smart Grid case. In the discussions I read that this feature will be added for the SG Ready templates in the future. Register 57 will be the one used for power limit during Smart Grid. That will differ from your case.

@tbiest I think we could quite easily add this to `type: sgready` but that would really only make sense when boosting from evcc perspective. We would especially NOT set a (different) power limit when disabling the "charger", i.e. switching back to normal. Would this still be helpful?

tbiest, 2025-05-19:
Hello @andig. If I understand you correctly we would only want to set the power limit for the Boost Mode, not interfering in any way in the normal power limit. Right?

In case of the Daikin homehub this is possible because there is a different modbus register for the SG power limit (register 57) and the General power limit. So when the device is not in boost, it will disregard evcc and use the general power limit (register 58).

So I think this will work fine for Daikin. It might not be possible for every manufacturer.

hybe2022, 2025-05-19:
Hello,
general power limit might be useful for 14a load management when signal comes from grid service provide

andig, 2025-05-19:
> Hello [@andig](https://github.com/andig). If I understand you correctly we would only want to set the power limit for the Boost Mode, not interfering in any way in the normal power limit. Right?
>
> In [...]

@tbiest could you check if https://github.com/evcc-io/evcc/pull/21319 works for your purpose?

Raudi1, 2025-05-19:
> So I think this will work fine for Daikin. It might not be possible for every manufacturer.

It will not work for use case 4. There we only have one power limit as far as I understand it.

tbiest, 2025-05-19:
@Raudi1 could it be that in your case register 1002 is intended for smart grid power limit? It says "Power limit for Demand Control". It might be something you can ask your installer.

Raudi1, 2025-05-19:
I could ask my installers. But it would be the same as asking my dog. They have no idea what Modbus is or how it works. They just know Daikin told them I need the Home Hub. They didn't even know why. And since our electricity provider is actually installing Rundsteuerempfänger for §14a compliance, they couldn't find a solution. For basically all other electricity providers it's enough to have some kind of theoretical control capability and a cable from the meter to the device. So when my electrician asked Daikin how he can connect the Rundsteuerempfänger, they didn't even understand the issue at first and didn't have any solution ready to go.
The electrician, the installer and several people at Daikin talked for a while without any solution other than waiting for a future product. When they told me, I found a solution in a couple of seconds. It's not straightforward, but works: I already have 2 Go-e chargers connected to the Rundsteuerempfänger. When I configure one to change to single phase when the contact is activated, I can see it in the status in Home Assistant. So HA just tells the Home Hub via  Modbus to throttle to 4,2kW when HA registers the status change of the Go-e.
As for how the power limit register works, it's probably easiest to read the manual and just test it.

deThommy, 2025-08-26:
Just read through the thread and got a question regarding the integeration of my heatpump: I have a Daikin Altherma 3 with Home Hub installed, configured the HomeHub and Shelly Pro 3 EM correctly running for a year now. 

Now the heatpump is shown as "heating" (only 28-74 W) in the evcc UI, which is the stand-by consumption, circulation pump,.. which is also measured by the Shelly.

Is this correct like this or is it possible to integrate some kind of "threshold" until which the heatpump is shown as "waiting" or "idling"? (e.g. < 80W = "off")

I have an [ESPAltherma](https://raomin.github.io/ESPAltherma/) installed (aswell to grab some more data directly frome the heatpump itself), so I'd know if the current state of the Heatpump is actually heating.

tbiest, 2025-08-26:
Hi @deThommy I use the Altherma 3 also. The heat pump will use a small number of watts in idle mode indeed. In my case around 18W idle and around 65w when SG-ready mode is active but idle. The wattage via Modbus will show 0.

In my case, Evcc will only show "heating" when the SG mode is active. When it is consuming power in a non-SG mode it will show 'standby'. Maybe you can check if this is also working that way in your situation? When you configure the Daikin heat pump from the UI, you can choose to not use the shelly as a meter. In that case evcc will use the measurement from the modbus (it will show 0 in idle and it will roughly show the consumption in a rounded number).

deThommy, 2025-08-26:
@tbiest thanks for your feedback! I just tried it, deleted the Shelly, it showed 0 W and when the Altherma produced hot water it showed 1,3 kw. so exactly as you described. 

however the shelly is way more accurate than the "wild guesses" from the Altherma itself. For me it seems that the Altherma "tries" to calculate every value in the most positive way so the result is a better COP than it actually is when you calculate it.

e.g. 
- lower energy consumption "measured" (more a guess than a measurement) by the Altherma vs. Shelly (even when I take the 70W for the pump out of the equasion). by my account it shows roughly 8% less energy in the first year
- more generated heat than what I get if i calculate it 

That leads to a way better COP (values in the Altherma Unit) vs. everything calculated. 

So since there really seems to be no downside, i guess i'll stick to the 70w constantly being measured. seems like it has no real impact on the signal towards the Altherma to start producing heat... or am I missing something?

tbiest, 2025-08-26:
Yes there is basically no problem. If you want accurate data then using the Shelly is a better option. I used homewizard the 1st year to check the power consumption but now I trust the machine and got rid of the kWh meter.

One thing to note: when the altherma is producing heat/cooling or hot water by itself and not via the SG command, evcc should still say “standby” while the power consumption is showing that the heat pump is active (showing f.i. 2kw). I included a screenshot as 

[image]

example for you.

tbiest, 2025-08-26:
Hi @deThommy the slider is not designed to work like this. The Daikin template is using the SG Ready principle: when PV surplus is detected, it will set the Altherma in the "Recommended on" mode. The Altherma will decide for itself if it will start heating/cooling the room (if you have buffering enabled on the altherma) or if it is going to produce hot water.

The slider is a standard function of evcc. When you choose to show the temperature (either room or hot water tank) the slider appears. I think it is somewhere on the roadmap to disable that and show the temperature of either the tank or the room depending on what the Altherma is working on. If you get annoyed by the slider then the best option for now is to not select room temperature or tank temperature when setting up in the UI: the slider will disappear.

When in PV mode, evcc will also send the maximum power allowed to the Altherma so that it will not consume more than the available PV surplus.

If you have trouble setting up the correct settings in the Altherma I'm happy to help.

tbiest, 2026-03-22:
I think Daikin advises not to install in direct sunlight. My installer moved the temperature sensor to the shadow side of the unit, that helps for the most part.

The way the SG function works has quite some limitations. That is why I switched to Niko Home Control beginning of this year. It works with a direct integration on the indoor unit that is developed by Dakin. No hardware required and it works quite nice: when excess power is available, it can set the thermostat to a higher (or lower) value and it uses the 'boost' method to boost hot water. This will result in nice, smooth and long runs of the heat pump and works also if you are actively heating or cooling your home. The integration is managed professionally so if there is something wrong you will have support.