evcc-io/evcc #20657: Docs update for custom tariffs plugin

chrisrock1010, 2025-04-16:
### Describe the bug

Hi, 
i have implemented the custom tariffs plugin but was struggling with the example in the docs:
https://docs.evcc.io/en/docs/tariffs#plugin

Therefore i think you need to update the sample output / jq to replace `price` with `value` otherwise its showing `0.000` for price / cost. 
In addition for me it was required to devide the price by `100` to get the correct cost per EUR/kWh
Not working: `25.0`
Working: `0.25`

```
 jq: '[.timestamps, .prices] | transpose | map({ "start": (.[0] | strptime("%Y-%m-%dT%H:%M:%S.%f%z") | strftime("%Y-%m-%dT%H:%M:%SZ")), "end": (.[0] | strptime("%Y-%m-%dT%H:%M:%S.%f%z") | mktime + 3600 | strftime("%Y-%m-%dT%H:%M:%SZ")), "price": .[1] }) | tostring'
```

Sample in the docs

```
[
  { "start": "2025-01-01T00:00:00Z", "end": "2025-01-01T01:00:00Z", "price": 25.0 },
  { "start": "2025-01-01T01:00:00Z", "end": "2025-01-01T02:00:00Z", "price": 30.0 },
]
```

Working sample for me: 
```
[
  { "start": "2025-04-16T14:00:00Z", "end": "2025-04-16T15:00:00Z", "value": 0.30 },
  { "start": "2025-04-16T15:00:00Z", "end": "2025-04-16T16:00:00Z", "value": 0.22 }
]

### Steps to reproduce

Use the sample of the current docs: https://docs.evcc.io/en/docs/tariffs#plugin

### Configuration details

```yaml
jq: '[.timestamps, .prices] | transpose | map({ "start": (.[0] | strptime("%Y-%m-%dT%H:%M:%S.%f%z") | strftime("%Y-%m-%dT%H:%M:%SZ")), "end": (.[0] | strptime("%Y-%m-%dT%H:%M:%S.%f%z") | mktime + 3600 | strftime("%Y-%m-%dT%H:%M:%SZ")), "price": .[1] }) | tostring'
```

### Log details

```text
grid:
From                   To                     Price/Cost (EUR/kWh)
2025-04-16 15:00:00    2025-04-17 04:00:00    0.000
2025-04-17 04:00:00    2025-04-17 06:00:00    0.000
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-04-16:
Happy to take a PR in the docs repo!

naltatis, 2025-04-16:
Doku PR https://github.com/evcc-io/docs/pull/777 already existed but was waiting for release. It's merged now.

closed 2025-04-16 by naltatis

andig, 2025-04-16:
Danke Micha 🙏