evcc-io/evcc #20670: item is not available in REST call - Solarwatt manager Flex

hkempf77, 2025-04-17:
### Describe the bug

Hi,

in my EVCC configuration I get all values instead of the SoC of my Batteryflex managed by Solarwatt Energymanager flex

in Line 71 of the  solarwatt-flex.yaml file the query for the SoC looks like this

      [(.[] | select(.name | test(".*_batteryChannel_state_of_charge$")).state | parseFirstNumber)] | add // 0

but when I check my setup via REST - the values are listed here:

 {
    "link": "http://energymanager.local/rest/items/solarwattBattery_batteryflex_BatteryFlex_0BB169_batteryChannelGroup_batteryStateOfCharge",
    "state": "0.0",
    "stateDescription": {
      "pattern": "%.0f",
      "readOnly": true,
      "options": []
    },
    "editable": false,
    "type": "Number",
    "name": "solarwattBattery_batteryflex_BatteryFlex_0BB169_batteryChannelGroup_batteryStateOfCharge",
    "label": "State Of Charge",
    "tags": [],
    "groupNames": []

means the line should be changed to 

[(.[] | select(.name | test(".*_batteryChannelGroup_batteryStateOfCharge$")).state | parseFirstNumber)] | add // 0

I run my EVCC as a LXC Container - build up with the helper-script.
EVCC-Version is v0.203.1
Solarwatt Manager Flex version is 10.25.13.0

### Steps to reproduce

1..
2.
3.
...

### Configuration details

```yaml
Solarwatt-Manger flex added with the experimental GUI - not included in evcc.yaml.
```

### Log details

```text
Battery 2 is the Solarwatt-Battery:
[lp-1 ] DEBUG 2025/04/17 10:26:38 charge power: 0W
[lp-1 ] DEBUG 2025/04/17 10:26:38 charge currents: [0 0 0]A
[site ] DEBUG 2025/04/17 10:26:38 grid power: 19W
[site ] DEBUG 2025/04/17 10:26:38 battery 2 power: -756W
[site ] DEBUG 2025/04/17 10:26:38 pv 2 power: 1146W
[site ] DEBUG 2025/04/17 10:26:39 battery 1 power: -0W
[site ] DEBUG 2025/04/17 10:26:39 pv 1 power: 313W
[site ] DEBUG 2025/04/17 10:26:39 pv power: 1459W
[site ] DEBUG 2025/04/17 10:26:39 grid powers: [-5 -23 -17]W
[site ] DEBUG 2025/04/17 10:26:39 grid currents: [-3.23 -2.29 -2.28]A
[site ] DEBUG 2025/04/17 10:26:39 battery 1 soc: 10%
[site ] DEBUG 2025/04/17 10:26:40 battery 2 soc: 0%
[site ] DEBUG 2025/04/17 10:26:40 battery power: -756W
[site ] DEBUG 2025/04/17 10:26:40 battery soc: 0%
[site ] DEBUG 2025/04/17 10:26:40 battery has priority at soc 0% (< 80%)
[site ] DEBUG 2025/04/17 10:26:40 site power: 119W
[lp-1 ] DEBUG 2025/04/17 10:26:40 charge total import: 7800.656kWh
[lp-1 ] DEBUG 2025/04/17 10:26:40 charger status: B
[lp-1 ] DEBUG 2025/04/17 10:26:40 next soc poll remaining time: 9m30s
[site ] DEBUG 2025/04/17 10:26:53 ----
```

### What type of operating system or environment does evcc run on?

other

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.203.1

andig, 2025-04-17:
Feel free to open a PR

hkempf77, 2025-04-17:
before I do a pull - i would like to validate this in my environment - therefore I´ve checked my filesystem on my container to find the template-files. nothing found. the only thing I find related to evcc is the config yaml in /etc and the database in /var/lib/evcc.

where can find the template files?

andig, 2025-04-17:
You'll need to compile from source for that

VolkerK62, 2025-04-17:
I guess, you can try this https://github.com/evcc-io/evcc/pull/19495

closed 2025-04-29 by github-actions[bot]