evcc-io/evcc #20674: EVCC startet Ladevorgang ständig neu.

Dici71, 2025-04-17:
### Describe the bug

Hallo,

Ich habe eine GO-E Wallbox mit EVCC verbunden. Das lief auch seit einigen Monaten stabil. 

Nun habe ich das Problem, dass der Ladevorgang ständig bei startet, obwohl kein PV Überschuss vorhanden ist. Auch wenn ich über EVCC Laden ausschalte startet der Ladevorgang trotzdem. 

EVCC Version : 0.203.1 ( der gleiche Fehler trat auch schon bei der vorherigen Version auf )

Ich habe mal testweise WLAN auf der GO-E Box, sowie die API V2 ausgeschaltet. dann tritt der Fehler nicht mehr auf.

### Steps to reproduce

1. GO-E einzeln, Kein Problem
2. GO-E API V2 und " ausstecken simulieren" aktiviert
3. GO-E kmit EVCC verbunden
4. Ladevorgang beginnt immer wieder von vorne

### Configuration details

```yaml
network:
 #  schema: http
  # host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 90a967ebccf3c9e65eb83b9c5f8658166de19480022891bc14c

interval: 5s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoiRGljaTcxIiwiZXhwIjoxODA5ODMDAsInNyYyI6ImdoIn0.lOyAbo>

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false 

vehicles:
- type: template
  template: audi
  title: Audi Q4
  user: andrea.cleven@me.com
  password: kerzos-4qicCi-nibfys
  vin: WAUZZZFZ2PP011667
  capacity: 55
  name: ev1

chargers:
  - name: GO-E
    type: template
    template: go-e-v3
    host: 192.168.1.40 # IP-Adresse oder Hostname

#chargers:
#  - name: GO-E
#    type: template
#    template: ocpp
#    stationid: 250057 # Station ID des Ladepunktes. Nur erforderlich wenn mehrere OCPP-Ladestationen eingerichtet sind um diese korrekt zuzuweisen. Eine einzeln>
   # connector: 1 # Anschlussnummer bei Ladestationen mit mehreren Anschlüssen. Die Zählung beginnt bei 1 für den ersten Anschluss. (optional)
   # remotestart: # Diese Option nur aktivieren wenn keinerlei Möglichkeit besteht Transaktionen seitens des Ladepunktes zu initiieren! Das ist nur der Fall wenn>
   # idtag: evcc # Diese Option ist nur in Ausnahmefällen erforderlich wenn der Ladepunkt für die Annahme externer Transaktionen einen spezifischen Token erforde>
   # connecttimeout: 5m # optional
   # meterinterval: 10s # optional
   # metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage # optional

[... 638 lines cut ...]
[lp-1  ] DEBUG 2025/04/17 13:11:40 set charge current limit: 10A
[site  ] DEBUG 2025/04/17 13:11:40 ----
[lp-1  ] DEBUG 2025/04/17 13:11:40 charge power: 6994W
[lp-1  ] DEBUG 2025/04/17 13:11:40 charge currents: [15.2 15.1 0.076]A
[site  ] DEBUG 2025/04/17 13:11:41 battery 1 power: 1096W
[site  ] DEBUG 2025/04/17 13:11:41 battery 1 soc: 18%
[site  ] DEBUG 2025/04/17 13:11:42 grid power: 1773W
[site  ] DEBUG 2025/04/17 13:11:42 pv 1 power: 2218W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

Lako-Unbekannt, 2025-04-17:
Hi,
es schein so als würde deine Wallbox nicht richtig auf evcc hören:
"charger logic error: current mismatch (got 10.5A measured, expected 6A) - make sure your interval is at least 30s"
dein Intervall ist ja auf 5 Sekunden gesetzt, könntest du testen ob das hochsetzten auf 30 das Prolblem löst?

TomF79, 2025-04-17:
**Bitte anonymisiere deine Config, es können alle deine Passwörter und Token gesehen werden !!!**

Dici71, 2025-04-17:
Ich probiere es mal aus

Dici71, 2025-04-17:
Das bringt leider nicht, das Fahrzeug startet ständig den Ladevorgang neu

Lako-Unbekannt, 2025-04-17:
Kannst du dann noch ein log schicken?

Lako-Unbekannt, 2025-04-17:
In dem Bild fehlt evcc auserdem Informationen, wo der Strom herkommt. Ist das immer so?

Dici71, 2025-04-17:
Klar. Hier nochmal ein Log

[lp-1  ] DEBUG 2025/04/17 14:16:41 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:16:41 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:16:41 battery 1 power: -1729W
[site  ] DEBUG 2025/04/17 14:16:41 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:16:43 grid power: 63W
[site  ] DEBUG 2025/04/17 14:16:50 pv 1 power: 2336W
[site  ] ERROR 2025/04/17 14:16:53 grid currents: Get "https://192.168.1.248/production.json?details=1": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site  ] ERROR 2025/04/17 14:17:00 pv 1 energy: Get "https://192.168.1.248/production.json": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[site  ] DEBUG 2025/04/17 14:17:00 battery has priority at soc 27% (< 90%)
[site  ] DEBUG 2025/04/17 14:17:00 site power: 163W
[lp-1  ] DEBUG 2025/04/17 14:17:00 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:17:00 charger status: A
[site  ] DEBUG 2025/04/17 14:17:11 ----
[lp-1  ] DEBUG 2025/04/17 14:17:11 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:17:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:17:11 battery 1 power: -1636W
[site  ] DEBUG 2025/04/17 14:17:11 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:17:12 grid power: 52W
[site  ] DEBUG 2025/04/17 14:17:15 pv 1 power: 2278W
[site  ] DEBUG 2025/04/17 14:17:21 grid currents: [5.33 -3.38 -2.78]A
[site  ] DEBUG 2025/04/17 14:17:21 battery has priority at soc 27% (< 90%)
[site  ] DEBUG 2025/04/17 14:17:21 site power: 152W
[lp-1  ] DEBUG 2025/04/17 14:17:21 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:17:21 charger status: A
[site  ] DEBUG 2025/04/17 14:17:41 ----
[lp-1  ] DEBUG 2025/04/17 14:17:41 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:17:41 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:17:41 battery 1 power: -1557W
[site  ] DEBUG 2025/04/17 14:17:41 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:17:41 grid power: 35W
[site  ] DEBUG 2025/04/17 14:17:42 pv 1 power: 2230W
[site  ] DEBUG 2025/04/17 14:17:48 grid currents: [5.13 -3.34 -2.73]A
[site  ] DEBUG 2025/04/17 14:17:48 battery has priority at soc 27% (< 90%)
[site  ] DEBUG 2025/04/17 14:17:48 site power: 135W
[lp-1  ] DEBUG 2025/04/17 14:17:48 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:17:48 charger status: A
[site  ] DEBUG 2025/04/17 14:18:11 ----
[lp-1  ] DEBUG 2025/04/17 14:18:11 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:18:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:18:11 battery 1 power: -1534W
[site  ] DEBUG 2025/04/17 14:18:11 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:18:12 grid power: 16W
[site  ] DEBUG 2025/04/17 14:18:15 pv 1 power: 2206W
[site  ] DEBUG 2025/04/17 14:18:30 grid currents: [4.99 -3.32 -2.7]A
[site  ] DEBUG 2025/04/17 14:18:30 battery has priority at soc 27% (< 90%)
[site  ] DEBUG 2025/04/17 14:18:30 site power: 116W
[lp-1  ] DEBUG 2025/04/17 14:18:30 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:18:30 charger status: A
[site  ] DEBUG 2025/04/17 14:18:41 ----
[lp-1  ] DEBUG 2025/04/17 14:18:41 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:18:41 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:18:41 battery 1 power: -1502W
[site  ] DEBUG 2025/04/17 14:18:41 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:18:41 pv 1 power: 2197W
[site  ] DEBUG 2025/04/17 14:18:43 grid power: -3W
[site  ] DEBUG 2025/04/17 14:18:51 grid currents: [4.9 -3.3 -2.69]A
[site  ] DEBUG 2025/04/17 14:18:51 battery has priority at soc 27% (< 90%)
[site  ] DEBUG 2025/04/17 14:18:51 site power: 97W
[lp-1  ] DEBUG 2025/04/17 14:18:51 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:18:51 charger status: A
[site  ] DEBUG 2025/04/17 14:19:11 ----
[lp-1  ] DEBUG 2025/04/17 14:19:11 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:19:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:19:11 battery 1 power: -1479W
[site  ] DEBUG 2025/04/17 14:19:11 battery 1 soc: 27%
[site  ] DEBUG 2025/04/17 14:19:11 pv 1 power: 2189W
[site  ] DEBUG 2025/04/17 14:19:13 grid power: 6W
[site  ] DEBUG 2025/04/17 14:19:19 grid currents: [4.89 -3.3 -2.66]A
[... 816 lines cut ...]
[lp-1  ] DEBUG 2025/04/17 14:53:22 charger status: A
[site  ] DEBUG 2025/04/17 14:53:41 ----
[lp-1  ] DEBUG 2025/04/17 14:53:41 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:53:41 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:53:41 battery 1 power: -1252W
[site  ] DEBUG 2025/04/17 14:53:41 battery 1 soc: 34%
[site  ] DEBUG 2025/04/17 14:53:42 pv 1 power: 2011W
[site  ] DEBUG 2025/04/17 14:53:44 grid power: -50W
[site  ] DEBUG 2025/04/17 14:53:49 grid currents: [4.27 -3.13 -2.46]A
[site  ] DEBUG 2025/04/17 14:53:49 battery has priority at soc 34% (< 90%)
[site  ] DEBUG 2025/04/17 14:53:49 site power: 50W
[lp-1  ] DEBUG 2025/04/17 14:53:49 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:53:49 charger status: A
[site  ] DEBUG 2025/04/17 14:54:11 ----
[lp-1  ] DEBUG 2025/04/17 14:54:11 charge power: 0W
[lp-1  ] DEBUG 2025/04/17 14:54:11 charge currents: [0 0 0]A
[site  ] DEBUG 2025/04/17 14:54:11 battery 1 power: -1297W
[site  ] DEBUG 2025/04/17 14:54:11 battery 1 soc: 34%
[site  ] DEBUG 2025/04/17 14:54:12 pv 1 power: 2033W
[site  ] DEBUG 2025/04/17 14:54:13 grid power: -20W
[site  ] DEBUG 2025/04/17 14:54:17 grid currents: [4.41 -3.14 -2.48]A
[site  ] DEBUG 2025/04/17 14:54:17 battery has priority at soc 34% (< 90%)
[site  ] DEBUG 2025/04/17 14:54:17 site power: 80W
[lp-1  ] DEBUG 2025/04/17 14:54:17 charge total import: 1175.893kWh
[lp-1  ] DEBUG 2025/04/17 14:54:17 charger status: A

Dici71, 2025-04-17:
> In dem Bild fehlt evcc auserdem Informationen, wo der Strom herkommt. Ist das immer so?

Normalerweise nicht.

SolarPower2024, 2025-04-17:
Kann es sein, dass dein Charger die Verbindung zum WLAN verliert? Hast du mal im Lastmanagement (des go-e chargers) den Fallback auf 0A gestellt?

Btw, dein grid meter / pv meter zeigt im log auch einen Fehler an.

Dici71, 2025-04-17:
> Kann es sein, dass dein Charger die Verbindung zum WLAN verliert? Hast du mal im Lastmanagement (des go-e chargers) den Fallback auf 0A gestellt?
> 
> Btw, dein grid meter / pv meter zeigt im log auch einen Fehler an.

Die WLAN Verbindung habe ich selber gestoppt, da die Preise derzeit sonst zu hoch sind und der Wagen nicht laden soll.
Den Fallback habe ich schonmal auf 0A gestellt. Das hat Leider nicht gebracht.

Die Fehlermeldung vom PV Meter ist normal. Zwischendurch kriegt EVCC nicht alle Daten. Bei der nächsten Abfrage ist dann aber wieder alles normal.

SolarPower2024, 2025-04-17:
Naja, als normal würde ich das jetzt nicht bezeichnen, wenn dauernd eine Verbindung im Netzwerk verloren geht. ;-)

Aber irgendetwas triggert wohl den Start der go-e. Ich denke aber nicht, dass dies evcc ist. Im log ist hier auch nichts zu erkennen, dass ein Ladesignal gesendet wird.

Hast du in der go-e Box irgendetwas aktiv? Ladezeitfenster, PV laden, daily trip, Energietarif oder etwas?

Und was für einen Status zeigt die Wallbox denn an? Es sollte wenn du sie in evcc auf "aus" hast in der go-e der status "aufladen nicht erlaubt stehen"

Dici71, 2025-04-17:
[image]

[image]

[image]

Dici71, 2025-04-17:
[image]

[image]

Dici71, 2025-04-17:
Also ich habe nochmal nachgeschaut.

 Ladezeitfenster - Nein

PV laden - aktiv ( war aber immer so )

daily trip - Nein

Energietarif - Nein

Ich kann es mir auch nicht erklären. Es ist allerdings so, dass das Laden aufhört, sobald ich das Standard WLAN abschalte, bzw. die API V2

Chris45671, 2025-04-17:
Stell doch bitte Mal die Wallbox auf Eco. Vielleicht geht es dann. So ist sie bei mir eingestellt und ich habe dieses Problem nicht.

andig, 2025-04-18:
> make sure your interval is at least 30s

Du willst erstmal die Fehlermeldung korrigieren…