evcc-io/evcc #20684: Modbus TCP communication with charger [Delta AC MAX Smart] broken with version 0.203.x

madmat17, 2025-04-18:
### Describe the bug

When updating from version 0.202.0 to 0.203.0 or 0.203.1 the modbus TCP communication to the charger is no longer working (sometimes on restart the error message is being thrown that the charger cannot be created, sometimes the creation is working but succeeding communication is not).

Downgrading to version 0.202.0 is solving the issue.

When creation of the charger is working the communication can be established as some time at least (after some unsuccessful communication attempts).

**REMARK related to the configuration posted with this issue:**
The configuration is showing 'homeassistant.lan' as modbus host. This is due to the fact that my Home Assisstant instance is also a modbus proxy.
The described error also persists with the modbus proxy being shut down and evcc pointed directly to the charger.
evcc is running on a Linux server, not as HomeAssistant add-on, nor as any other sort of docker instance.

### Steps to reproduce

1. upgrade evcc from any version to 0.203.0 or 0.203.1
2. wait evcc to start, or restart evcc manually

### Configuration details

```yaml
root@evcc:/etc# evcc dump --cfg --database /var/lib/evcc/evcc.db
[main  ] INFO 2025/04/18 09:08:10 evcc 0.203.0
[main  ] INFO 2025/04/18 09:08:10 using config file: /etc/evcc.yaml
[db    ] INFO 2025/04/18 09:08:10 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/04/18 09:08:10 connecting evcc-1199125773 at tcp://homeassistant.lan:1883
[mqtt  ] DEBUG 2025/04/18 09:08:10 tcp://homeassistant.lan:1883 connected

Fehlermeldung:

    charger [wallbox5] cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'delta': failsafe timeout enabled: read tcp 192.168.5.111:50260->192.168.5.100:503: i/o timeout

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-sun2000
  id: 1
  host: homeassistant.lan
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 15s
  storageunit: 1
  name: grid1
- type: template
  template: huawei-sun2000
  id: 1
  host: homeassistant.lan
[... 325 lines cut ...]
[delta ] TRACE 2025/04/18 09:13:34 modbus: recv 00 63 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2025/04/18 09:13:34 charge power: 0W
[delta ] TRACE 2025/04/18 09:13:34 modbus: send 00 64 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/04/18 09:13:35 modbus: recv 00 64 00 00 00 05 01 04 02 00 00
[lp-1 ] DEBUG 2025/04/18 09:13:35 charger status: A
[delta ] TRACE 2025/04/18 09:13:35 modbus: send 00 65 00 00 00 06 01 03 06 40 00 02
[delta ] TRACE 2025/04/18 09:13:36 modbus: close connection: read tcp 192.168.5.111:43318->192.168.5.100:503: i/o timeout
[lp-1 ] ERROR 2025/04/18 09:13:36 charger enabled: read tcp 192.168.5.111:43318->192.168.5.100:503: i/o timeout
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.230.0 and 0.230.1

Juljoo, 2025-04-18:
Kann ich bestätigen.

Christalias, 2025-04-18:
same problem here. downgrade to 202.1 "fixed" it

andig, 2025-04-18:
In 203, we've changed the modbus TCP timeout from 10s to 1s. This doesn't seem to be enough. Could someone try to add this in delta.go to see what a sufficient timeout would be?

```go
conn.Timeout(3 * time.Second)
```

Christalias, 2025-04-18:
i´m a novice but i can try to do it when you tell me how to get to this file. i run evcc as an addon on homeassistant (i´m to stupid to talk via console to evcc because i cant find it ^^)

StefanSchoof, 2025-04-18:
You can not just change this inside the ha addon. You need to compile evcc. https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md

madmat17, 2025-04-18:
I can run a test on Monday most likely (incl. a code change and compilation).
@andig will you guys create a branch, or should I fork?

andig, 2025-04-18:
Just give it a try, we'll work the rest out ;)

JanVanr, 2025-04-18:
Any chance this could affect odd charging/1P-3P switching on my Keba 3P? I noticed today that my phase switching relay was toggling a lot and the charging was starting and stopping every couple minutes.
Disabling auto phase switching makes the charging run smooth on  1 phase only.
I'm not seeing any 'close connection' messages in the modbus logging.
I'm running 0.203.1 on Home Assistant.

Started an old 202.1 on my Mac and that was working as expected and was able to run on 3 phases.

andig, 2025-04-18:
If you don‘t have an error then you‘re not affected…

Kishi85, 2025-04-19:
I'm seeing this as well with my setup. Standalone evcc that fails to connect to my SolarEdge System via Modbus TCP. Works fine with 0.202.x but cannot connect with 0.203.x logging an i/o timeout upon trying to connect. 

You've already stated that you lowered the timeout from 10 to 1 second but for a Modbus connection to some Industrial/IoT hardware stack this is might not be enough and the old value was actually pretty decent. On SolarEdge it can take up to 15 seconds (but usually less than 10) according to the Solaredge Docs for certain setting write operations before the whole TCP stack will start to respond again (which can easily happen using evcc if you use modbus proxy with some other app/device and evcc in parallel). I've also seen such timeouts happen when my device does it's internal maintenance operation twice a day. 

Easiest option if you want to continue to enable lower timeouts would be to make this a (potentially globally) configurable option but 1 second is certainly calling it close when working with real world hardware that is out today and I'm honestly not sure if 3 seconds will cover all currently possible cases reliably.

For now i'll stay on 0.202.x and wait for a fix. Thanks in advance.

andig, 2025-04-19:
The only thing relevant at this time is to actually try if longer timeout solves the peoblem.

Kishi85, 2025-04-19:
> The only thing relevant at this time is to actually try if longer timeout solves the peoblem.

I agree. Hopefully this is enough to fix the issue. Also a configurable option (even if it is just in the config file) would be preferable to a hard coded value.

r1st0, 2025-04-19:
#20247 maybe they are related
I compiled with ```conn.Timeout(3 * time.Second)```
and i get no modbus timeouts anymore

closed 2025-04-20 by andig