evcc-io/evcc #20686: Export to a evcc.yaml file from the experimental UI configuration

asdfsafd, 2025-04-18:
**Is your feature request related to a problem? Please describe.**
It was a quick start setup using the new UI feature. But now I want to enhance the evcc functionalities without having to wait for the UI feature to be implemented.

**Describe the solution you'd like**
In the experimental UI a button "Export to yaml" exists for a yaml download to start.

**Describe alternatives you've considered**
I do not want to invest time to start fresh with a yaml configuration when I have working system through the experimental UI.

**Additional context**
n/a

SolarPower2024, 2025-04-18:
You can mix UI and yaml configuration.

Just keep the UI as is and the missing components you state in the evcc.yaml.
No, need to export the UI settings to yaml.

asdfsafd, 2025-04-18:
Works for me, thank you for your reply.

closed 2025-04-18 by andig