evcc-io/evcc #20702: WebSocket-Fehler nach Update auf 0.203 – http.ResponseWriter does not implement http.Hijacker (Reverse Proxy + Auth)

Basti77, 2025-04-18:
### Describe the bug

Nach dem Update auf Version 0.203 treten bei meiner Reverse-Proxy-Absicherung (nginx + Basic Auth) massive Probleme auf.
Vor der WAF im Intranet wird im evcc-Log mehrfach der Fehler http.ResponseWriter does not implement http.Hijacker geworfen.
Hinter der WAF ist gar kein Zugriff mehr auf die Anwendung möglich.
Bis einschließlich Version 0.202 lief diese Konfiguration ohne Probleme.

### Steps to reproduce

.Nginx-Server mit Reverse Proxy und Basic Auth einrichten (siehe Konfiguration unten).

WebSocket- und Frontend-Verbindungen über nginx leiten.

Update auf evcc Version 0.203 durchführen.

Fehler erscheinen sofort im Log, WebSocket-Verbindungen schlagen fehl.

### Configuration details

```yaml
Configuration details:
nginx Reverse Proxy Konfiguration:

nginx
Kopieren
Bearbeiten
server {
    listen 7071;
    server_name _;

    # WebSocket ohne Passwortschutz
    location /ws {
        allow 192.168.0.0/16;
        deny all;

        proxy_pass http://localhost:7070/ws;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
    }

    # Frontend mit Basic Auth
    location / {
        auth_basic "evcc Zugang";
        auth_basic_user_file /etc/nginx/.htpasswd-evcc;

        proxy_pass http://localhost:7070/;
        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
    }
}
evcc.yaml: (keine sensiblen Daten enthalten)

yaml
Kopieren
Bearbeiten
network:
  port: 7070
  basepath: ""
  auth:
    users:
      - user: nachbar
        password: geheim
(Wenn auth aktiv ist, ändert sich nichts am Verhalten.)
```

### Log details

```text
Fehlermeldungen im evcc-Log direkt nach dem Start:

pgsql
Kopieren
Bearbeiten
[server] ERROR 2025/04/18 20:46:54 failed to accept WebSocket connection: http.ResponseWriter does not implement http.Hijacker
[server] ERROR 2025/04/18 20:46:56 failed to accept WebSocket connection: http.ResponseWriter does not implement http.Hijacker
[server] ERROR 2025/04/18 20:46:59 failed to accept WebSocket connection: http.ResponseWriter does not implement http.Hijacker
[server] ERROR 2025/04/18 20:47:02 failed to accept WebSocket connection: http.ResponseWriter does not implement http.Hijacker
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-18:
Mir fällt nix ein wo ich suchen könnte. Dir bleibt nur den schuldigen Commit zu identifizieren, z.B. über die Nightlies oder selbst bauen.

andig, 2025-04-18:
@naltatis kann es sein, dass das an deinem Request Logging liegt? Auch wenn ich nicht sehe, warum?

naltatis, 2025-04-18:
@andig ja, das scheint das Problem zu sein. Hab das Logging jetzt vereinfacht. Kein Request-Wrapping mehr. Damit sollte das Problem behoben sein.

closed 2025-04-19 by andig