evcc-io/evcc #20718: falsche Energy Werte bei Shelly EM (Gen1) und Shelly PlugS (Gen1)

VolkerK62, 2025-04-19:
### Describe the bug

Beim Shelly EM (Gen1) (als Charger an meiner Wärmepumpe) werden beide Energiewerte (total und total_returned) miteinander verrechnet.

 Beim Shelly PlugS (Gen1) ist mir nicht klar, wo der Energiewert herkommt. `total` ist es nicht.

Fehler tritt erst mit dem letzten Nightly auf.
Liegt wohl an #20570 

### Steps to reproduce

.

### Configuration details

```yaml
Shelly EM (config UI)
db:10 {Type:template Title: Icon: Product:} map[channel:0 heating:true host:192.168.178.21 icon:heatpump integrateddevice:true standbypower:10 template:shelly]

Shelly PlugS (yaml)
- name: extW #Waschen
  type: template
  template: shelly-1pm
  usage: charge
  host: 192.168.178.72
```

### Log details

```text
Shelly EM
[shelly] TRACE 2025/04/19 05:37:06 GET http://192.168.178.21/shelly
[shelly] TRACE 2025/04/19 05:37:06 {"type":"SHEM","mac":"C45BBE77DE81","auth":false,"fw":"20230913-114150/v1.14.0-gcb84623","discoverable":true,"longid":1,"num_outputs":1,"num_meters":0,"num_emeters":2,"report_period":1}
[shelly] TRACE 2025/04/19 05:37:06 GET http://192.168.178.21/status
[shelly] TRACE 2025/04/19 05:37:06 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.21","rssi":-63},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"05:37","unixtime":1745033824,"serial":17295,"has_update":false,"mac":"C45BBE77DE81","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}],"emeters":[{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":232.65,"is_valid":true,"total":1268868.6,"total_returned":1390938.0},{"power":-40.63,"reactive":-22.93,"pf":0.87,"voltage":232.65,"is_valid":true,"total":1299646.8,"total_returned":1402502.7}],"update":{"status":"idle","has_update":false,"new_version":"20230913-114150/v1.14.0-gcb84623","old_version":"20230913-114150/v1.14.0-gcb84623","beta_version":"20231107-164916/v1.14.1-rc1-g0617c15"},"ram_total":51064,"ram_free":34644,"fs_size":233681,"fs_free":156373,"uptime":2167167}
[shelly] TRACE 2025/04/19 05:37:06 GET http://192.168.178.21/relay/0
[shelly] TRACE 2025/04/19 05:37:06 {"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}
Power:         0W
Energy:        -122.1kWh
Charge status: B
Enabled:       true
Features:      [IntegratedDevice Heating]

Shelly PlugS
[shelly] TRACE 2025/04/19 05:54:58 GET http://192.168.178.72/shelly
[shelly] TRACE 2025/04/19 05:54:58 {"type":"SHPLG2-1","mac":"C45BBEE308F1","auth":false,"fw":"20230913-113610/v1.14.0-gcb84623","discoverable":true,"num_outputs":1,"num_meters":1}
[shelly] TRACE 2025/04/19 05:54:58 GET http://192.168.178.72/status
[shelly] TRACE 2025/04/19 05:54:58 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.72","rssi":-47},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":true},"time":"05:54","unixtime":1745034898,"serial":30127,"has_update":false,"mac":"C45BBEE308F1","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":true,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"source":"input"}],"meters":[{"power":0.00,"overpower":0.00,"is_valid":true,"timestamp":1745042098,"counters":[0.000, 0.000, 0.000],"total":1310750}],"update":{"status":"idle","has_update":false,"new_version":"20230913-113610/v1.14.0-gcb84623","old_version":"20230913-113610/v1.14.0-gcb84623","beta_version":"20231107-164401/v1.14.1-rc1-g0617c15"},"ram_total":52096,"ram_free":39912,"fs_size":233681,"fs_free":165409,"uptime":6112636}
Power:  0W
Energy: 21.8kWh
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1 (8fb06f4c)

andig, 2025-04-19:
/cc @thierolm

thierolm, 2025-04-19:
@VolkerK62 danke für die Info, hab' den Fehler gefunden denke ich gefunden, liegt an https://github.com/evcc-io/evcc/blob/2319e0d7ae0ea3f12f76f4f7e03c92dfdc464c7e/meter/shelly/gen1.go#L131-L132

fix PR folgt.

thierolm, 2025-04-19:
@VolkerK62 ich glaube die Werte waren wenn in der alten Version falsch. Dein Plug S liefert einen Total Wert von 1310750.

[image]

Laut API1 Doku sind das Watt MINUTEN!
Deswegen 1310750 / 60 = 21.845,83333333333 Wh
Dass dann geteilt durch 1000 sind 21.8 kWh ... yay :-)

[image]

thierolm, 2025-04-19:
Link zur API1 Doku: https://shelly-api-docs.shelly.cloud/gen1/#shelly-plug-plugs-status

thierolm, 2025-04-19:
Mit https://github.com/evcc-io/evcc/pull/20570 will ich Struktur in die historisch gewachsene nicht nur allein von mir umgesetzte Shelly Logik bringen. Bitte helft bei der Fehlersuche und ggf. anderen Usern.

thierolm, 2025-04-19:
@VolkerK62 was zeigt denn die Shelly App als Energie Wert für den PlugS Shelly?

VolkerK62, 2025-04-19:
ah, ok. An Wm hab ich nicht gedacht und an die APP auch nicht.
Vergiss das Issue bzgl PlugS.
evcc und App liefern den selber Wert. Sorry.

Bleibt noch die Verrechnung beim EM

thierolm, 2025-04-19:
OK, welchen Channel hast du konfigiuriert und welchen der beiden Werte ist dann der Richtige?

[image]

Btw: das ist genau die Ecke, die im Moment noch nicht ganz klar ist, wie wir sinnvoll entscheiden können, welchen Wert Total Energy zurück liefern soll. Da bin ich mit @premultiply in KOntakt.

VolkerK62, 2025-04-19:
Ja, hab den Kommentar im PR gelesen.
Als Charger muss ich Channel 0 nutzen. 
Richtig, in Bezug auf den alten Wert, ist `total: 1268868.6`

thierolm, 2025-04-19:
OK, dann mache ich einen fix PR für https://github.com/evcc-io/evcc/blob/2319e0d7ae0ea3f12f76f4f7e03c92dfdc464c7e/meter/shelly/gen1.go#L117-L118

Ist auch bei der Gen2 API so, dass wir nur Total zurückgeben. Mal schauen, was es für Usecases für den Ret Wert gibt.

VolkerK62, 2025-04-19:
Danke für den schnellen Fix

thierolm, 2025-04-19:
@VolkerK62 kurze Frage: Wo kommt denn der `Total_Returned` Wert von 1390938.0 für deine Wärmepumpe her? Liefert die auch Strom :-) oder hattest du den EM vorher andres eingesetzt?

VolkerK62, 2025-04-20:
Der EM hat anfangs nur meine Batterie gemessen. 
Dann kam die Sache mit SGReady und deshalb ist die Batterie jetzt auf Channel1.

thierolm, 2025-04-20:
ok, d.h. aber auch, hättest du bei der Umstellung die Energie Werte für Kanal 1 zurück gesetzt, hättest die aktuelle Version auch den richtigen Wert angezeigt ... dann wäre `total_returned` für Kanal 1 gleich 0.

thierolm, 2025-04-20:
Gibt es in der Shelly App die Möglichkeit die Energiewerte einzeln zurückzusetzen, oder zu überschreiben?

VolkerK62, 2025-04-20:
einzeln nicht. Ich habe jetzt alle Energiewerte zurückgesetzt.
und ja, hast recht, dann würde der Summenwert stimmen.

andig, 2025-04-20:
….also alles korrekt oder muss ich den PR reverten?

thierolm, 2025-04-20:
Ich würde dann auch erst mal nichts ändern und PR #20724 nicht mergen (lassen) wollen.

andig, 2025-04-20:
Bin nicht sicher ob ich es verstanden habe. Wir wollten ja egtl. kein Verhalten ändern. Jetzt doch?

VolkerK62, 2025-04-20:
>Ist auch bei der Gen2 API so, dass wir nur Total zurückgeben.

Ja, das Thema Energy ist noch pending, aber wo ist der Unterschied zwischen Gen1 und Gen2

thierolm, 2025-04-20:
Es gab ursprünglich ja zwei Shelly Implementierungen:
`energymeter.go` und `switch.go`.
Diese sind jetzt zusammengeführt.
energymeter.go hatte offensichtlich nur Total berücksichtigt, deswegen gab es keine Probleme, wenn man zusätzlich auch in Total_Returned Werte hatte.
Das ist durch die Konsolidierung der Logik nicht mehr so. Bei der Nutzungsänderung des EM Shellies die Energywerte zurücksetzen.

thierolm, 2025-04-20:
> wo ist der Unterschied zwischen Gen1 und Gen2

Aktuell werden in der Gen1 API Logik die Total_Returned Werte verrechnet. Das geschieht aufgrund des aktuellen Diskussionsstandes mit @premultily zur Gen2 API nicht.
Es ja einige Anfragen Usage abhängig auch die Returned Werte zu berücksichtigen. Die genauen Szenarien sind aber noch nicht ausgearbeitet.

Wie wäre der gewollte TotalEnergy Wert denn, wenn du deinen EM einphasig als Batterie-Meter einsetzen würdest. Welcher der beiden Werte oder Welche Logik würde für TotalEnergy erwartet?

premultiply, 2025-04-20:
Es wird nur der Energiezähler in die positive Leistungsflussrichtung erwartet.

thierolm, 2025-04-20:
Ok, das spricht dann für den Merge von PR https://github.com/evcc-io/evcc/pull/20724, dann sind wir auch in Sync mit der Gen2 Logik, richtig?

VolkerK62, 2025-04-20:
>@premultiply meinte man darf nicht verrechnen. Aber die TotalEnergy Methode gibt nur einen Energie Wert zurück.

sehe ich auch so. Verrechnen ist mMn nicht richtig.
Wenn, dann sollte es dafür einen zweiten Energie Wert geben. Der Wunsch/die Frage kommt gelegentlich in den Diskussionen.
Ist aber dann ein "globales" Thema.

thierolm, 2025-04-20:
@andig :
Kurze Zusammenfassung:

1. Feeback von @premultiply :
Es wird nur der Energiezähler in die positive Leistungsflussrichtung erwartet (setzt die korrekte Installation des Shelly voraus).
2. Das spricht dann für den Merge von PR https://github.com/evcc-io/evcc/pull/20724, dann sind wir auch in Sync mit der Gen2 Logik.
3. Der gelegentliche Wunsch/die Frage in den Shelly Diskussionen nach dem Returned Energiewert ist aber dann ein "globales" Thema und sollte wenn, in einem neuen PR behandelt werden.

premultiply, 2025-04-20:
Genau. Exportenergiezähler ist wenn dann ein eigenes Thema.
Bisher kennt evcc den überhaupt nicht und brauch ihn nirgends.

PR #20724 (2025-04-20): Shelly: Gen1 API - remove offsetting energies - keep Total only
Fixes #20718

evcc expects only the energy meter in the positive power flow direction.
closed 2025-04-20 by andig