evcc-io/evcc #20720: LG ESS: Two same devices are not supported?

retsifp, 2025-04-19:
### Describe the bug

Hi, I have two LG ESS Home 8, one with battery and one without.

I added both to evcc, unfortunately it fails to start with
`meter [pv4] cannot create meter 'pv4': cannot create meter type 'template': cannot create meter type 'lgess8': uri mismatch: https://192.168.2.92 vs https://192.168.2.49`

As soon as I remove 'pv4', it works like a charm. Would be great if the values of both could be fetched! :)

### Steps to reproduce

1. Add two LG ESS Home 8 to evcc
2. Doesn't work

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <id>

interval: 30s # control cycle interval

meters:
- type: template
  template: lg-ess-home-8-10 
  usage: grid  
  host: 192.168.2.92  
  password: <pw1>  
  name: grid1
- type: template
  template: lg-ess-home-8-10 
  usage: pv  
  host: 192.168.2.92  
  password: <pw1> 
  name: pv2
- type: template
  template: lg-ess-home-8-10 
  usage: battery  
  host: 192.168.2.92  
  password: <pw1> 
  name: battery3
- type: template
  template: lg-ess-home-8-10 
  usage: pv  
  host: 192.168.2.49  
  password: <pw2> 
  name: pv4

site:
  title: retsifp
  meters:
    grid: grid1
    pv:
    - pv2
    - pv4
    battery:
    - battery3
```

### Log details

```text
thomas@eq12:~$ evcc --database ~/.evcc/evcc.db
[main  ] INFO 2025/04/19 18:09:26 evcc 0.203.1
[main  ] INFO 2025/04/19 18:09:26 using config file: /home/thomas/evcc.yaml
[db    ] INFO 2025/04/19 18:09:26 using sqlite database: /home/thomas/.evcc/evcc.db
[eebus ] INFO 2025/04/19 18:09:26 Local SKI: f50e32f9f5a975d9fc767ce094d18ba8cc07ef4f
[eebus ] DEBUG 2025/04/19 18:09:26 starting websocket server on :4712
[eebus ] DEBUG 2025/04/19 18:09:26 mdns: announce
[eebus ] DEBUG 2025/04/19 18:09:26 mdns: using avahi
[main  ] INFO 2025/04/19 18:09:26 listening at :7070
[lgess ] DEBUG 2025/04/19 18:09:27 re-login
[main  ] FATAL 2025/04/19 18:09:27 meter [pv4] cannot create meter 'pv4': cannot create meter type 'template': cannot create meter type 'lgess8': uri mismatch: https://192.168.2.92 vs https://192.168.2.49
[main  ] FATAL 2025/04/19 18:09:27 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.1

andig, 2025-04-20:
Indeed, currently that‘s a singleton but not too difficult to change.

closed 2025-04-20 by andig

retsifp, 2025-04-23:
Wow, that was so fast. Thank you guys very much! It works like a charm now 🥳

PR #20731 (2025-04-20): LG ESS: support multiple instances
Fix #20720