evcc-io/evcc #20721: FR: diverse Parameter per source änderbar machen

VolkerK62, 2025-04-19:
Es gibt diverse Parameter, die können zwar per API geändert werden, müssen aber in der Konfiguration als fester Wert definiert werden.
Um diese situationsbedingt ändern zu können ist dann immer eine "externe" Aktion nötig.
Das könnte oft verhindert werden, wenn diese Parameter (ebenfalls) per source definiert werden könnten.

Ein Beispiel:
Wenn meine Heizung startet, braucht sie ca. 10 Minuten lang 1000W. Mein normales disable.delay steht auf 3 Minuten. Im Grenzbereich wird die Ladnung dann gestoppt, um ein paar Minuten später wieder zu starten.
Da das ein absehbarer Zusatzverbrauch ist, habe ich mir eine externe Logik gebaut, die in dem Fall das disable.delay per API erhöht, um es danach wieder auf den alten Wert einzustellen.
Die Heizung ist als ext-Meter konfiguriert. Somit steht die Power in /api/state zur Verfügung. Wenn disable.delay als source zur Verfügung stünde, könnte man diese Logik dort unterbringen und die externe Aktion wäre unnötig.

Das würde die Gestaltungsmöglichkeiten auch auf die Nutzer erweitern, die kein HomeAssist/NodeRed/ o.ä. zur Verfügung haben.

Um es nciht für alle kompliziert zu machen, sollte die Eingabe fester Werte weiterhin möglich sein.
Evtl. lässt sich das ja analog Lastmanagement (GetMaxPower) umsetzen.

Leider habe ich keine Ahnung, wie groß der Aufwand dafür ist und ob Kosten/Nutzen dann noch angemessen sind.

andig, 2025-04-20:
Was meinst du mit „als source verfügbar“ und was sind die „diversen parameter“ ausser dem erwähnten?

VolkerK62, 2025-04-20:
>Was meinst du mit „als source verfügbar“ 

bleiben wir mal bei dem Beispiel von oben
dann sähe das so aus
```
enable:
  delay:
    source: http
    uri: localhost:7070/api/state
    jq: if .result.ext[0].power >400 then 15m else 3m end
```

>was sind die „diversen parameter“ ausser dem erwähnten?

Jetzt wo du so fragst und ich die API Doku angesehen habe, fällt mir auf, dass gar nicht mehr viel da, weil immer mehr ins UI "gewandert" ist und in der Konfig gar nicht mehr definierbar .... bleiben eigentlich nur noch
- enable.threshold
- enable. delay
- disable.threshold
- disable.delay
- residualpower

Hätte die Idee wohl vorher mal besser durchdenken sollen 😄

andig, 2025-04-20:
Mir ist nicht so richtig klar, wofür Du das brauchst (abgesehen davon, dass es nicht zum config-ui passt): für alle diese Werte gibts ja schon APIs die den gleichen Zweck erfüllen. Ich denke wontfix?

closed 2025-04-20 by andig

VolkerK62, 2025-04-20:
Ja, es passt nicht zum config-UI.
APIs müssen aber irgendwie von "extern" geschickt werden oder durch manuelle Änderungen im UI.
Idee war, diese Möglichkeit innerhalb von evcc zu automatisieren, für einfache Logiken.

andig, 2025-04-20:
Ich verstehe den Punkt, ist mir aber ehrlich gesagt zuviel Magie und das echte Einsatzfeld zu begrenzt. Wer soviel Eingriff möchte hat oft auch Automatisierung im Einsatz.