evcc-io/evcc #20735: sgready um limitsoc bzw. limittemp erweitern

Tombra1889, 2025-04-20:
**Is your feature request related to a problem? Please describe.**
Aktuell nutze ich für die Wärmepumpe das custom-Template. Gerne würde ich aber auf das sgready Template wechseln, um mir insb. die Fehlermeldung "charger out of sync: expected disabled, got enabled" zu ersparen.

Zur Visualisierung verwende ich soc und limitsoc.

[image]

Im sgready-Template gibt es statt soc das temp-Attribut, aber leider keines für limitsoc "limittemp", welches ich ansprechen könnte.

Es geht mir hier vorwiegend um die Visualisierung. Evtl. könnte in den Zuge auch minsoc "mintemp" mit implementiert werden.

Tombra1889, 2025-04-20:
Eine kleine Anregung noch: bzgl. set-/getmode sind die Zustände aktuell:

1 Normalbetrieb
2 Boostbetrieb
3 reduzierte Leistung

Laut sgready-Vorgabe wäre die Reihenfolge:

1 reduzierte Leistung
2 Normalbetrieb
3 Boostbetrieb 
(https://www.waermepumpe.de/normen-technik/sg-ready/)

andig, 2025-04-21:
> Eine kleine Anregung noch: bzgl. set-/getmode sind die Zustände aktuell:

@premultiply sollen wir das flächendeckend einmal ändern?

andig, 2025-04-21:
> Im sgready-Template gibt es statt soc das temp-Attribut, aber leider keines für limitsoc "limittemp", welches ich ansprechen könnte.

@Tombra1889 genau das gibt es, ist also schon erledigt.

closed 2025-04-21 by andig

premultiply, 2025-04-21:
> > Eine kleine Anregung noch: bzgl. set-/getmode sind die Zustände aktuell:
> 
> @premultiply sollen wir das flächendeckend einmal ändern?

Können wir gern machen.
Viel Klarheit von sich aus wird es aber leider nicht geben, da es keine Norm ist sondern nur ein Empfehlung (jetzt in zwei Versionen) und die Realität auch häufig nochmal von beiden abweicht. Bis es neue Geräte gibt die die neue Version ansatzweise Umsetzen dürfte es noch dauern.

Aber für den reduzierten Betrieb mit §14a EnWG gibt es nun zumindest schon mal einen definierten Ansatz. Das ist gut. 👍🏼

andig, 2025-04-21:
Top- das wird dann https://github.com/evcc-io/evcc/issues/20746