evcc-io/evcc #20741: OCPP: Support for AuthEnabled: false AuthEnabledOffline: false

poml88, 2025-04-20:
**Is your feature request related to a problem? Please describe.**
Kind of. Some chargers need authentification via RFID e.g. when in OCPP mode.

**Describe the solution you'd like**
Implement the OCPP commands
```
AuthEnabled: false / true
AuthEnabledOffline: false / true
```
**Describe alternatives you've considered**
Cannot think of any. Would like to avoid the `remotestart: true`.

**Additional context**
None.

premultiply, 2025-04-21:
Nothing to do here.
Everything is implemented.

Swipe your card, locally setup your charger correctly or use `remotestart: true` to use remote transactions. There are no other options in OCPP.

closed 2025-04-21 by premultiply

poml88, 2025-04-21:
Sorry, there are: see the two commands above. 
The Wallbox is correctly setup as „free“.
But there are various models who require authentication then in ocpp mode. For those are those commands. It was previously also discussed here. 
Here in the repository people are often discouraged from using remotestart: true. 
Sorry, fact, not fantasy.

poml88, 2025-04-21:
Please reconsider.

premultiply, 2025-04-21:
These configuration keys are not part of the OCPP spec.

poml88, 2025-04-23:
Ok, thanks. I was referring to this:
https://support.wallbox.com/na/knowledge-base/ocpp-activation-and-setup-guide/#:~:text=AuthEnabled%20in%20CSMS

premultiply, 2025-04-23:
It may not be the best idea to look for OCPP details at Wallbox.com of all places... 😅

What you are looking for needs to be setup in your local charger configuration or set `remotestart`: true in evcc configuration. These are the only two ways according to the OCPP spec.