evcc-io/evcc #20754: Niu Abfrage aktuellen km stand

Michel83, 2025-04-21:
Frage nach Möglichkeit der km-Abfrage wie beim PKW zur weitern Auswertung.

Datenpunkt der Aktuellen km ist unter:

```
 "/motoinfo/overallTally": {
			"post": {
				"summary": "Returns the odometer value and days of usage",
				"requestBody": {
					"required": true,
					"content": {
						"application/x-www-form-urlencoded": {
							"schema": {
								"type": "object",
								"properties": {
									"sn": {
										"$ref": "#/components/schemas/MopedDesc/properties/sn"
									},
									"token": {
										"$ref": "#/components/schemas/Token"
									}
								}
							}
						}
					}
				},
				"responses": {
					"200": {
						"description": "Odometer and time value",
						"content": {
							"application/json": {
								"schema": {
									"type": "object",
									"properties": {
										"data": {
											"type": "object",
											"properties": {
												"bindDaysCount": {
													"type": "integer"
												},
												"totalMileage": {
													"type": "integer"
												}
											}
										}
									}
								}
							}
						}
					}
				}
			}
		},

```

```
const API_BASE_URL = 'https://app-api-fk.niu.com';
    const ACCOUNT_BASE_URL = 'https://account-fk.niu.com';

    public static function get_token($email, $password, $country_code) {
        $data = new UrlRequest(self::ACCOUNT_BASE_URL . '/appv2/login', array('account' => $email, 'countryCode' => $country_code, 'password' => $password));
        if (is_object($data) && property_exists($data, 'response') && property_exists($data->response, 'data') && property_exists($data->response->data, 'token')) {
            return $data->response->data->token;
        }
        return false;
    }

    public static function get_vehicles() {
        if (self::$token === null)
            return false;
        return self::return_response_data(new UrlRequest(self::API_BASE_URL . '/motoinfo/list', [], self::$token));
    }

    public static function get_motor_info() {
        if (self::$token === null || self::$serial_no === null)
            return false;
        return self::return_response_data(new UrlRequest(self::API_BASE_URL . '/v3/motor_data/index_info?sn=' . self::$serial_no, null, self::$token));
    }

    public static function get_overall_tally() {
        if (self::$token === null || self::$serial_no === null)
            return false;
        return self::return_response_data(new UrlRequest(self::API_BASE_URL . '/motoinfo/overallTally', ['sn' => self::$serial_no], self::$token));
    }
```

`data":{"bindDaysCount":632,"totalMileage":1061.464},"desc":"Success","trace":"Success","status":0`

[image]

zu finden.

Leider habe ich es selbst nicht hin bekommen die niu.go so zu editieren das es Funktioniert.

Grüße

andig, 2025-04-21:
Es braucht bitte einfach den HTTP Request, keine OpenApi Definition davon. PR welcome.

closed 2025-04-21 by andig

thierolm, 2025-04-21:
Soweit ich das oben erkennen kann, müsste der Request wie folgt aussehen:

Request: 
`https://app-api-fk.niu.com//motoinfo/overallTally?sn=123xxx45`

Response:
```json
{
  "data": {
    "bindDaysCount": 632,
    "totalMileage": 1061.464
  },
  "desc": "Success",
  "trace": "Success",
  "status": 0
}
```

andig, 2025-04-21:
Ich möchte nicht raten. Wenn wir den Request haben gehts schnell. Danke.

Michel83, 2025-04-28:
Hi,

ich glaube die Adresse des Https stimmt nicht es müsst sein:

Da auch keine km geschrieben werden.

[image]

```
{
    "id": "fa50ac1b.e215e",
    "type": "http request",
    "z": "4b4281a5.b723e8",
    "name": "API CALL",
    "method": "POST",
    "ret": "obj",
    "paytoqs": false,
    "url": "https://app-api.niu.com/motoinfo/overallTally",
    "tls": "",
    "proxy": "",
    "authType": "basic",
    "x": 568,
    "y": 660,
    "wires": [
        [
            "21002b84.7f89fc",
            "aabb60e.21b592"
        ]
    ]
```

Grüße

thierolm, 2025-04-28:
@Michel83 da es keinen Fehler gibt nutzen uns die Bilder und JSon Teile ohne Quellenangabe nichts!

Bitte Ausgabe von `evcc -l trace vehicle` posten.

Michel83, 2025-04-28:
Kein Problem:

```
NiuRoller
---------
[niu   ] TRACE 2025/04/28 14:21:02 POST https://account-fk.niu.com/v3/api/oauth2/token
[niu   ] TRACE 2025/04/28 14:21:02 account=***&app_id=niu_8xt1afu6&grant_type=password&password=ba780hhhhhhhhhhbc8cde6d8e3a9b7&scope=base
--
{"data":{"token":{"access_token":"eyJhbmtpZCI6Ikg3VUlKUXhVMlE0NW5uNVNIeWM1MTNLemNKR1RRbU41eGJZZyIsInR5cCI6IkpXVCJ9.eyJhdWQiOiIyNjB2QlZBSmZnUUl2ZWc3QUFDNVdTaElzVEdkTWRNZjFsOXFSamRKN3l1MmQ1MkxuVG13bHVBaGswUG5YREFUIiwiZXhwIjoxNzQ4NDM0ODYyLCJpYXQiOjE3NDU4NDI4NjI2OTgxNjM2MzgsInN1YiI6InM3MzJtM3U2eDc5YWZ6ejQxdHV6Y2NrZCJ9.Rldd78v3y1YlF4K_t1XhrvZK9iD8kX6-Khxjwgl7EUQwGpygeoCgpQHp8H7iKQCklCK1oKMBnvSHU7UCQRvqnQ","refresh_token":"MDC0KNDKTODCZMI01ODC2LWFLNZMTZTG1MDRKZMM1ZTHK","refresh_token_expires_in":1748434862,"token_expires_in":1748434862},"user":{"user_id":"s732m3u6x79afzz41tuzcckd","mobile":"1791401519","email":"***","country_code":"49","nick_name":"Mike","real_name":"MichaelFerber","last_name":"","identification_code":"","birthdate":"","gender":0,"avatar":"https://s.niucache.com/static/user_image/avater.jpg","thumb_avatar":"https://store-v3.niucache.com/2005b202004291708046291.jpg?x-oss-process=image/resize,m_fill,h_400,w_400","profession":0,"income":0,"car_owners":0,"purpose":"","sign":"A NIU WAY FORWARD","background":"https://download.niucache.com/static/upload/20200507/57f16705-48e9-4080-8246-b2fde8df30b5.jpg","height":0,"weight":0}},"desc":"ok","status":0}
[niu   ] TRACE 2025/04/28 14:21:02 GET https://app-api-fk.niu.com/v3/motor_data/index_info?sn=NT2B89699917AKNNJ
[niu   ] TRACE 2025/04/28 14:21:03 {"data":{"isCharging":0,"lockStatus":0,"isAccOn":0,"isFortificationOn":1,"isConnected":true,"postion":{"lat":50.588444,"lng":8.462152},"hdop":0,"time":1745842677078,"batteries":{"compartmentA":{"bmsId":"BN72CY2G91200048","isConnected":true,"batteryCharging":97,"gradeBattery":"96.6"},"compartmentB":{"bmsId":"BN72CY2G91200165","isConnected":true,"batteryCharging":95,"gradeBattery":"96.7"}},"leftTime":"0.0","estimatedMileage":88,"gpsTimestamp":1745842677078,"infoTimestamp":1745842677078,"nowSpeed":0,"shakingValue":3,"locationType":1,"batteryDetail":true,"centreCtrlBattery":94,"ss_protocol_ver":3,"ss_online_sta":"1","gps":0,"gsm":20,"lastTrack":{"ridingTime":87,"distance":636,"time":1745244231988}},"desc":"成功","trace":"成功","status":0}
[niu   ] TRACE 2025/04/28 14:21:03 GET https://app-api-fk.niu.com/overallTally?sn=NT2B89699917AKNNJ
[niu   ] TRACE 2025/04/28 14:21:03 404: Page not Found
Soc:           97%
Capacity:      4.0kWh
Charge status: B
Range:         88km
Odometer:      unexpected status: 404 (Not Found)
```

thierolm, 2025-04-28:
@Michel83 kannst du zeitnah das nightly Release testen, wenn der PR gemerged ist?

Michel83, 2025-04-30:
Kurzes update, bekomme das testen leider erst am Wochenende hin.

Grüsse

Michel83, 2025-05-04:
Geht leider nicht.

```
[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/vehicle/vehicle_2/status

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/vehicle/vehicle_3/status

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/vehicle/vehicle_4/status

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/vehicle/vehicle_5/status

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/vehicle/db:1/status

[niu ] TRACE 2025/05/04 15:34:04 GET https://app-api-fk.niu.com/motoinfo/overallTally?sn=NT2L4AB8AKNNJ

[niu ] TRACE 2025/05/04 15:34:04 404: Page not Found

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/charger/charger_1/status

[httpd ] TRACE 2025/05/04 15:34:04 GET /api/config/devices/charger/charger_2/status

)
```

Die Aktuell Version:

[image]

Grüsse

thierolm, 2025-05-04:
Hm, URI stimmt, dann müsste man mal die App tracen. Wo hattest du denn deine Screenshots oben her? Gibt es ein Git Projekt, dass dir die Daten wie oben gezeigt ausliest? Wenn ja bitte teilen, dann können wir dich unterstützen, um dessen Logik zu tracen ...

Michel83, 2025-05-04:
Also die Daten stammen aus iobroker da gibt es ein Adapter für:

https://github.com/TA2k/ioBroker.niu

Hier wurde es vor zwei Wochen eingefügt, beim tracen über iobroker sehe ich leider den Http Aufruf nicht.

Was ich noch gefunden habe ist das:

```
// Response is the Niu motor_data api response
// https://app-api-fk.niu.com/v3/motor_data/index_info?sn=<ScooterSerialNumber>
type Response struct {
	Data struct {
		IsCharging  int   `json:"isCharging,omitempty"`
		IsConnected bool  `json:"isConnected,omitempty"`
		Timestamp   int64 `json:"time,omitempty"`
		Batteries   struct {
			CompartmentA struct {
				BatteryCharging int64 `json:"batteryCharging,omitempty"`
			} `json:"compartmentA"`
		}
		// LeftTime         float32 `json:"leftTime,omitempty"`
		EstimatedMileage int64 `json:"estimatedMileage,omitempty"`

		// overallTally
		TotalMileage float64 `json:"totalMileage,omitempty"`
	}
}

```

thierolm, 2025-05-04:
Wo hast du den letzten Auszug her?

thierolm, 2025-05-04:
Laut IOBroker Projekt ist die von uns genutzte overTally URL die Richtige:
https://github.com/TA2k/ioBroker.niu/blob/bb6cc29dc1cea582643854c7b0183f3efc1aa6f4/main.js#L209

thierolm, 2025-05-04:
Via `curl` komme ich auch auf die overTally Url:
```bash
xxx@xxx:~/git-repos/evcc$ curl -X POST "https://app-api-fk.niu.com/motoinfo/overallTally" \
  -H "Content-Type: application/json" \
  -H "Accept: application/json" \
  -H "Authorization: Bearer DEIN_TOKEN_HIER" \
  -d '{
    "sn": "DEINE_SERIENNUMMER_HIER"
  }'
{"data":"","desc":"登录信息错误","trace":"Fail!TOKEN ERROR","status":1131}
```

Michel83, 2025-05-04:
Sorry das war Kopie misst von mir........ ich hatte die url´s durchprobiert wo ich eine Antwort bekomme.

Michel83, 2025-05-04:
`niu.0 | 2025-05-04 16:44:04.461 | debug | {"data":{"bindDaysCount":648,"totalMileage":1089.398},"desc":"Success","trace":"Success","status":0}
`

mehr kommt leider nicht im debug mode

thierolm, 2025-05-04:
ok, Problem erkannt, wir fragen mit der `GET` Methode ab, müssen aber mit der `POST` Methode arbeiten.

thierolm, 2025-05-04:
ich muss jetzt los, versuche einen fix PR asap, evtl. ist @andig schneller ;-)

Michel83, 2025-05-04:
Kein Problem, vielen Dank!!!

Grüße

thierolm, 2025-05-04:
@Michel83 kannst du evtl https://github.com/evcc-io/evcc/pull/21051/commits/46b6a69e459b12b363c75d167f2c4a4edd5f1ee2 testen?

Michel83, 2025-05-05:
Klar, heute abend. Grüsse

Michel83, 2025-05-05:
```
NiuRoller
---------
[niu   ] TRACE 2025/05/05 12:13:17 POST https://account-fk.niu.com/v3/api/oauth2/token
[niu   ] TRACE 2025/05/05 12:13:17 account=***&app_id=niu_8xt1afu6&grant_type=password&password=ba780f56d8e3a9b7&scope=base
--
{"data":{"token":{"access_token":"eyJhbGciOiJIUzUxMiIsImtpZCI6IjhUekVtMnI2S0swa0IwcDJNRE51WXVnOU5oMVJqd0YwRkk4cyIsInR5cCI6IkpXVCJ9.eyJhdWQiOiI3QzB0Q2JueWJkcEY2Qnp6NTBndVloU0c2YN1YiI62eDc5YWZ6ejQxdHV6Y2NrZCJ9.rauBU8YTvMF23BoTfe4WdMF-j1GtCNV03uCY4kSZvyAZyZsbrepQayICBheL65MWXCRJwwzoCdk1Jc31XHA","refresh_token":"ZMQ3ETYWJMYI01Y2Q3LTKYYWYTZDNKODMZGUY","refresh_token_expires_in":1749031997,"token_expires_in":1749031997},"user":{"user_id":"s732m3ucckd","mobile":"1","email":"***","country_code":"49","nick_name":"","real_name":"","last_name":"","identification_code":"","birthdate":"","gender":0,"avatar":"https://s.niucache.com/static/user_image/avater.jpg","thumb_avatar":"https://store-v3.niucache.com/2005b8046291.jpg?x-oss-process=image/resize,m_fill,h_400,w_400","profession":0,"income":0,"car_owners":0,"purpose":"","sign":"A NIU WAY FORWARD","background":"https://download.niucache.com/static/upload/20200507/57f16705-48e9-4080-8246-b2fde8df30b5.jpg","height":0,"weight":0}},"desc":"ok","status":0}
[niu   ] TRACE 2025/05/05 12:13:17 GET https://app-api-fk.niu.com/v3/motor_data/index_info?sn=NTNJ
[niu   ] TRACE 2025/05/05 12:13:18 {"data":{"isCharging":0,"lockStatus":0,"isAccOn":0,"isFortificationOn":1,"isConnected":true,"postion":{"lat":,"lng":},"hdop":0,"time":1746439949752,"batteries":{"compartmentA":{"bmsId":"BN720048","isConnected":true,"batteryCharging":88,"gradeBattery":"96.6"},"compartmentB":{"bmsId":"BN791200165","isConnected":true,"batteryCharging":89,"gradeBattery":"96.7"}},"leftTime":"0.0","estimatedMileage":80,"gpsTimestamp":174649752,"infoTimestamp":17469752,"nowSpeed":0,"shakingValue":3,"locationType":1,"batteryDetail":true,"centreCtrlBattery":94,"ss_protocol_ver":3,"ss_online_sta":"1","gps":2,"gsm":20,"lastTrack":{"ridingTime":54,"distance":235,"time":1745945856449}},"desc":"成功","trace":"成功","status":0}
[niu   ] TRACE 2025/05/05 12:13:18 POST https://app-api-fk.niu.com/motoinfo/overallTally
[niu   ] TRACE 2025/05/05 12:13:19 sn=NT2L4AB8917AKNNJ
--
{"data":"","desc":"参数错误","trace":"参数错误","status":1301}
Soc:           88%
Capacity:      4.0kWh
Charge status: B
Range:         80km
Odometer:      json: cannot unmarshal string into Go struct field Response.Data of type struct { IsCharging int "json:\"isCharging,omitempty\""; IsConnected bool "json:\"isConnected,omitempty\""; Timestamp int64 "json:\"time,omitempty\""; Batteries struct { CompartmentA struct { BatteryCharging int64 "json:\"batteryCharging,omitempty\"" } "json:\"compartmentA\"" }; EstimatedMileage int64 "json:\"estimatedMileage,omitempty\""; TotalMileage float64 "json:\"totalMileage,omitempty\"" }
```

Grade auf die neuste Version aktualisiert

thierolm, 2025-05-05:
ok, der Post Ufruf funktioniert, es hapert noch an der Parameter Übergabe.

thierolm, 2025-05-05:
Bist du in der evcc Slack Gruppe? Es wäre sehr hilfreich, wenn du mir deine Niu App Credentials zur Entwicklung einmal geben könntest, dann müssten wir nicht über Git in mehreren Anläufen testen.

Michel83, 2025-05-05:
Nein in der Gruppe bin ich nicht aber kann ich gerne zur Verfügung stellen , im Prinzip brauchst ja nur die Zugangsdaten ?

thierolm, 2025-05-05:
> Nein in der Gruppe bin ich nicht aber kann ich gerne zur Verfügung stellen , im Prinzip brauchst ja nur die Zugangsdaten ?

ja, Zugangsdaten reichen.

thierolm, 2025-05-05:
Alternativ, wenn du selbst den evcc Master mit `make` bauen kannst, ändere die `post` Methode in `niu.go` wie folgt und teste das Vehicle:
```go
func (v *Niu) post(uri string) (*http.Request, error) {
	data := url.Values{
		"sn":    {v.serial},
		"token": {v.token.AccessToken},
	}
	return v.newRequest(http.MethodPost, uri, strings.NewReader(data.Encode()))
}
```
Ich habe nur noch das token ergänzt.

PR #20925 (2025-04-28): Niu: Correct odometer endpoint
Fixes #20754

## Summary by Sourcery

Bug Fixes:
- Fix incorrect API endpoint for Niu odometer readings.