evcc-io/evcc #20756: Lademodus schnell ohne Netz (nur PV + Batteriespeicher)

taran-2, 2025-04-21:
**Is your feature request related to a problem? Please describe.**
Ich möchte die Möglichkeit haben, mein E-Auto so schnell wie möglich zu laden, ohne Netzstrom zu beziehen.

**Describe the solution you'd like**
Weiterer Lademodus, der auf PV + Batteriespeicher zurückgreift, jedoch keinen Strom aus dem Netz bezieht.

**Describe alternatives you've considered**
Ich könnte den Lademodus "schnell" mittels Konfiguration dahin abändern, dass er Strombezug aus dem Netz unterlässt. Dann hätte ich jedoch das Problem, dass ich nicht mehr mit voller Geschwindigkeit laden kann, wenn dies mal notwendig ist. Eine weitere Lösung wäre, wenn man für den Lademodus "schnell" in der App mit einem Schalter definieren könnte, ob Strom aus dem Netz bezogen werden soll oder nicht.

TobiasHuber1980, 2025-04-21:
Gibt es indirekt schon, https://docs.evcc.io/docs/features/battery#batterie-boost

taran-2, 2025-04-21:
Danke dir! Ich hatte experimentelle UI ausgeblendet, daher wurde mir das bisher nicht angezeigt. Dann werde ich das Feature demnächst testen.

closed 2025-04-21 by andig

premultiply, 2026-04-14:
Nein, da technisch kaum anders lösbar und auch am Ende finanziell quasi irrelevant.

Je nach Phasenkonfiguration beim Laden wird sich ein restlicher Netzbezug von ca. 100 W im Battery Boost Modus nicht vermeiden lassen. Im 3p-Modus ggf. deutlich mehr.