evcc-io/evcc #20759: Fehlermeldung BYD HVS timeout

haegele100, 2025-04-21:
### Describe the bug

Hallo, seit dem Update auf 0.203 habe ich ein Problem mit dem Akku. In unregelmäßigen Abständen kommt eine Fehlermeldung timeout.
Kostal hat mit einem update die zeit für den timeout der externen Batteriesteuerung von max 180ms auf 60ms gesenkt. Evtl. hat es auch damit zu tun.

[evcc-20250421-180622-debug.log](https://github.com/user-attachments/files/19836601/evcc-20250421-180622-debug.log)

### Steps to reproduce

1. unklar
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
#  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
#  host: 192.168.178.54
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
#  port: 7070

# interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken: XXXXX

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true
plant: cc8d5be2ecfa24ad5d7f0ab8858dc67f2d769cc0db3e468a08afefb3da7af57d

# log settings
# log: info
# levels:
#  site: debug
#  lp-1: debug
#  lp-2: debug
#  cache: error
#  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true
[... 871 lines cut ...]
[site  ] DEBUG 2025/04/21 18:06:16 battery 1 soc: 98%
[site  ] DEBUG 2025/04/21 18:06:16 pv 1 power: 1091W
[site  ] DEBUG 2025/04/21 18:06:16 pv power: 1851W
[site  ] DEBUG 2025/04/21 18:06:16 site power: 76W
[lp-2  ] DEBUG 2025/04/21 18:06:16 charge total import: 17.849kWh
[lp-2  ] DEBUG 2025/04/21 18:06:17 charger status: B
[lp-2  ] DEBUG 2025/04/21 18:06:17 pv charge current: 0A = 0A + -0.332A (76W @ 1p)
[site  ] DEBUG 2025/04/21 18:06:17 solar forecast: accumulated 0.076kWh, produced 0.134kWh, scale 1.769
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.1

andig, 2025-04-21:
> I have verified that the issue is reproducible with the latest nightly build

Hast du?

haegele100, 2025-04-22:
Ich habe fast alles über die Config-UI gemacht. Ich habe das jetzt auch bei der nightly gemacht. 
Der Fehler tritt in der nightly nicht auf!

closed 2025-04-22 by andig