evcc-io/evcc #20760: Wärmeerzeuger: Support von Alpha Innotec (ait) Wärmepumpen, Modbus TCP & SG ready

lorn10, 2025-04-21:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I think it would be very useful if at some point also Alpha Innotec (ait) heat-pumps could be supported in evcc. (In Central Europe those are quite widely used.)

**Additional context**

Hallo an alle
An dieser Stelle folgt ein feature request bezüglich der Unterstützung von Alpha Innotec Wärmepumpen in evcc. (Das wird im evcc Wiki so empfohlen.)

Ab firmware **V3.90.3** ist im Rahmen des Smart-Home-Interface ein einfacher **Modus TCP support** enthalten. So weit ich weiss ist das nun bei allen ait Modellen welche über die Luxtronik 2.1 Steuerung verfügen der Fall. Ausgenommen sind die älteren Modelle mit Luxtronik 2.0 Steuerung. Und vermutlich werden mit Stand April 2025 auch die ait Wärmepumpen welche von NIBE sind noch nicht über diese Funktion verfügen. Diese Modelle haben eine "Emmy" genannte Steuerung. Ich denke aber das auch bei denen ein Smart-Home-Interface mit einfachem Modus TCP support noch kommen wird. Die neusten NIBE Modelle der S Serie unterstützen Modbus bereits ab Werk, alle anderen brauchen (momentan) noch ein externes Modul, dass [Nibe Modbus 40](https://talotuote.fi/p45605/nibe-modbus-40).

Wie auch immer, die entsprechenden ait Modbusregister wurden auf folgender Seite vorbildhaft dokumentiert:
https://github.com/raibisch/mylibs/blob/main/LuxModbusSHI/Modbusregister.md

Eine Anleitung wie Modbus TCP auf der ait Wärmepumpe eingerichtet wird gibt es hier:
https://github.com/raibisch/mylibs/blob/main/LuxModbusSHI/LuxtronikSHI.md

Ganz vielen Dank an dieser Stelle an @raibisch für all die Infos. Sein Projekt findet man unter:
https://github.com/raibisch/ESP32_Luxtronik2

Mehr Infos gibt es zudem im Haustechnik Dialog Forum thread [Eigene Regelung PV + Luxtronik 2.1 Smart Home Interface (SHI)](https://www.haustechnikdialog.de/Forum/t/284442/Eigene-Regelung-PV-Luxtronik-2-1-Smart-Home-Interface-SHI)

Zu guter Letzt, ait Wärmepumpen (inklusive jenen welche von NIBE kommen) verfügen auch über **SG ready** Unterstützung. Soweit ich weis ist das schon lange der Fall. Somit könnte in Verbindung mit ait also auch nur ein SG ready support in evcc hinzugefügt werden, - wenn das einfacher geht.

waja, 2025-05-17:
„Completed“ means supported or just closed without implementation?

andig, 2025-05-17:
> https://github.com/evcc-io/evcc/issues/20760#issuecomment-2819159159

andig, 2025-05-17:
See https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md#device-templates for testing templates.

phagemann, 2025-05-17:
I'm working on a PR.
The heatpump requires writing 2 separate registers, in order to accept an increased temperature or accept a power limit via the SmartHomeInterface. 
At this point i don't know how to solves this via templates.

I expect a dedicated Go implementation would be required. As i am new to evcc, I would like to have a second opinion whether this is solvable through templates or not.
In case templates are not an option, I would switch to Go implementation.

andig, 2025-05-17:
Feel free to open the PR so we can discuss what exactly is missing.

> The heatpump requires writing 2 separate registers

maybe use the `sequence` plugin?

lorn10, 2025-06-22:
For the record, the support regarding Luxtronik 2.1 based heat-pumps was finally added with merge request https://github.com/evcc-io/evcc/pull/21516.

Big thanks to @daniel309 for his great work in this topic! 👍